DSheirer/sdrtrunk #1547: Playlist editor stops writing to default.xml and Teardowns happen afterward

tylerwatt12, 2023-05-12:
**sdrtrunk Version**
0.5.3

**Describe the bug**
When I make large changes to aliases, at some point the default.xml file will stop updating until I completely reopen sdrtrunk
When I go to close and re-open the program, none of the changes I made since it stopped writing are displayed.
Also noticed Teardowns start to occur after default.xml stops being written to

See video https://youtu.be/CWN9M_ATvEQ?t=390
(Start at 6:30)

This video shows what happens a little clearer 
https://www.youtube.com/watch?v=DjuumOdbKrs

**To Reproduce**
Import Ohio MARCS-IP from RRDB, 
Import all Talkgroups
Delete talkgroups for all except Lorain and Cuyahoga County.
At some point something internally breaks and default.xml isn't updated

**Expected behavior**
Default.xml updates

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Application Log**
If applicable, include either the full application log,or a snippet of the error logging.

**Desktop (optional - complete the following information):**
 - OS: ESXI 8, Windows 11 Pro LTSC VM
 - CPU Cores: Intel Xeon E5-2697 v4 16 cores on VM
 - RAM: 16GB on VM

**Additional context**
I initially discovered that my XML file wasn't saving it while trying to parse it. I then noticed the teardowns that would occur when I do anything CPU/File intensive. I'm hoping this helps track down the teardown issue

DSheirer, 2023-05-15:
@tylerwatt12 I tested this by importing the MARCS talkgroups into an empty playlist and removing a large chunk of those talkgroups as you suggested, but I'm not able to reproduce any error or to cause the playlist to not be saved.

There may be something else going on ... channel teardown issues normally wouldn't be related to playlist save issues, but they could both happen if/when the application is in a degraded state, such as if it runs out of heap space or if there is an I/O error.

If you can recreate the scenario/symptoms, can you please send me the application log and we can see if there were any errors logged that might be related to the symptoms you're seeing?

tylerwatt12, 2023-05-15:
I was able to reproduce it, but I don't get anything out of the log when that happens. 

I did find the logs from the previous video I uploaded

https://www.youtube.com/watch?v=DjuumOdbKrs

[20230512 113536.847 [main] INFO  i..txt](https://github.com/DSheirer/sdrtrunk/files/11478129/20230512.113536.847.main.INFO.i.txt)

Does sdrtrunk have more verbose debug options?

If you want remote access to this VM to test/debug I can accommodate.

DSheirer, 2023-05-21:
@tylerwatt12 I updated the alias editor to make add/delete operations more efficient.  I'm not sure if this was contributing to the playlist save failure that you're seeing or not.  However, I did notice with that system in partucular that it was taking a long time to both add and to delete.  I'll merge these changes to master and close this issue for now.  However, if the playlist save issue continues, we can reopen this issue and look into it some more.

PR #1560 (2023-05-21): #1547 Playlist Fails To Save When Add/Delete Large Quantity of Aliases
Closes #1547 

Updates playlist alias editor to make add/delete operations more efficient to possibly resolve issue where playlist may not be saving when adding or removing large chunks of aliases.
closed 2023-05-21 by DSheirer