DSheirer/sdrtrunk #1549: Error: OutOfMemoryError

ah1102, 2023-05-14:
There was such an error, after a minute of time, after listening to several receiving channels in AM mode. It probably happens in other modes as well. The 20 MHz device bandwidth I was using made the error appear faster.

System: macOS Catalina
Device: HackRF One (in 20 MHz band mode)
Program: sdr-trunk-0.6.0-alpha6

```
Exception in thread "AWT-EventQueue-0" java.lang.OutOfMemoryError: Java heap space
	at java.desktop/java.awt.image.DataBufferInt.<init>(DataBufferInt.java:76)
	at java.desktop/java.awt.image.Raster.createPackedRaster(Raster.java:539)
	at java.desktop/java.awt.image.DirectColorModel.createCompatibleWritableRaster(DirectColorModel.java:1032)
	at java.desktop/java.awt.image.BufferedImage.<init>(BufferedImage.java:351)
	at java.desktop/sun.java2d.loops.GraphicsPrimitive.convertFrom(GraphicsPrimitive.java:568)
	at java.desktop/sun.java2d.metal.MTLGeneralBlit.Blit(MTLBlitLoops.java:781)
	at java.desktop/sun.java2d.SurfaceDataProxy.updateSurfaceData(SurfaceDataProxy.java:499)
	at java.desktop/sun.java2d.SurfaceDataProxy.replaceData(SurfaceDataProxy.java:456)
	at java.desktop/sun.java2d.SurfaceData.getSourceSurfaceData(SurfaceData.java:235)
	at java.desktop/sun.java2d.pipe.DrawImage.renderImageScale(DrawImage.java:634)
	at java.desktop/sun.java2d.pipe.DrawImage.tryCopyOrScale(DrawImage.java:321)
	at java.desktop/sun.java2d.pipe.DrawImage.transformImage(DrawImage.java:260)
	at java.desktop/sun.java2d.pipe.DrawImage.scaleImage(DrawImage.java:126)
	at java.desktop/sun.java2d.pipe.DrawImage.scaleImage(DrawImage.java:1077)
	at java.desktop/sun.java2d.SunGraphics2D.drawImage(SunGraphics2D.java:3383)
	at java.desktop/sun.awt.image.ImageRepresentation.drawToBufImage(ImageRepresentation.java:845)
	at java.desktop/sun.java2d.pipe.DrawImage.scaleImage(DrawImage.java:1084)
	at java.desktop/sun.java2d.pipe.ValidatePipe.scaleImage(ValidatePipe.java:207)
	at java.desktop/sun.java2d.SunGraphics2D.drawImage(SunGraphics2D.java:3383)
	at java.desktop/sun.java2d.SunGraphics2D.drawImage(SunGraphics2D.java:3321)
	at io.github.dsheirer.spectrum.WaterfallPanel.paintComponent(WaterfallPanel.java:297)
	at java.desktop/javax.swing.JComponent.paint(JComponent.java:1128)
	at java.desktop/javax.swing.JComponent.paintToOffscreen(JComponent.java:5318)
	at java.desktop/javax.swing.RepaintManager$PaintManager.paintDoubleBufferedImpl(RepaintManager.java:1656)
	at java.desktop/javax.swing.RepaintManager$PaintManager.paintDoubleBuffered(RepaintManager.java:1631)
	at java.desktop/javax.swing.RepaintManager$PaintManager.paint(RepaintManager.java:1569)
	at java.desktop/javax.swing.RepaintManager.paint(RepaintManager.java:1336)
	at java.desktop/javax.swing.JComponent._paintImmediately(JComponent.java:5266)
	at java.desktop/javax.swing.JComponent.paintImmediately(JComponent.java:5076)
	at java.desktop/javax.swing.RepaintManager$4.run(RepaintManager.java:878)
	at java.desktop/javax.swing.RepaintManager$4.run(RepaintManager.java:861)
	at java.base/java.security.AccessController.executePrivileged(AccessController.java:776)
Exception in thread "AWT-EventQueue-0" java.lang.OutOfMemoryError: Java heap space
	at java.desktop/java.awt.image.DataBufferInt.<init>(DataBufferInt.java:76)
	at java.desktop/java.awt.image.Raster.createPackedRaster(Raster.java:539)
	at java.desktop/java.awt.image.DirectColorModel.createCompatibleWritableRaster(DirectColorModel.java:1032)
	at java.desktop/java.awt.image.BufferedImage.<init>(BufferedImage.java:351)
	at java.desktop/sun.java2d.loops.GraphicsPrimitive.convertFrom(GraphicsPrimitive.java:568)
	at java.desktop/sun.java2d.metal.MTLGeneralBlit.Blit(MTLBlitLoops.java:781)
	at java.desktop/sun.java2d.SurfaceDataProxy.updateSurfaceData(SurfaceDataProxy.java:499)
	at java.desktop/sun.java2d.SurfaceDataProxy.replaceData(SurfaceDataProxy.java:456)
	at java.desktop/sun.java2d.SurfaceData.getSourceSurfaceData(SurfaceData.java:235)
	at java.desktop/sun.java2d.pipe.DrawImage.renderImageScale(DrawImage.java:634)
	at java.desktop/sun.java2d.pipe.DrawImage.tryCopyOrScale(DrawImage.java:321)
	at java.desktop/sun.java2d.pipe.DrawImage.transformImage(DrawImage.java:260)
	at java.desktop/sun.java2d.pipe.DrawImage.scaleImage(DrawImage.java:126)
	at java.desktop/sun.java2d.pipe.DrawImage.scaleImage(DrawImage.java:1077)
	at java.desktop/sun.java2d.SunGraphics2D.drawImage(SunGraphics2D.java:3383)
	at java.desktop/sun.awt.image.ImageRepresentation.drawToBufImage(ImageRepresentation.java:845)
	at java.desktop/sun.java2d.pipe.DrawImage.scaleImage(DrawImage.java:1084)
	at java.desktop/sun.java2d.pipe.ValidatePipe.scaleImage(ValidatePipe.java:207)
	at java.desktop/sun.java2d.SunGraphics2D.drawImage(SunGraphics2D.java:3383)
	at java.desktop/sun.java2d.SunGraphics2D.drawImage(SunGraphics2D.java:3321)
	at io.github.dsheirer.spectrum.WaterfallPanel.paintComponent(WaterfallPanel.java:297)
	at java.desktop/javax.swing.JComponent.paint(JComponent.java:1128)
	at java.desktop/javax.swing.JComponent.paintToOffscreen(JComponent.java:5318)
	at java.desktop/javax.swing.RepaintManager$PaintManager.paintDoubleBufferedImpl(RepaintManager.java:1656)
	at java.desktop/javax.swing.RepaintManager$PaintManager.paintDoubleBuffered(RepaintManager.java:1631)
	at java.desktop/javax.swing.RepaintManager$PaintManager.paint(RepaintManager.java:1569)
	at java.desktop/javax.swing.RepaintManager.paint(RepaintManager.java:1336)
	at java.desktop/javax.swing.JComponent._paintImmediately(JComponent.java:5266)
	at java.desktop/javax.swing.JComponent.paintImmediately(JComponent.java:5076)
[... 93 lines cut ...]
Exception: java.lang.OutOfMemoryError thrown from the UncaughtExceptionHandler in thread "sdrtrunk polyphase channel thread 1"

Exception: java.lang.OutOfMemoryError thrown from the UncaughtExceptionHandler in thread "sdrtrunk polyphase channel thread 1"

Exception: java.lang.OutOfMemoryError thrown from the UncaughtExceptionHandler in thread "sdrtrunk polyphase channel thread 1"

Exception: java.lang.OutOfMemoryError thrown from the UncaughtExceptionHandler in thread "sdrtrunk polyphase ifft processor thread 1"

Exception: java.lang.OutOfMemoryError thrown from the UncaughtExceptionHandler in thread "sdrtrunk polyphase channel thread 1"

Exception: java.lang.OutOfMemoryError thrown from the UncaughtExceptionHandler in thread "sdrtrunk scheduled thread 2"

Exception: java.lang.OutOfMemoryError thrown from the UncaughtExceptionHandler in thread "sdrtrunk polyphase channel thread 1"

Exception: java.lang.OutOfMemoryError thrown from the UncaughtExceptionHandler in thread "sdrtrunk polyphase channel thread 1"

Exception: java.lang.OutOfMemoryError thrown from the UncaughtExceptionHandler in thread "sdrtrunk scheduled thread 3"

Exception: java.lang.OutOfMemoryError thrown from the UncaughtExceptionHandler in thread "sdrtrunk scheduled thread 4"

Exception: java.lang.OutOfMemoryError thrown from the UncaughtExceptionHandler in thread "sdrtrunk scheduled thread 1"

Exception: java.lang.OutOfMemoryError thrown from the UncaughtExceptionHandler in thread "Reference Handler"

```

DSheirer, 2023-05-15:
There's not much I can do to fix this issue.

Try running the application on a more powerful machine, or one with more RAM.  Make sure that you're not imposing a RAM constraint (-Xmx **g)

closed 2023-05-15 by DSheirer

ah1102, 2023-05-15:
Is it normal for an application not to take up a fixed amount of RAM +/-, but keep growing until an error occurs?

It seems that the program saves something in RAM and does not clear it.

I have 16 GB of RAM and I have seen that with a receiver bandwidth of 20 MHz, the RAM consumption is 3 GB and grows faster. And with a bandwidth of 5 MHz, the application takes up 500 MB of RAM, but the memory consumption grows much more slowly. This happens not only in AM mode, but also in P25P1.

DSheirer, 2023-05-15:
Java operates differently than other software because Java uses a Garbage Collector (GC) to clean up (clear) any unused memory.  The Java GC runs independently from the application.  Sometimes the GC will allow the garbage to build up and consume a lot of 'available' system memory and then the GC will run and cleanup that memory.

There's a CPU cost when the GC runs and Java attempts to find a balance between running the GC and letting garbage memory pile up and it tries not to impact program execution.  This can cause the total memory usage to rise and fall over time.

If sdrtrunk is running out of memory, it means that either a)you're not allowing the OS to give it enough resources or b) your CPU is not capable of running everything that you are asking it to run.

There could certainly be a memory leak that allows the application to continuously use more and more memory, but I've been heavily profiling sdrtrunk for the last 4 months and I believe that I've found any remaining issues.

matt-keys, 2024-12-29:
My apologies for reviving an old thread. I ran into this as well and wanted to share something I noticed. I'm running in a VM which I initially provisioned with 4 vcpus and 8GB ram. OOO occurred, so I bumped to 12 vcpus and 16GB of RAM and retested. 

In normal operation, ie launching the app and auto monitoring trunks, I noticed the 'allocated memory' and 'used memory' stays pretty consistent 1/4 to 1/3 of the ram in a cycle of consumption and release. However when I navigate to the playlist editor area and it starts to pull data from radio reference (while still monitoring trunks), the allocated memory shoots up and continues to climb. I suspect there may be a memory leak or lack of GC in that particular area.

DSheirer, 2025-01-03:
@matt-keys can you post it to a google drive or similar and send me a link to download?

matt-keys, 2025-01-03:
> @matt-keys can you post it to a google drive or similar and send me a link to download?

https://drive.google.com/file/d/1ir3Bdllg-r6TsEgE7n2P7uVfoEhk6uGx/view?usp=sharing

DSheirer, 2025-01-03:
The VisualVM screenshot shows that the memory leak is in the tuner's Polyphase Channelizer.  The incoming samples (from the tuner) queue has filled up and the channelizer dispatcher thread is not able to process the samples faster than they are arriving.  This could happen if channelizer has a blocked thread/consumer downstream, or if the CPU is simply not able to keep up with the processing.

From your application log it looks like you're running a 4-core CPU and you've set the HackRF sample rate to the highest (20 MHz).  

I need to understand if this is an issue where your CPU is underpowered for the workload you're attempting, or if there is a downstream blocked thread issue that's causing back pressure and causing the queue to fill up.  In order to diagnose, I need a VisualVM heap dump from when the application runs out of memory and I also need a diagnostic report run just before the application runs out of memory.  From your application log, it looks like you attempted to run the diagnostic report after the application ran out of memory and it looks like it failed.  You may be able to detect when/once it has run out of memory from the memory consumption chart in Visual VM ... when you see it climb to the top like in the screen shot you posted above.

I looked through the diagnostic report that you attached and there doesn't appear to be any blocked threads in the downstream channel consumers.

matt-keys, 2025-01-03:
> From your application log it looks like you're running a 4-core CPU and you've set the HackRF sample rate to the highest (20 MHz).
> 

I assumed the same initially, however I can reproduce this with the smallest sample size of 5MHz. It does take longer to achieve at 5MHz sample. 

CPU doesn't seem to be the either issue as I allocated 12 vcpus in the original VM use case. It's not struggling until the oom occurs, then after cpu spikes. 

To confirm what you'd like for next steps, I should leave the 8GB max / 1GB min heap settings in place, run a 5 mhz sample and collect diagnostic reports just before oom? Is there anything additional in visualvm you'd like?

Are you able to reproduce this in your test lab?

DSheirer, 2025-01-03:
Looking at the thread dump in the diagnostic report a bit further ... the primary tuner sample stream processing thread appears to be running and processing and not in a blocked state.  Reading your posted notes again, I see that you were running 4x virtual CPUs during this report period.  Is it possible that your VM is not getting enough dedicated/consistent CPU time?  This tuner sample stream processing thread is a critical process that needs consistent CPU time otherwise the incoming sample queue will build up.

```
Thread[#66,sdrtrunk polyphase buffer processor thread 1,5,main] RUNNABLE
	at app//io.github.dsheirer.dsp.filter.channelizer.ComplexPolyphaseChannelizerM2.process(ComplexPolyphaseChannelizerM2.java:343)
	at app//io.github.dsheirer.dsp.filter.channelizer.ComplexPolyphaseChannelizerM2.receive(ComplexPolyphaseChannelizerM2.java:222)
	at app//io.github.dsheirer.dsp.filter.channelizer.PolyphaseChannelManager$NativeBufferReceiver.receive(PolyphaseChannelManager.java:521)
	at app//io.github.dsheirer.dsp.filter.channelizer.PolyphaseChannelManager$NativeBufferReceiver.receive(PolyphaseChannelManager.java:472)
	at app//io.github.dsheirer.util.Dispatcher.process(Dispatcher.java:184)
	at app//io.github.dsheirer.util.Dispatcher$Processor.run(Dispatcher.java:207)
	at java.base@23.0.1/java.util.concurrent.Executors$RunnableAdapter.call(Unknown Source)
	at java.base@23.0.1/java.util.concurrent.FutureTask.runAndReset(Unknown Source)
	at java.base@23.0.1/java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(Unknown Source)
	at java.base@23.0.1/java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.base@23.0.1/java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.base@23.0.1/java.lang.Thread.runWith(Unknown Source)
	at java.base@23.0.1/java.lang.Thread.run(Unknown Source)
```

matt-keys, 2025-01-03:
>  Is it possible that your VM is not getting enough dedicated/consistent CPU time? This tuner sample stream processing thread is a critical process that needs consistent CPU time otherwise the incoming sample queue will build up.

I can run the VM test again if you'd prefer and collect heap there. The heap dump I've provided is from the physical machine. [The hackrf documentation doesn't recommend running through a VM](https://hackrf.readthedocs.io/en/latest/virtual_machines.html), so I moved to the physical machine to rule out a virtualization layer issue. The physical machine is indeed weaker in cpu power, but has 16gb ram.

DSheirer, 2025-01-03:
> I assumed the same initially, however I can reproduce this with the smallest sample size of 5MHz. It does take longer to achieve at 5MHz sample. CPU doesn't seem to be the either issue as I [...]

As the Java VM runs out of memory, it dedicates more/all CPU cycles to managing the remaning (small) amount of free memory attempting to keep from crashing. 

> To confirm what you'd like for next steps, I should leave the 8GB max / 1GB min heap settings in place, run a 5 mhz sample and collect diagnostic reports just before oom? Is there anything additional in visualvm you'd like?
> 

You can run at the full 20 MHz sample rate ... it'll just get to OOM faster.  8GB max should be good.

> Are you able to reproduce this in your test lab?

I can't reproduce this.  I haven't tested with the HackRF yet, but I do routinely test with Airspy at 10 MHz sample rate and the code where you're having the memory leak is the same as what's used downstream from the Airspy.  I'll try to setup a test with HackRF to see if I can reproduce.  I don't have a VM test infrastructure.

DSheirer, 2025-01-03:
Testing now on bare iron (AMD 5800X w/32GB)
[image]

matt-keys, 2025-01-03:
@DSheirer Would you mind providing an email address that can I use to restrict the sharing link data to just you? The retest log data and screen capture video I'm about to post contain my radioreference username.

DSheirer, 2025-01-03:
@matt-keys What CPU is this running on?

DSheirer, 2025-01-03:
Unfortunately, that CPU is not up to the task of running sdrtrunk with the HackRF running 20 MHz sample rate.  It's just not keeping up with the processing workload and it's causing the buffers to fill up with unprocessed data.

In looking at the logs and heap dump, there's no code issues other than sdrtrunk could probably start dumping tuner sample data once it gets that far behind.

matt-keys, 2025-01-03:
> Unfortunately, that CPU is not up to the task of running sdrtrunk with the HackRF running 20 MHz sample rate. It's just not keeping up with the processing workload and it's causing the buffers to [...]

Is 5MHz sample a more appropriate size? It takes about an hour to reach oom at 5MHz rather than about 15 minutes at 20MHz. I'll send you the 5MHz test data link in a bit.

matt-keys, 2025-01-11:
Just to update I'm still working on this. Given your mention of cpu scheduling, and the hardware is an amd APU, I did some digging and discovered the amdgpu module was using a frequency scaling model of 'balanced' / 'auto' by default, which I have adjusted to high/high (https://docs.kernel.org/gpu/amdgpu/thermal.html#power-dpm-force-performance-level) . I am testing with that setting now.

Is it expected for polyphase channelizer to run with these high settings:

`20250111 073823.984 [sdrtrunk cached thread 1] INFO  i.g.d.d.f.c.ComplexPolyphaseChannelizerM2 - Sample Rate [5000000.0] providing [200] channels at [25000.0] Hz each  [153MB/1GB 15%]
`

... though the tuner is set to 5mhz and I'm only monitoring 3 channels with 12.5 bandwidth set?