DSheirer/sdrtrunk #1550: DMR T3 LSN Automatic Calculations

Nokoa, 2023-05-14:
**Is your feature request related to a problem? Please describe.**
Manually entering all the frequencies and their LSN for large DMR Tier 3 networks, is a timely task, especially when they are not all known in advance. User can use a Uniden scanner LCN Finder to find the LCN for entered frequencies. But this still requires the user to know all the frequencies associated for the site.

**Describe the solution you'd like**
Apparently it's possible to calculate all LSN frequencies for DMR Tier 3 networks by knowing only one frequency LSN. I think it could be possible to implement into SDRTrunk where either user defines a single LSN, or the LSN for the control channel is found automatically. I am not certain if the control channel identifies its LSN in the data. If not, it could be possible to wait for a activity on the second slot of the control channel, and identify the channel that way. Then calculating the rest of the LSNs can occur by adding 12.5 khz for each LSN higher.

**Describe alternatives you've considered**
Manually calculating the LSNs myself with the same method and adding them to the list. However I don't know what frequencies will be used. 

**Additional context**
This method as I understand it, only works for DMR Trunked Tier 3. Not Capacity Plus, or any others.

DSheirer, 2023-05-15:
This capability to calculate DMR Tier 3 channel frequencies already exists in the sdrtrunk application.

When adding a DMR Tier 3 channel configuration you only have to enter the control frequency.  The LCN channel map is only used for systems like cap+ or con+

Nokoa, 2023-05-15:
I cannot seem to get this working with both Motorola DMR T3 and a Tait DMR T3. With nothing filled in the LSN table.

[image]

Additionally from the console you requested the following be sent to you:
```
WARN  i.g.d.m.d.d.m.d.p.UDTShortMessageService - Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:3180000018070001AC2F0000007FFFFFFFFFFFFFFFFF09C9 As String:1∩┐╜↑☺∩┐╜/∩┐╜∩┐╜∩┐╜∩┐╜∩┐╜∩┐╜∩┐╜∩┐╜    ∩┐╜  
2023-05-15 14:41:22.462 WARN  i.g.d.m.d.d.m.d.p.UDTShortMessageService - Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:31800000AB07000123360000007FFFFFFFFFFFFFFFFF7D1A As String:1∩┐╜∩┐╜☺#6∩┐╜∩┐╜∩┐╜∩┐╜∩┐╜∩┐╜∩┐╜∩┐╜}→  
```