DSheirer/sdrtrunk #1552: Cap+ - Rest beacon does not stop channel rotation

wedgef5, 2023-05-15:
master-branch

The local Cap+ system sends a Rest beacon (CSBKO=62) every 2 seconds, which I believe is pretty standard Cap+ configuration. Upon detecting the beacon, the software should stop channel rotation, as the current Rest frequency/timeslot is where the next voice call will occur. However, channel rotation continues after the delay time.

One contributor to this issue is that the maximum rotation delay is 2 seconds. If I am interpreting the code correctly, channel rotation will resume if no activity is detected for "delay" seconds. Even at the maximum setting, rotation could resume before the next beacon message is heard. That said, I modified ChannelRotationMonitor and increased mRotationDelay to 3000 ms to force it to sit on the frequency long enough to definitely hear the beacon, and still rotation continued.

IMO, on a Cap+ system, channel rotation should not resume unless it is clear that the Rest beacon has been lost. i.e. only continue if nothing is heard for something quite a bit longer than the beacon interval. What is "quite a bit longer"? I'm not sure. Perhaps 4000 ms?

Steps to reproduce the behavior:
1. Configure a multi-frequency DMR Cap+ system with a single channel.
2. Set channel rotation delay to the maximum of 2.0s
3. Start monitoring
4. Watch the Messages pane for a Site Status message, followed by channel rotation resuming.

I would expect the software to stay on the Rest frequency once it is found and follow to a new Rest frequency as directed by the system, unless the Rest frequency is lost for some reason. This is one (of many) aspects of Cap+ that I'm not clear on. When a voice call starts on the current Rest frequency/timeslot, where is the new Rest frequency/timeslot information found?

lwvmobile, 2023-05-16:
Just FWIW, a few of my own personal observations:

1. While most Cap+ systems always have one RF channel beaconing at all times during periods of inactivity, I have also observed at least one Cap+ system cease to beacon altogether for minutes at a time. This could either be that I am missing a frequency for it, or that it just 'stops' for a period of time. I never observe the rest channel change when this occurs to a value that would indicate that I am missing a frequency. Which leads me to my second observation:

2. Cap+ systems may rotate to a new rest channel during beaconing mode without signalling it in the CSBK or SLC. While its often noted that the rest channel will change during call activity, when there is no activity, and its just beaconing, I've observed that there is no change of rest channel in the CSBK or the SLC, and it just rotates to the next value. Usually, my observation is that it always rotates to the next sequential value, or if its at its max value, it just rotates back to 1, but could be on a per system basis.

3.

> When a voice call starts on the current Rest frequency/timeslot, where is the new Rest frequency/timeslot information found?

If there is no activity in the other timeslot for the RF channel you are observing, you will usually still have the CSBK 62 signalling there to indicate the new rest channel and any current system activity on other LSNs. If both timeslots are busy with voice and/or data, then the slow link control can be consulted for the rest channel, but I've observed that occassionally, on busy systems that information can lag due to slow link control needing four timeslots to complete itself and by the time its complete, a new rest channel may already be assigned.

lwvmobile, 2023-05-16:
PS,

I compiled the latest main branch, and tested it, and can confirm, it does not stop on the current rest channel for long enough to get subsequent beacons. I have tested this on a marginal signal strength system, and a good signal system, and the same result. The only time it stops is when there is prolonged activity (longer site status broadcasts, voice or data activity).

I would usually conjecture a guess at this point, so its possible that the bursts are so short, that SDRTrunk is having a hard time getting good sync and demodulation of those bits, and without a positive decode of at least one message, perhaps it doesn't know to remain stationary long enough to continue decoding. 

One thing I may try to do is just set up a DMR system in the playlist with a single frequency (covering just LSN 1 and 2) on the strong signal so it'll remain stationary and see if it decodes the Site Status CSBK on the short beacons when they are active there.

wedgef5, 2023-05-18:
I have not ever seen our local system stop beaconing, but then again, it is a sample size of one, so it is certainly possible. With regard to these observations, and others about the handling of the rest channel, I wonder if there might be some value in discussing them on [this RR thread](https://forums.radioreference.com/threads/understanding-capacity-plus-trunking-some-more.452566/). The folks on that thread have taken some pretty deep dives into Cap+ and may be able to shed some light.

I was just doing the single-frequency monitoring that @lwvmobile mentioned, and observed exactly what you are referring to. I have a pretty solid signal from the local site, and I see a number of beacons that SDRtrunk fails to decode. I see quite a few CRC-ERRORs and other bursts where there is nothing at all in the decoded messages. At one point, I went 20s without a good decode on the beacon. I'm thinking this should be a new issue. Opinions?

wedgef5, 2024-08-05:
I've looked into this a little bit more. I see that when rest beacons are successfully decoded, the channel goes to an ACTIVE state, but then immediately goes back to IDLE. I'm wondering if perhaps it should stay ACTIVE when a Cap+ rest beacon is detected for some longer period of time, even when no carrier is being detected. For example, maybe there is some new Cap+ REST Timeout threshold of like 5 seconds. If that is reached, then two successive beacons have been missed, so resume channel rotation.

wedgef5, 2025-03-25:
Circling back to this after the DMR decoder redesign (#2115). I tested again hardcoding the channel rotation delay to 4000 ms in ChannelRotationMonitor, and it is following the system pretty well now, including following a move to a new Rest channel. I'm thinking there are at least 3 ways to deal with this.:

1. Allow a higher maximum rotation delay, and simply instruct users with a Cap+ system to set the delay to 2-3x the burst interval.
2. Add a configuration option to the GUI to have the user explicitly set the Cap+ Site Status interval, then have the channel rotation delay be 2-3x that,
3. Measure the time between bursts in DMRDecoderState and send an EventBus message to ChannelRotationMonitor to adjust the delay accordingly.

I'm guessing this would apply to other Motorola systems (e.g. Cap Max).

DSheirer, 2026-11-08:
Recent DMR decoder redesign should resolve this.

closed 2026-11-08 by DSheirer