DSheirer/sdrtrunk #1558: Out of range tuning in Het mode

AC9BX, 2023-05-20:
**sdrtrunk Version**
0.6.0

**Describe the bug**
In heterodyne mode when attempting to tune a frequency substantially above (by about 3.59 MHz) the range of a tuner locked frequency it will drop anything locked and tune to said attempted frequency versus ignoring it or displaying a "no tuner available" message. Going below the locked frequency does not appear to do this. This is problematic because if monitoring a trunk system for example with frequencies too far apart for a single tuner it will willingly drop the control channel if it needs to reach a frequency substantially too high.

**To Reproduce**
1 Put program in Heterodyne mode
2a Use any trunk system too wide for a single tuner, wait for a call on a frequency too high for the tuner to reach while locked on the control channel, or try tuning a conventional frequency too high.
2b or tune to any frequency. (In this example an always on VHF channel) Then tune to another frequency too high for the tuner to reach. The range moves to include the new frequency, dropping anything too low. 
3 Stopping the 'too high' channel, the software does NOT move back to the original when in analog. It will eventually get back to normal when trunking.
4 Selecting a frequency too low, the range moves to tune the highest frequency.

**Expected behavior**
"Cannot tune, no tuner available" message
"fade" indication when trunking that the channel cannot be reached.

**Desktop (optional - complete the following information):**
Windows and Linux

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

[sdrtrunk_app.log](https://github.com/DSheirer/sdrtrunk/files/11522889/sdrtrunk_app.log)

jamren01, 2023-05-21:
I too am observing this; however, only on 0.6.0 alpha 4 and newer, observing 700/800MHz P25 traffic on 3 SDRplay RSP1a devices.

DSheirer, 2023-05-21:
@AC9BX Thanks for the detailed bug report.

PR #1561 (2023-05-21): #1558 Heterodyne Channelizer Allows Channels Exceeding Tuner Bandwidth
Closes #1558 

Resolves issue where heterodyne channelizer allows tuning a set of channels that exceed the tuner's bandwidth.
closed 2023-05-21 by DSheirer