DSheirer/sdrtrunk #1564: After use of MPT1327 channel, tuner remains locked despite deselecting in Playlist Editor

geoffb1723, 2023-05-28:
Was fooling around with some MPT1327 channels in the 454 MHz band. Despite de-selecting them, the tuner shows as LOCKED in the  "tuners' window, with no check mark in the Playlist editor appearing for any system.  (3 RTL-USB tuners) It is locked, as I start getting 'Tuner Unavailable' messages for some other P25 systems when I try to select them, as only 2 of 3 tuners are available. It is locked and stuck on the MPT freq with no check mark in Playlist Editor.

Minor, a restart clears it up, not a big deal, just FYI. It's reproduceable, just select any MPT channel, now deselect, a Tuner remains locked.   Even better, since no one really cares about MPT, dont choose those freqs !

Interesting side note, Tuner #3 always comes up on the same NBFM freq, with nothing in Playlist Editor. Selecting  2 x P25 systems to monitor unlocks it and inserts the P25 appropriate freq and locks it, as required. My guess is that on the first P25 system there is not a lot of traffic, so SDRT has no need to utilize the 3rd tuner, adding in the second P25 system causes it to act correctly and utilize all tuners as required. Am I off base on this ? Probably !

Ver 6.0a5 on osx  13.3.1,  aarch64

thanks !

geoff

DSheirer, 2023-06-05:
I'm unable to recreate the issue.  I can create a new MPT1327 channel configuration, start it, stop it, and the tuner shows zero channels (unlocked).

Can you please look in the application log and see if there is an error that's being logged when this happens for you?  If so, can you please attach the log to this issue?

geoffb1723, 2023-06-05:
thanks, appreciate your looking at it, will check logs.

g

geoffb1723, 2023-06-05:
Interesting, it is not logging the MPT channel being selected/deselected, it is only logging the P25s, I selected and tried to deselect the MPT 3-4 times. Of course, the 
behavior of the program locking the tuner permanently seems to have stopped. So maybe you close this as not reproducible, if it shows up again I will re-open it ?

Interesting FYI -  it seems not to log any NBFM  or MPT or DMR selections, it seems only to log the P25 systems being selected/deselected.

thanks

g

20230605 034832.747 [main] INFO  i.g.d.log.ApplicationLog - SDRTrunk Version  : 0.6.0-alpha5  [18MB/112MB 16%]
20230605 034832.749 [main] INFO  i.g.d.log.ApplicationLog - Gradle Version    : Gradle 7.6  [18MB/112MB 16%]
20230605 034832.749 [main] INFO  i.g.d.log.ApplicationLog - Build Timestamp   : 2023-04-12T03:25:47.587-0400  [18MB/112MB 16%]
20230605 034832.749 [main] INFO  i.g.d.log.ApplicationLog - Build-JDK         : 19.0.1 (BellSoft 19.0.1+11  [18MB/112MB 16%]
20230605 034832.749 [main] INFO  i.g.d.log.ApplicationLog - Build OS          : Linux (amd64 5.19.0-38-generic  [18MB/112MB 16%]
20230605 034832.749 [main] INFO  i.g.d.log.ApplicationLog -   [19MB/112MB 17%]
20230605 034832.749 [main] INFO  i.g.d.log.ApplicationLog - *******************************************************************  [19MB/112MB 17%]
20230605 034832.750 [main] INFO  i.g.d.log.ApplicationLog - **** sdrtrunk: a trunked radio and digital decoding application ***  [19MB/112MB 17%]
20230605 034832.750 [main] INFO  i.g.d.log.ApplicationLog - ****  website: https://github.com/dsheirer/sdrtrunk             ***  [19MB/112MB 17%]
20230605 034832.750 [main] INFO  i.g.d.log.ApplicationLog - *******************************************************************  [19MB/112MB 17%]
20230605 034832.750 [main] INFO  i.g.d.log.ApplicationLog - Memory Logging Format: [Used/Allocated PercentUsed%]  [19MB/112MB 17%]
20230605 034832.750 [main] INFO  i.g.d.log.ApplicationLog - Host OS Name:          Mac OS X  [19MB/112MB 17%]
20230605 034832.750 [main] INFO  i.g.d.log.ApplicationLog - Host OS Arch:          aarch64  [19MB/112MB 17%]
20230605 034832.750 [main] INFO  i.g.d.log.ApplicationLog - Host OS Version:       13.3.1  [19MB/112MB 17%]
20230605 034832.750 [main] INFO  i.g.d.log.ApplicationLog - Host CPU Cores:        20  [19MB/112MB 17%]
20230605 034832.750 [main] INFO  i.g.d.log.ApplicationLog - Host Max Java Memory:  29 GB  [19MB/112MB 17%]
20230605 034832.750 [main] INFO  i.g.d.log.ApplicationLog - Storage Directories:  [19MB/112MB 17%]
20230605 034832.751 [main] INFO  i.g.d.log.ApplicationLog -  Application Root: /Users/geoffb/SDRTrunk  [19MB/112MB 17%]
20230605 034832.751 [main] INFO  i.g.d.log.ApplicationLog -  Application Log:  /Users/geoffb/SDRTrunk/logs  [19MB/112MB 17%]
20230605 034832.751 [main] INFO  i.g.d.log.ApplicationLog -  Event Log:        /Users/geoffb/SDRTrunk/event_logs  [19MB/112MB 17%]
20230605 034832.751 [main] INFO  i.g.d.log.ApplicationLog -  Playlist:         /Users/geoffb/SDRTrunk/playlist  [19MB/112MB 17%]
20230605 034832.751 [main] INFO  i.g.d.log.ApplicationLog -  Recordings:       /Users/geoffb/SDRTrunk/recordings  [19MB/112MB 17%]
20230605 034832.807 [main] INFO  i.g.dsheirer.util.ThreadPool - Application thread pool created SCHEDULED and CACHED executors threads  [22MB/112MB 20%]
20230605 034832.807 [main] INFO  i.github.dsheirer.gui.SDRTrunk - Home path: /Users/geoffb/SDRTrunk  [22MB/112MB 20%]
20230605 034832.808 [main] INFO  i.g.d.p.SystemProperties - SystemProperties - loaded [/Users/geoffb/SDRTrunk/SDRTrunk.properties]  [24MB/112MB 21%]
20230605 034832.808 [main] INFO  i.g.d.p.SystemProperties - SystemProperties - application properties loaded [/Users/geoffb/SDRTrunk/SDRTrunk.properties]  [24MB/112MB 21%]
20230605 034832.854 [main] INFO  i.g.d.s.t.manager.TunerManager - Discovering tuners ...  [27MB/112MB 24%]
20230605 034833.114 [main] INFO  i.g.d.s.t.manager.TunerManager - LibUsb API Version: 1.0.264  [27MB/112MB 24%]
20230605 034833.115 [main] INFO  i.g.d.s.t.manager.TunerManager - LibUsb Version: 1.0.24.11584  [27MB/112MB 24%]
20230605 034833.135 [main] INFO  i.g.d.s.t.manager.TunerManager - LibUsb - discovered [38] potential usb devices  [27MB/112MB 24%]
20230605 034833.140 [main] INFO  i.g.d.s.t.manager.TunerManager - Discovered tuner at USB Bus [13] Port [1.2.1.3] Tuner Class [RTL-2832]  [27MB/112MB 24%]
20230605 034833.141 [main] INFO  i.g.d.s.t.manager.TunerManager - Discovered tuner at USB Bus [13] Port [1.2.1.4] Tuner Class [RTL-2832]  [27MB/112MB 24%]
20230605 034833.142 [main] INFO  i.g.d.s.t.manager.TunerManager - Discovered tuner at USB Bus [12] Port [3] Tuner Class [RTL-2832]  [27MB/112MB 24%]
20230605 034833.143 [main] INFO  i.g.d.s.t.manager.TunerManager - Tuner: USB Tuner - RTL-2832 USB Bus:13 Port:1.2.1.3 - Added / Starting ...  [27MB/112MB 24%]
20230605 034834.318 [main] INFO  i.g.d.d.f.c.ComplexPolyphaseChannelizerM2 - Sample Rate [2400000.0] providing [96] channels at [25000.0] Hz each  [33MB/112MB 29%]
20230605 034834.506 [main] INFO  i.g.d.d.f.c.ComplexPolyphaseChannelizerM2 - Sample Rate [2048000.0] providing [80] channels at [25600.0] Hz each  [33MB/112MB 29%]
20230605 034834.548 [main] INFO  i.g.d.s.t.manager.TunerManager - Tuner: USB Tuner - RTL-2832 USB Bus:13 Port:1.2.1.4 - Added / Starting ...  [33MB/112MB 30%]
20230605 034835.687 [main] INFO  i.g.d.d.f.c.ComplexPolyphaseChannelizerM2 - Sample Rate [2400000.0] providing [96] channels at [25000.0] Hz each  [33MB/112MB 30%]
[... cut ...]
20230605 035107.102 [sdrtrunk cached thread 1] INFO  i.g.d.m.d.p.P25TrafficChannelManager - Stopping traffic channel: San Mateo Cty Trunk_County Center_TRAFFIC_42  [318MB/544MB 58%]
20230605 035533.021 [sdrtrunk polyphase channel thread 1] INFO  i.g.d.a.c.mbe.AmbeAudioModule - AMBE CODEC successfully loaded - P25-2/DMR/NXDN audio will be available  [264MB/544MB 48%]
20230605 035535.448 [sdrtrunk cached thread 6] INFO  i.g.d.m.d.p.P25TrafficChannelManager - Stopping traffic channel: Silicon Valley Regional Communications System (SVRCS)_Santa Clara_TRAFFIC_84  [376MB/592MB 63%]

DSheirer, 2023-06-05:
> Interesting FYI - it seems not to log any NBFM or MPT or DMR selections, it seems only to log the P25 systems being selected/deselected. thanks g

The P25 traffic channel manager handles all of the allocated traffic channels and logs the traffic channels that it's stopping when the control channel is stopped/disabled.

When the user starts or stops a standard decode channel, those do not get logged.

I'll close this issue and, as you said, you can reopen later if needed.  Thanks for the followup!

closed 2023-06-05 by DSheirer

geoffb1723, 2023-06-05:
Copy, got it.

thanks

g

geoffb1723, 2023-06-05:
> I'll close this issue and, as you said, you can reopen later if needed. Thanks for the followup!

Appreciate your following up ! (and the explanation).

rgds

geoff