DSheirer/sdrtrunk #1566: Recordings file name delimiter interference

tylerwatt12, 2023-06-04:
Hi, I noticed some potential issues with recordings when the system, site or channel have spaces in them.

Take for example the following scenario:
System: Ohio MARCS-IP: Multi-Agency Radio Communications
Site: Cuyahoga

Recordings from this system get saved as this, where any spaces in the system name turn into underscores
`20230604_115401Ohio_MARCS-IP__Multi-Agency_Radio_Communications_Cuyahoga_TRAFFIC__TO_13570_FROM_1891771.mp3
`
My code is unable to discern the delimitation of these fields when either of them have spaces in the field value. 
This is due to both the field delimiter, and str.replace(" ","_") function in sdrtrunk using the underscore character.

I would like to submit a feature request or bug fix to fix this.

A potential feature request fix: Add the ability for a user to select their own delimiter character, and leave _ as default for backwards compatibility.

A potential bug fix: Replace field delimiters with a caret or similar unique character so a recording would appear as

`20230604_115401Ohio_MARCS-IP__Multi-Agency_Radio_Communications^Cuyahoga^TRAFFIC^^TO^13570^FROM^1891771.mp3`

DSheirer, 2023-06-24:
@tylerwatt12 for simplicity I use the underscore as both a delimiter and an invalid character replacement.  This ensures that the sdrtrunk filename validation code won't generate an error due to the user entering a non-valid character for file naming purposes.  Allowing the user to specify either the delimiter or the replacement character opens the potential for the user to introduce an invalid character and cause the code to fail.

There's a couple different third party applications and custom end-user processing scripts that are depending on those file names to remain constant, so I'm hesitant to change the format ... considering that the issue can be resolved by the end user not using invalid symbols in the channel configuration.

Thoughts?

tylerwatt12, 2023-06-29:
Hi Denny,
I was able to find a workaround by pulling this data from the ID3 tag

closed 2023-06-29 by tylerwatt12