DSheirer/sdrtrunk #1571: Automatically Restart SDRTrunk on Pipe Errors

RotatingWires, 2023-06-06:
**Is your feature request related to a problem? Please describe.**
Yes, the problem is related to pipe errors.

**Describe the solution you'd like**
When a pipe error occurs, I would want SDRTrunk to automatically restart. I know that SDRT tries to restart the tuner 5 times after a pipe error, but that never works for me. Even if success of the tuner restart can't be detected, I'd still prefer SDRT to restart anyway, as again, the restart for tuners never fixes pipe errors for me.

**Describe alternatives you've considered**
I've tried switching out the USB ports that the tuners are plugged into, but that didn't fix the problem either. I'm not using any hubs or USB extensions.

**Additional context**
This can be an optional setting in the user preferences.

louisik1, 2023-06-06:
Hi there, I'm just a lowly user and a new one at that, but I found myself and others were running into pipe errors when using USB 3 ports, I presume hubs or not. Maybe something you haven't checked yet. PS are you discussing this on a thread anywhere?

RotatingWires, 2023-06-06:
I can double check which type of ports I'm using for the tuners, but I'm pretty sure most of the ports on the computer I'm using to run SDRT are USB 3. I will try to see if I can fit the tuners in ports that aren't USB 3 and see if it works. If not, I'm not sure what else I could do. 

I'm not discussing this on a thread anywhere. Thanks for the tip though!

RotatingWires, 2023-06-10:
@louisik1 and for anyone interested, I found this: 

https://github.com/LDZPLN1/SDRTrunk-Monitor

It does what I want and more. Check it out!

closed 2023-06-10 by RotatingWires