DSheirer/sdrtrunk #1572: YSF Support?

troyBORG, 2023-06-09:
The local Amateur Radio Club Repeater near me uses YSF.
I was wondering if you can add support for this, or if its already implemented why isn't it listed as YSF in the encoders

DSheirer, 2023-06-10:
@troyBORG I'd like to implement this at some point.  However, I don't have any sample recordings to test against.  Could you please make some baseband channel recordings from this repeater when user's are transmitting and send them to me?

You can turn on baseband recording in the playlist editor in the channel configuration.  You can use any of the protocol decoders (NBFM, DMR, etc) ... it won't decode the signal but it will record the channel's baseband I/Q sample stream.

troyBORG, 2023-06-11:
How long of a recording do you need?

I can try to get you two different ones.  One from a close strong signal, and one from a further away antenna.

troyBORG, 2023-06-12:
I don't know if this is "YSF".   But this 2m Repeater near me was broadcasting this (it starts and stops) I'm wondering if it was some kinda data being sent?

 
[20230612_001005_147315000_HAM_KD0WY_2m_Repeater_Clinton__IA_54_baseband.zip](https://github.com/DSheirer/sdrtrunk/files/11718049/20230612_001005_147315000_HAM_KD0WY_2m_Repeater_Clinton__IA_54_baseband.zip)

DSheirer, 2023-06-13:
Awesome.  I downloaded it.  Once I get a free second, I'll dig into it.

mattdrum, 2024-09-04:
Has this been worked on at all? I wound like to 2nd this request and am willing to contribute to make it happen.

DSheirer, 2024-09-04:
It hasn't been worked. Feel free to try it.

troyBORG, 2024-09-05:
I know OpenWebRX has it working.   But needs p25.

To bad you two couldn't work together and get p25 working on openwebrx and
then get all the YSF, DMR, FAX, SSTV, FT8, WSPR.... from their stuff...