DSheirer/sdrtrunk #1576: alpha 6 text glitch

Linuxuser1234, 2023-06-21:
**sdrtrunk Version**
alpha 6 
**Describe the bug**
im having this issue where the text is not showing properly 
**To Reproduce**
Steps to reproduce the behavior:
open sdrtrunk 

[image]

Linuxuser1234, 2023-06-22:
@DSheirer do you know what is going on how do i fix this issue

DSheirer, 2023-06-22:
I have no idea what would cause that.  Are you using any non- English keyboard layout?

Linuxuser1234, 2023-06-22:
i do have ENG INT but most of the time i use ENG US

Linuxuser1234, 2023-06-22:
and i do have the latest jdk version installed and i have tried other sdrtrunk versions but i keep getting the same issue

DSheirer, 2023-06-24:
sdrtrunk ships with its own Java packaged with the application, so it doesn't depend on you having the correct or most-current java installed on your workstation.

It's interesting to note that the display characters in your screenshot are off by two, meaning that instead of displaying a capitol A, yours would be a C.  In your screenshot, the 'Radio Reference' tab is displayed as:

```
R a d i o  R e f e r e n c e  << Should display as this
S b e j p  S f g f s f o d f  << Off by 1
T c f k q  T g h g t g p e g  << Off by 2 - Actual Display from screenshot
```

You might try checking the Windows character set settings and try changing that value to different alphabets and then back to US English to see if that clears the issue.  This almost seems as if the ASCII alphabet characters have been remapped by a version of the ASCII alphabet that's shifted by two places.

Linuxuser1234, 2023-06-25:
@DSheirer where do you change the Windows character set is it in control panel->clock & region

closed 2026-11-08 by DSheirer