DSheirer/sdrtrunk #1577: ![image](https://github.com/DSheirer/sdrtrunk/assets/136218043/958c6016-80a6-4aa4-a42e-3728ec765f0f)

TRON0903, 2023-06-22:
[image]

[image]

so i think im making process after switching but im not hearing anything, the right side the frequency is changing but no sound.

_Originally posted by @TRON0903 in https://github.com/DSheirer/sdrtrunk/issues/907#issuecomment-1590184802_

TRON0903, 2023-06-22:
[image]

TRON0903, 2023-06-22:
[image]

CGUTH, 2023-06-23:
The tuner showing in the first image is tuned to the 100MHz band, looks like you have one tuner, what do the settings look like for it?

TRON0903, 2023-06-23:
im more new to this ,which specific settings are you referring too? sorry for not fulling knowing.

bradjtrammell, 2023-06-23:
What system are you trying to scan, and what model of SDR are you using?

If possible, include a screenshot of your "Tuners" tab, and of your Playlist Editor > Channels tab.

TRON0903, 2023-06-23:
[image]

TRON0903, 2023-06-23:
[image]

TRON0903, 2023-06-23:
here's some pics inside my settings.im still not hearing anything,thank you for helping and working with me

bradjtrammell, 2023-06-23:
It appears that it is locked on to the frequency now.

Do your "Now Playing" tab still show Idle? Or does it show channels being detected you just simply can't hear them?

DSheirer, 2023-06-23:
> [image]

From this screenshot, it doesn't appear that you have any p25 channels visible.  Those 2x signals that are showing appear to be broadcast FM radio stations bleading over.

I recommend two things:

1. Get the broadcast FM in line filter from rtl-sdr.com website and place it between your antenna cable and your tuner.
2. Get a good antenna and feedline,  if you don't already have one.

TRON0903, 2023-06-23:
@bradjtrammell i was trying to pick the main part of dickson where i am and charlotte.i imported from radio refrence,its not muted.yes im useing one of the stock antennas.its still idle.

[image]

[image]

bradjtrammell, 2023-06-23:
So I see multiple issues here. Are you running this on Windows or Linux? (If Linux what Distro)

a.) Your instance is coming back with "Unknown" for all the important network information. This can either mean you have an issue with your actual SDR hardware (defective), or you are too far away from the transmitting tower for the stock antenna to actually pick anything up of value.

b.) Your configuration specifies TACN as a Phase I system, but the TACN system is a **APCO P25 Phase II** system. Which means it's not setup correctly.

[image]

Personally, based on what I'm seeing. I have to wager a guess is that your trunk system configuration is not setup correctly.

I would recommend completely deleting your configuration in SDRTrunk, and starting fresh. Import everything from Radio Reference again, ensuring that you are leaving all imported values alone and as default, only selecting a single county site to test (whichever is closest you to geographically). When it asks you about what frequencies you want to select, make sure you check "ALL" instead of just "Control"

[image]

When completed, notice there is no option for C4FM or LSM as was in your earlier screenshots. Those options are only available in APCO P25 Phase I trunk. Also do not set a value for "Default Tuner", you only have a single one, so it makes no difference having this value specified or blank. Please refer to the screenshot below for what your configuration should look like (your frequencies may vary from the ones I randomly selected). 

[image]

Finally, as @DSheirer had alluded to. If you still continue to not have any luck, install a program like SDR++ (free) and tune to each of the control channels. If you hear what [sounds like this](https://www.youtube.com/watch?v=7Ir1gvkQXEE) {ignore the person talking as their commentary doesn't pertain to this subject}, then you are in fact hearing a P25 signal, and your configuration still needs to be adjusted so SDRTrunk can process it.

If you do not hear that sound, and instead hear talking, voices, music, or any other form of communication that is not that sound. Return the radio and get a new one. It's not tuned properly.

If you are still not able to get it working. Try using this as a starting point, and ONLY change the frequencies to those that are listed on Radio Reference. I wouldn't recommend changing anything else as far as System Configuration, then adjust the Aliases to those you wish to hear. 

_Keep in mind that most if not all Law Enforcement on TACN is said to be encrypted, which means you'll see it be picked up, but it won't be audible._ 

[TRON0903.zip](https://github.com/DSheirer/sdrtrunk/files/11850451/TRON0903.zip)

TRON0903, 2023-06-23:
i will try this and see what happens

TRON0903, 2023-06-23:
what is the easiest way to delete all this and start fresh? thank you

bradjtrammell, 2023-06-23:
> 

The easiest way would be to create a new XML file in your playlists.

TRON0903, 2023-06-23:
is that the zip you left up  there...how would i do that.i watched a video to set it up....thank you also for the help.im new and learning all  the time.

DSheirer, 2023-06-24:
@TRON0903 This appears to be a user setup issue and not a bug with the software, so I'm closing this issue.  The GitHub issues feature is for reporting software issues, not user support issues.

Continuing to use the stock antenna that comes with the RTL tuners is going to cause a significant waste of your time for decoding anything other than broadcast FM radio.  Recommend getting a better antenna before you expend any more effort toward decoding trunked radio channels.

Once you have a better antenna installed, you can ask any followup questions in one of the user support forums that are listed here:

https://github.com/DSheirer/sdrtrunk/wiki/User-Manual
(right-hand column and click the 'Help / Support' link)

cheers,
Denny

closed 2023-06-24 by DSheirer