DSheirer/sdrtrunk #1597: latest windows update and inability to monitor

Clats97, 2023-07-09:
**sdrtrunk Version**
Master branch most recent update

**Describe the bug**
No reception and no latching to control channel after the latest windows update

**To Reproduce**
Steps to reproduce the behavior:
1. Update Windows computer
2. try to listen to to a local system

**Expected behavior**
I expected it to work as per usual

**Desktop (optional - complete the following information):**
 - OS: Windows 11
 - CPU Cores: 8
 - RAM: 20GB

DSheirer, 2023-07-09:
I need more details.  Are you seeing signals in the spectral display?  Can you post a screenshot?  Have you tried reinstalling the zadig driver?

Clats97, 2023-07-09:
All good! Turns out it was a simple tuner error. I restarted it and boom it's working. Sorry for bothering you mate.

Thanks,
Josh Clatney

DSheirer, 2023-07-22:
Closing as resolved.

closed 2023-07-22 by DSheirer