DSheirer/sdrtrunk #1600: USB (Libusb) failing to detect anything

Intenna, 2023-07-13:
**sdrtrunk Version**
0.6.0-alpha 6 (issue occurs in alpha5 and 0.5.3 as well.

Issue is upon launching SDTrunk cannot select any receivers (using 2832 nooelec v5), the libusb shows -99 devices and reports error, i have SDR tuner working with SDR++ and others. (FYI this is an AM5 / X670E chipset with AMD and INTEL USB 3 host controllers)

application log below (paths edited for privacy reasons)
```
WARNING: Using incubator modules: jdk.incubator.vector
2023-07-13 18:54:30.826 INFO  i.g.d.icon.IconModel - loading icons file [blanked]\SDRTrunk\settings\icons.xml]  [15MB/500MB 3%]
2023-07-13 18:54:30.828 INFO  i.g.d.icon.IconModel - Icons file not found at [blanked]\SDRTrunk\settings\icons.xml]  [15MB/500MB 3%]
2023-07-13 18:54:31.066 INFO  i.g.d.log.ApplicationLog - Application Log File: [blanked]\SDRTrunk\logs\sdrtrunk_app.log  [23MB/500MB 4%]
2023-07-13 18:54:31.095 INFO  i.g.d.log.ApplicationLog - SDRTrunk Version  : 0.6.0-alpha6  [7MB/500MB 1%]
2023-07-13 18:54:31.095 INFO  i.g.d.log.ApplicationLog - Gradle Version    : Gradle 7.6  [7MB/500MB 1%]
2023-07-13 18:54:31.095 INFO  i.g.d.log.ApplicationLog - Build Timestamp   : 2023-05-21T06:27:06.392-0400  [7MB/500MB 1%]
2023-07-13 18:54:31.095 INFO  i.g.d.log.ApplicationLog - Build-JDK         : 19.0.1 (BellSoft 19.0.1+11  [7MB/500MB 1%]
2023-07-13 18:54:31.095 INFO  i.g.d.log.ApplicationLog - Build OS          : Linux (amd64 5.19.0-41-generic  [7MB/500MB 1%]
2023-07-13 18:54:31.095 INFO  i.g.d.log.ApplicationLog -   [7MB/500MB 1%]
2023-07-13 18:54:31.095 INFO  i.g.d.log.ApplicationLog - *******************************************************************  [7MB/500MB 1%]
2023-07-13 18:54:31.095 INFO  i.g.d.log.ApplicationLog - **** sdrtrunk: a trunked radio and digital decoding application ***  [7MB/500MB 1%]
2023-07-13 18:54:31.096 INFO  i.g.d.log.ApplicationLog - ****  website: https://github.com/dsheirer/sdrtrunk             ***  [7MB/500MB 1%]
2023-07-13 18:54:31.096 INFO  i.g.d.log.ApplicationLog - *******************************************************************  [7MB/500MB 1%]
2023-07-13 18:54:31.096 INFO  i.g.d.log.ApplicationLog - Memory Logging Format: [Used/Allocated PercentUsed%]  [7MB/500MB 1%]
2023-07-13 18:54:31.096 INFO  i.g.d.log.ApplicationLog - Host OS Name:          Windows 11  [7MB/500MB 1%]
2023-07-13 18:54:31.096 INFO  i.g.d.log.ApplicationLog - Host OS Arch:          amd64  [7MB/500MB 1%]
2023-07-13 18:54:31.096 INFO  i.g.d.log.ApplicationLog - Host OS Version:       10.0  [7MB/500MB 1%]
2023-07-13 18:54:31.096 INFO  i.g.d.log.ApplicationLog - Host CPU Cores:        32  [7MB/500MB 1%]
2023-07-13 18:54:31.096 INFO  i.g.d.log.ApplicationLog - Host Max Java Memory:  7 GB  [7MB/500MB 1%]
2023-07-13 18:54:31.096 INFO  i.g.d.log.ApplicationLog - Storage Directories:  [7MB/500MB 1%]
2023-07-13 18:54:31.097 INFO  i.g.d.log.ApplicationLog -  Application Root: [blanked]\SDRTrunk  [7MB/500MB 1%]
2023-07-13 18:54:31.097 INFO  i.g.d.log.ApplicationLog -  Application Log:  [blanked]\SDRTrunk\logs  [7MB/500MB 1%]
2023-07-13 18:54:31.098 INFO  i.g.d.log.ApplicationLog -  Event Log:        [blanked]\SDRTrunk\event_logs  [7MB/500MB 1%]
2023-07-13 18:54:31.098 INFO  i.g.d.log.ApplicationLog -  Playlist:         [blanked]\SDRTrunk\playlist  [7MB/500MB 1%]
2023-07-13 18:54:31.100 INFO  i.g.d.log.ApplicationLog -  Recordings:      [blanked]\SDRTrunk\recordings  [7MB/500MB 1%]
2023-07-13 18:54:31.101 INFO  i.g.d.util.ThreadPool - Application thread pool created SCHEDULED and CACHED executors threads  [7MB/500MB 1%]
2023-07-13 18:54:31.102 INFO  i.g.dsheirer.gui.SDRTrunk - Home path: [blanked]\SDRTrunk  [7MB/500MB 1%]
2023-07-13 18:54:31.105 INFO  i.g.d.p.SystemProperties - SystemProperties - loaded [blanked]\SDRTrunk\SDRTrunk.properties]  [7MB/500MB 1%]
2023-07-13 18:54:31.105 INFO  i.g.d.p.SystemProperties - SystemProperties - application properties loaded [blanked]\SDRTrunk\SDRTrunk.properties]  [7MB/500MB 1%]
2023-07-13 18:54:31.114 INFO  i.g.d.s.t.m.TunerManager - Discovering tuners ...  [8MB/500MB 1%]
2023-07-13 18:54:31.175 INFO  i.g.d.s.t.m.TunerManager - LibUsb API Version: 1.0.262  [9MB/500MB 1%]
2023-07-13 18:54:31.176 INFO  i.g.d.s.t.m.TunerManager - LibUsb Version: 1.0.22.11312  [9MB/500MB 1%]
2023-07-13 18:54:31.186 INFO  i.g.d.s.t.m.TunerManager - LibUsb - discovered [-99] potential usb devices  [9MB/500MB 1%]
2023-07-13 18:54:31.186 ERROR i.g.d.s.t.m.TunerManager - LibUsb - error during USB device discovery  [10MB/500MB 2%]
java.lang.IllegalStateException: deviceListPointer is not initialized
        at org.usb4java.LibUsb.freeDeviceList(Native Method)
        at io.github.dsheirer.source.tuner.manager.TunerManager.discoverUSBTuners(TunerManager.java:233)
        at io.github.dsheirer.source.tuner.manager.TunerManager.start(TunerManager.java:127)
        at io.github.dsheirer.gui.SDRTrunk.<init>(SDRTrunk.java:178)
        at io.github.dsheirer.gui.SDRTrunk.main(SDRTrunk.java:738)
2023-07-13 18:54:31.188 INFO  i.g.d.s.t.m.TunerManager - LibUsb Hotplug event notification Is Not Supported on this platform.  [10MB/500MB 2%]
2023-07-13 18:54:31.192 INFO  i.g.d.s.t.s.api.SDRplay - Loading API Library from default install path: C:\Program Files\SDRplay\API\x64\sdrplay_api  [10MB/500MB 2%]
...CONTINUES to load GUI successfully...
```

Intenna, 2023-07-14:
just to add to this i've tried forcing java in the batch file to java 15.X.X and re did zadig multiple times.

DSheirer, 2023-08-18:
Resolved under #1628

closed 2023-08-18 by DSheirer