DSheirer/sdrtrunk #1603: Version 0.6.0 Alpha 7 > Now Playing > Channel > Events sub panel > Duration value rendering adding rows

jpryor223, 2023-07-14:
**sdrtrunk Version**
Version 0.6.0 Alpha 7

**Describe the bug**
The Now Playing > active Channel > Events sub panel > "Duration" column value rendering process is adding new rows continuously with incrementing duration time values while an active call plays.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to and click on a Status CALL active channel in Now Playing.
2. Click on the "Events" sub panel.
3. View the incrementing Duration values and new rows being added while the channel is active.

**Expected behavior**
In previous versions (Alpha 6 and prior) the Duration column current row value would increment the duration value only, and not add new rows as the active call progressed.

**Screenshots**
Active Channel with Events Duration repeating rows (during active channel):
[image]

Idle Channel with no Events Duration repeated rows (leave channel and come back to idle channel):
[image]

**Application Log**
N/A, no error in application log.  This Bug is visible in Events sub panel.

**Desktop (optional - complete the following information):**
 - OS: Windows 10
 - CPU Cores: 4
 - RAM: 8GB

**Additional context**
Able to reproduce the Events Duration repeating rows with AM, NBFM (with and without DCS decoder enabled), P25-1 conventional channel and trunking traffic channel Decoders.
After leaving the active channel and coming back to a previous idle channel, the Events sub panel no longer shows the repeating rows.
Also noted by and further screen shots in Discord from user "M.C.B.": https://discord.com/channels/565715676749103115/565715676749103117/1129496726022131752

linuxnut79, 2023-07-15:
i also have this issue.  I have noticed that i can remove the duplicated events by selecting another channel upon refresh the single event is shown.

dave3825us, 2023-07-17:
I have this issue also. Win 10 64 bit

Ctpolicelive, 2023-07-17:
Same issue on Linux.. Also if you toggle the event slider lower and then toggle any filter on and off. Then if you turn go back and toggle the slider again it loses all channel aliases and other info in events and just repeats (Identifier Collection Timeslot) 

[image]
[image]

Seagravebuff60, 2023-07-18:
Same issue here. Im actually the original one who reported it in Discord. https://discord.com/channels/565715676749103115/565715676749103117/1129496726022131752

PR #1608 (2023-07-22): #1603 De-Duplicate Call Events
Closes #1603 

Resolves issue with de-duplicating events to allow call duration values to display correctly and minimize the quantity of duplicate events.
closed 2023-07-22 by DSheirer