DSheirer/sdrtrunk #1605: No Tuner Selected Error

CGUTH, 2023-07-16:
**sdrtrunk Version**
[Version 0.6.0 Alpha 7](https://github.com/DSheirer/sdrtrunk/releases/tag/v0.6.0-alpha-7)
Previous versions including stable branch have same issue for me.

**Describe the bug**
Program runs and decodes multiple P25 systems fine and my multiple Airspy & RTL-2832 tuners show on the Tuners tab, however when I click on any of the tuners the right pane shows "No Tuner Selected". I have to go into the tuner config file to set gain levels. I have tried fresh installs and trying only one tuner at a time with same results. Issue only happens on my Ubuntu server, works fin on my Windows 10 machines.

**To Reproduce**
See above

**Expected behavior**
When clicking on a tuner I should be able to change the tuner settings

**Screenshots**
[image]

**Application Log**
[sdrtrunklog.txt](https://github.com/DSheirer/sdrtrunk/files/12063101/sdrtrunklog.txt)

**Desktop (optional - complete the following information):**
 - OS: Ubuntu 22.04
 - CPU Cores: 56
 - RAM: 32GB
 - Dell R630 Server

**Additional context**
Tried asking for help in the discussion area and google group with no luck.
My first thought is it is a driver issue for the dongles however other sdr apps work fine with the dongles.

DSheirer, 2023-07-22:
Relevant Stack Trace:

Exception in thread "AWT-EventQueue-0" java.lang.IndexOutOfBoundsException: invalid hotSpot
	at java.desktop/sun.awt.CustomCursor.<init>(Unknown Source)
	at java.desktop/sun.awt.X11CustomCursor.<init>(Unknown Source)
	at java.desktop/sun.awt.X11.XCustomCursor.<init>(Unknown Source)
	at java.desktop/sun.awt.X11.XToolkit.createCustomCursor(Unknown Source)
	at io.github.dsheirer.gui.control.JFrequencyControl.init(JFrequencyControl.java:116)
	at io.github.dsheirer.gui.control.JFrequencyControl.<init>(JFrequencyControl.java:70)
	at io.github.dsheirer.source.tuner.ui.TunerEditor.getFrequencyControl(TunerEditor.java:281)
	at io.github.dsheirer.source.tuner.ui.TunerEditor$FrequencyPanel.<init>(TunerEditor.java:607)
	at io.github.dsheirer.source.tuner.ui.TunerEditor.getFrequencyPanel(TunerEditor.java:241)
	at io.github.dsheirer.source.tuner.ui.TunerEditor.lambda$receive$9(TunerEditor.java:491)
	at java.desktop/java.awt.event.InvocationEvent.dispatch(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.base/java.security.AccessController.doPrivileged(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.run(Unknown Source)

DSheirer, 2023-07-22:
Issue is with the JFrequencyControl editor's usage of custom cursor causes initialization error on certain Linux configurations and fails to render the tuner editor panel.  Removed usage of custom cursor to resolve issue.

PR #1611 (2023-07-22): #1605 Blank Tuner Editor Panel on Certain Linux Configurations
Closes #1605 
Resolves issue where certain Linux users were unable to display the tuner editor panel due to an issue with the frequency control editor's usage of custom cursor.
closed 2023-07-22 by DSheirer