DSheirer/sdrtrunk #1609: Option To Disable Aliasing Of Streamed Identifiers (AlphaTags)

DSheirer, 2023-07-22:
SDRTrunk currently encodes any group/user aliases that it has in the playlist and sends them as alpha tags, which is a neat feature. However I have some feeds where I'd like to disable this feature and not send any alpha tags, or at least disable the sending of the aliases (raw ID is okay).

doug-hoffman, 2023-08-01:
For icecast, #1395 would accomplish the latter with a configuration like this:

[image]

louisik1, 2023-08-21:
I think you may be able to create separate alias lists for those feeds where you keep the talkgroup name empty, or create a talk group group with all your nameless IDs in it.