DSheirer/sdrtrunk #1615: Errors when changing IF AGC Mode on SDRPlay RSPdx

ErikSwan, 2023-07-28:
**sdrtrunk Version**
sdrtrunk v0.6.0-alpha7
SDRUno package version 1.42 build 1014 (appears to include API v3.11)

**Describe the bug**
Changing the IF AGC Mode option in the Tuner settings sometimes has no effect and shows an error in the application log (see below).

**To Reproduce**
Steps to reproduce the behavior:
1. Launch sdrtrunk and select the RSPdx in the Tuners tab.
2. Click on the IF AGC Mode option and select a different option that currently selected (e.g. select "DISABLE" from "ENABLE" or vice versa).
3. Observe that there is no change in the spectrum view as would be expected for enabling/disabling the AGC, and an error appears in the application log (see below).

**Expected behavior**
I would expect that changing the AGC mode would immediately show a change in the spectrum view and that there would be no errors in the application log.

**Application Log**
```
20230728 091243.620 [main] INFO  i.g.d.log.ApplicationLog - SDRTrunk Version  : 0.6.0-alpha7  [7MB/512MB 1%]
20230728 091243.621 [main] INFO  i.g.d.log.ApplicationLog - Gradle Version    : Gradle 8.2  [7MB/512MB 1%]
20230728 091243.621 [main] INFO  i.g.d.log.ApplicationLog - Build Timestamp   : 2023-07-14T07:12:26.321-0400  [7MB/512MB 1%]
20230728 091243.621 [main] INFO  i.g.d.log.ApplicationLog - Build-JDK         : 20.0.1 (BellSoft 20.0.1+10  [7MB/512MB 1%]
20230728 091243.621 [main] INFO  i.g.d.log.ApplicationLog - Build OS          : Linux (amd64 5.19.0-46-generic  [7MB/512MB 1%]
20230728 091243.621 [main] INFO  i.g.d.log.ApplicationLog -   [7MB/512MB 1%]
20230728 091243.621 [main] INFO  i.g.d.log.ApplicationLog - *******************************************************************  [7MB/512MB 1%]
20230728 091243.621 [main] INFO  i.g.d.log.ApplicationLog - **** sdrtrunk: a trunked radio and digital decoding application ***  [7MB/512MB 1%]
20230728 091243.622 [main] INFO  i.g.d.log.ApplicationLog - ****  website: https://github.com/dsheirer/sdrtrunk             ***  [8MB/512MB 1%]
20230728 091243.622 [main] INFO  i.g.d.log.ApplicationLog - *******************************************************************  [8MB/512MB 1%]
20230728 091243.622 [main] INFO  i.g.d.log.ApplicationLog - Memory Logging Format: [Used/Allocated PercentUsed%]  [8MB/512MB 1%]
20230728 091243.622 [main] INFO  i.g.d.log.ApplicationLog - Host OS Name:          Windows 11  [8MB/512MB 1%]
20230728 091243.622 [main] INFO  i.g.d.log.ApplicationLog - Host OS Arch:          amd64  [8MB/512MB 1%]
20230728 091243.622 [main] INFO  i.g.d.log.ApplicationLog - Host OS Version:       10.0  [8MB/512MB 1%]
20230728 091243.622 [main] INFO  i.g.d.log.ApplicationLog - Host CPU Cores:        16  [8MB/512MB 1%]
20230728 091243.623 [main] INFO  i.g.d.log.ApplicationLog - Host Max Java Memory:  7 GB  [8MB/512MB 1%]
20230728 091243.623 [main] INFO  i.g.d.log.ApplicationLog - Storage Directories:  [8MB/512MB 1%]
20230728 091243.623 [main] INFO  i.g.d.log.ApplicationLog -  Application Root: C:\Users\eriks\SDRTrunk  [8MB/512MB 1%]
20230728 091243.623 [main] INFO  i.g.d.log.ApplicationLog -  Application Log:  C:\Users\eriks\SDRTrunk\logs  [8MB/512MB 1%]
20230728 091243.624 [main] INFO  i.g.d.log.ApplicationLog -  Event Log:        C:\Users\eriks\SDRTrunk\event_logs  [8MB/512MB 1%]
20230728 091243.624 [main] INFO  i.g.d.log.ApplicationLog -  Playlist:         C:\Users\eriks\SDRTrunk\playlist  [8MB/512MB 1%]
20230728 091243.624 [main] INFO  i.g.d.log.ApplicationLog -  Recordings:       C:\Users\eriks\SDRTrunk\recordings  [8MB/512MB 1%]
20230728 091243.624 [main] INFO  i.g.dsheirer.util.ThreadPool - Application thread pool created SCHEDULED and CACHED executors threads  [8MB/512MB 1%]
20230728 091243.624 [main] INFO  i.github.dsheirer.gui.SDRTrunk - Home path: C:\Users\eriks\SDRTrunk  [8MB/512MB 1%]
20230728 091243.628 [main] INFO  i.g.d.p.SystemProperties - SystemProperties - loaded [C:\Users\eriks\SDRTrunk\SDRTrunk.properties]  [8MB/512MB 1%]
20230728 091243.629 [main] INFO  i.g.d.p.SystemProperties - SystemProperties - application properties loaded [C:\Users\eriks\SDRTrunk\SDRTrunk.properties]  [8MB/512MB 1%]
20230728 091243.787 [main] INFO  i.g.d.s.t.manager.TunerManager - Discovering tuners ...  [16MB/56MB 29%]
20230728 091243.844 [main] INFO  i.g.d.s.t.manager.TunerManager - LibUsb API Version: 1.0.262  [17MB/56MB 30%]
20230728 091243.845 [main] INFO  i.g.d.s.t.manager.TunerManager - LibUsb Version: 1.0.22.11312  [17MB/56MB 30%]
20230728 091243.856 [main] INFO  i.g.d.s.t.manager.TunerManager - LibUsb - discovered [8] potential usb devices  [17MB/56MB 30%]
20230728 091243.857 [main] INFO  i.g.d.s.t.manager.TunerManager - LibUsb Hotplug event notification Is Not Supported on this platform.  [17MB/56MB 30%]
20230728 091243.862 [main] INFO  i.g.d.s.t.sdrplay.api.SDRplay - Loading API Library from default install path: C:\Program Files\SDRplay\API\x64\sdrplay_api  [17MB/56MB 31%]
20230728 091243.894 [main] INFO  i.g.d.s.t.sdrplay.api.SDRplay - API library - open status: Success  [20MB/56MB 36%]
20230728 091243.899 [main] INFO  i.g.d.s.t.sdrplay.api.SDRplay - API library v3.11 - loaded  [20MB/56MB 36%]
20230728 091243.920 [main] INFO  i.g.d.s.t.manager.TunerManager - Discovered [1] RSP devices from SDRplay API  [21MB/56MB 37%]
20230728 091243.929 [main] INFO  i.g.d.s.t.manager.TunerManager - Tuner: RSPdx SER#2237014548 - Added / Starting ...  [22MB/56MB 40%]
20230728 091243.975 [main] INFO  i.g.d.d.f.c.ComplexPolyphaseChannelizerM2 - Sample Rate [10000000.0] providing [400] channels at [25000.0] Hz each  [26MB/56MB 47%]
20230728 091243.979 [main] INFO  i.g.d.settings.SettingsManager - SettingsManager - loading settings file [C:\Users\eriks\SDRTrunk\settings\settings.xml]  [27MB/56MB 48%]
20230728 091244.217 [main] WARN  i.g.d.vector.VectorUtilities - CPU supports maximum SIMD instructions of Species[float, 8, S_256_BIT]  [42MB/56MB 75%]
20230728 091244.624 [main] INFO  i.g.d.playlist.PlaylistManager - Loading playlist [C:\Users\eriks\SDRTrunk\playlist\default.xml]  [26MB/64MB 41%]
20230728 091244.718 [main] INFO  i.github.dsheirer.gui.SDRTrunk - starting main application gui  [32MB/64MB 51%]
20230728 091246.249 [AWT-EventQueue-0] INFO  i.g.d.c.c.ChannelAutoStartFrame - Starting [1] channels now - user invoked  [150MB/288MB 52%]
20230728 091246.288 [AWT-EventQueue-0] INFO  i.g.d.a.c.mbe.JmbeAudioModule - Loading JMBE library from [C:\Users\eriks\SDRTrunk\jmbe\jmbe-1.0.9.jar]  [162MB/288MB 56%]
20230728 091246.293 [AWT-EventQueue-0] INFO  i.g.d.a.c.mbe.JmbeAudioModule - JMBE audio conversion library loaded: JMBE Audio Conversion Library v1.0.9  [163MB/288MB 56%]
20230728 091246.293 [AWT-EventQueue-0] INFO  i.g.d.a.c.mbe.ImbeAudioModule - JMBE audio conversion library IMBE CODEC successfully loaded - P25-1 audio will be available  [163MB/288MB 56%]
20230728 091246.307 [AWT-EventQueue-0] INFO  i.g.d.d.f.c.ComplexPolyphaseChannelizerM2 - Sample Rate [10000000.0] providing [400] channels at [25000.0] Hz each  [167MB/288MB 58%]
20230728 091332.032 [AWT-EventQueue-0] ERROR i.g.d.s.t.s.RspTunerEditor - Error setting AGC mode on RSP device  [253MB/672MB 37%]
```

**Desktop (optional - complete the following information):**
 - OS: Windows 11 Pro 22H2 (build 22621.1992)
 - CPU Cores: 8 cores/16 threads
 - RAM: 32GB

**Additional context**
I'm using the SDRUno package version 1.42 build 1014 (which appears to include SDRPlay API v3.11) with an RSPdx. I do not have any other SDRPlay devices so I don't know if this is unique to the RSPdx or not.

DSheirer, 2023-11-05:
@ErikSwan I do not have an RSPdx to test against.  This issue may be specific to that tuner because I can't reproduce with RSP1A or RSPduo.

DSheirer, 2026-11-07:
Closing.  sdrtrunk now uses the jSDRPlay wrapper library.

closed 2026-11-07 by DSheirer