DSheirer/sdrtrunk #1616: Audio and left/right channel alias display are out of sync

ErikSwan, 2023-07-28:
**sdrtrunk Version**
v0.6.0-alpha7

**Describe the bug**
The left/right audio channel display (where the aliases of the current talkgroups being played are shown) and the actual audio are slightly out of sync, which can make it difficult to match the audio you are hearing to the talkgroups displayed in the left/right channel information boxes, especially for short voice calls in a very busy system.

**To Reproduce**
Steps to reproduce the behavior:
1. Setup sdrtrunk as usual so that you can hear a busy trunked radio system with multiple talkgroups, where it is common to hear multiple talkgroups at the same time.
2. Observe that at the beginning of a voice call, the talkgroup alias is shown in the LEFT or RIGHT information box before the audio starts playing and disappears from the box before the audio ends. They are not synchronized. This is especially apparent for very short voice calls where the alias name may appear and disappear from the info box before the audio even starts playing.

**Expected behavior**
I would expect the audio to be perfectly synchronized with the aliases displayed in the LEFT/RIGHT channel information displays. This would make it a lot easier to keep track of which talkgroups the audio is coming from, especially when a talkgroup happens to be jumping between the LEFT/RIGHT channel and with very short voice calls.

**Application Log**
I reviewed the application log and didn't find any relevant errors.

**Desktop (optional - complete the following information):**
 - OS: Windows 11 Pro 22H2 (build 22621.1992)
 - CPU Cores: 8 cores/16 threads
 - RAM: 32GB

**Additional context**
I'm guessing this is occurring because there is latency/buffering involved in decoding the audio with the JMBE library, but no latency/buffering is added to the displayed aliases so that they are synchronized with the audio. If the audio decoding latency could be reduced, that would be ideal, but adding some buffering to the displayed aliases in the LEFT/RIGHT channel display so that it is synchronized with the audio decoding latency would also fix this.

DSheirer, 2023-11-05:
@ErikSwan sorry for the delay in working this.  I just fixed an issue with RSP tuners where the audio becomes excessively delayed.  Did you experience this out-of-sync issue when decoding from an RSP tuner?

It's possible that this issue that you've reported may have been addressed under #1695

ErikSwan, 2023-12-01:
I don't think this is the same issue. The time difference between the display and audio is subtle (maybe on the order of 200ms-400ms or so as an estimate) but I still experience it even with an Airspy R2 tuner.

It only becomes an annoyance on a busy system with lots of short calls going on back-to-back, where the delay is enough to make it hard to follow where each piece of audio is coming from.

Like I said earlier, I haven't looked into the codebase but I suspect this is coming from whatever latency/buffering is occurring in the audio decoding/audio pipeline, while the talk group display in the GUI has no such corresponding latency. So perhaps this should be considered more as a feature request for sdrtrunk to measure the audio playback latency and add corresponding latency to the talk group display so that it is correctly synchronized with the audio playback.

DSheirer, 2026-02-08:
Resolved under #2356