DSheirer/sdrtrunk #1619: View dialog windows open minimized

hans109h, 2023-07-31:
**sdrtrunk Version**
master-branch (6.0-alpha7)

**Describe the bug**
Playlist editor, DMR recording viewer, icon manager, and user preferences dialog windows open as completely minimized (vertical black bar) after one of them has been opened and then closed.

**To Reproduce**
Steps to reproduce the behavior:
1.  Start SDRTrunk
2. Open one of the dialog windows via the view menu.
3. Close the dialog window.
4. Open any other dialog window via the view menu.

**Expected behavior**
Window should open in default size and location with window contents visible.

**Screenshots**
[image]

**Application Log**
Nothing useful.

**Desktop (optional - complete the following information):**
 - OS: Ubuntu 23.04
 - CPU Cores: 8
 - RAM: Lots
 -
**Additional context**
It's been doing this for a while, I haven't had a chance to see if I can find the regression.

DSheirer, 2023-11-10:
@hans109h and @mwsltn  I'm unable to reproduce this error. 

However, I did find a code error in the JavaFxWindowManager where it wasn't monitoring the Java FX window size and position correctly.  It is supposed to monitor the windows so that it can record where the user last placed the window so that on restoration it will resize and position the window at the same place.

I'm going to submit that change against this issue and that will close out this issue.  If the change doesn't resolve this issue, please reopen this issue and we'll try again.

hans109h, 2023-11-10:
@DSheirer  unfortunately it didn't resolve the issue.  I can confirm that it does indeed remember the location (and size) of the window, so that when initially opened it returns there, but any opening of the dialog window after the initial result in the window completely minimized and only visible as a thin line on the screen.

DSheirer, 2023-11-11:
Reopening.  Will attempt to force the window out of the 'iconified' state before invoking show and focus.

DSheirer, 2023-11-11:
@hans109h I merged some additional changes to attempt to fix this issue.  The nightly build process just completed generating a new build with this change.  Can you please access the nightly build and verify if the issue is resolved?

hans109h, 2023-11-11:
Negative, issue still present.  Window opens normal on first open, after being closed it only opens in its iconified state.  If location of window is changed, the location is accurately remembered when opened even in iconified state.

hans109h, 2023-11-11:
On any open of the Playlist Editor, getting this in the syslog:
Window manager warning: Buggy client sent a _NET_ACTIVE_WINDOW message with a timestamp of 0 for 0xe00003

On open of the Icon Manager, getting this in the syslog:
mutter-x11-fram[461397]: gtk_widget_size_allocate(): attempt to allocate MetaFrameContent widget 0x55825bb76110 with width 1 and height -28
gnome-shell[459369]: *** BUG ***
gnome-shell[459369]: In pixman_region32_init_rect: Invalid rectangle passed
gnome-shell[459369]: Set a breakpoint on '_pixman_log_error' to debug
gnome-shell[459369]: *** BUG ***
gnome-shell[459369]: In pixman_region32_init_rect: Invalid rectangle passed
gnome-shell[459369]: Set a breakpoint on '_pixman_log_error' to debug

lwvmobile, 2023-11-12:
Same as above, on the latest nightly release (as of Nov 12, 2023) I get same message after opening the playlist editor once, closing it, and then reopening it, it opens in an extremely small/skinny window.

```
Window manager warning: Buggy client sent a _NET_ACTIVE_WINDOW message with a timestamp of 0 for 0x8600004
```

DSheirer, 2023-11-12:
@hans109h @lwvmobile have these warnings just started to appear with the 'tweaks' that I've been making to get the minimized windows to restore?

lwvmobile, 2023-11-12:
I just went back and downloaded Version 0.5.3 (final) and it also occurs there, so its nothing to do with any recent updates unless it predates that release somehow. I honestly think this is more to do with newer Gnome or Mutter updates in my case, but could be present in other desktop environments as well. This is using GNOME version 45.0 with Mutter window manager (Arch).

DSheirer, 2023-11-12:
I'm running KDE ... not seeing the error logging and not seeing this issue with restoring windows.  I'll have to dig a bit more to see if it's been identified elsewhere.

hans109h, 2023-11-12:
I feel like I would have noticed the syslog message back in July had it been there.  I'm currently on a later release of gnome and mutter since then, so I'm thinking the log message is likely a result of those updates rather than changes to sdrtrunk.

DSheirer, 2023-11-14:
@hans109h it seems like the window is restoring from its minimized state and it's positioning (x/y) correctly, but it appears that the width and height might be set to zero.  I'm wondering if this is a bug for JavaFX.

Could you run from master code branch and add the following logger to the end of the ```JavaFxWindowManager.restore()``` method and launch the application, open one of the JavaFX windows, close it, and then reopen it from the menu option?

```
        System.out.println("Stage [" + stage.getTitle() + "] restored - width[" + stage.getWidth() + "] height [" + stage.getHeight() + "]");

```

I'm curious what Java reports as the height and width of that stage once it thinks the window has been restored.

hans109h, 2023-11-14:
Line 1, initial playlist editor open.  Line 2, 2nd open.  After the second open I resized the iconified window and moved it, then closed it.  Line 3, third open.

Just not doing what it's told!

`Stage [sdrtrunk - Playlist Editor] restored - width[959.0] height [355.0]
Stage [sdrtrunk - Playlist Editor] restored - width[959.0] height [355.0]
Stage [sdrtrunk - Playlist Editor] restored - width[834.0] height [328.0]`

DSheirer, 2023-11-15:
@hans109h Thank you for the PR!

closed 2023-11-15 by DSheirer

DSheirer, 2024-10-27:
I reverted the changes from the PR associated with this issue here: #2047 

If this issue reappears, we'll need a different solution.

PR #1719 (2023-11-10): #1619 JavaFX Windows Show Minimized
Closes #1619 

Resolves issue where JavaFX window size and placement might show minimized after application is started and an initial JavaFX window is opened and then closed, where it impacted subsequent JavaFX window openings.  Corrects issue where JavaFX stage monitors were not being registered correctly, so that they could be deregistered prior to application shutdown.
PR #1738 (2023-11-15): Update JavaFxWindowManager.java for #1619
Fixes #1619

You pointed me in enough of a direction.
Based on comments in https://github.com/AlmasB/FXGL/issues/764
commit 65d16da9 (2023-11-15): Update JavaFxWindowManager.java for #1619

Fixes #1619