DSheirer/sdrtrunk #1621: Add audio Automatic Gain Control (AGC)

ErikSwan, 2023-08-10:
**Is your feature request related to a problem? Please describe.**
On my local P25 system, some individual users are much quieter than others. It can be frustrating listening to a diverse group of users because some users/radios are so quiet it is difficult to discern what they are saying, but if I turn up the global volume some other users are way too loud.

**Describe the solution you'd like**
It would be really nice if a feature was added to normalize audio volume (basically AGC for the decoded audio), so that all transmissions are roughly the same volume.

**Describe alternatives you've considered**
I can't think of very many other alternatives to solve the issue; perhaps a manual volume adjustment/offset per unit ID could also be effective, but that would be extremely cumbersome and complicated to properly configure, since in a large P25 system there are likely to be hundreds of unit IDs and it would be a pain to have to manually adjust the offset for each one as  you hear it (not to mention, the volume difference may be due to environmental factors like how far the user is speaking from the radio and may not be constant for a specific unit ID).

grahampaull, 2023-12-13:
Yes, this would be an excellent feature. I find that Analog conventional is very very loud. Having the ability to change the output volume for each system would be an excellent feature.

Brantlab, 2024-03-05:
Would love this but did accomplish this with VB cable and a program called https://github.com/catright/Wale
This is a good workaround but +1 for this one.

DSheirer, 2026-11-08:
Implemented in latest JMBE v2.1.1 library.

closed 2026-11-08 by DSheirer