DSheirer/sdrtrunk #1634: DMR trunked system issue in 6.0 beta 1

KrumbEMT, 2023-08-19:
Switching from build 6.0alpha7 to beta1, I didn't update a playlist (reusing the one built in a7). Trunked system (DMR Cap+) can start the control channel, but when it creates a traffic channel it only says 'No frequencies available, check LSN'.
However they are set up appropriately (RR api pull) but still the error persists. This is what the console window shows

_ org.usb4java.LibUsbException: USB error 9: error writing byte buffer: Pipe error
	at io.github.dsheirer.source.tuner.rtl.RTL2832TunerController.write(RTL2832TunerController.java:798)
	at io.github.dsheirer.source.tuner.rtl.RTL2832TunerController.write(RTL2832TunerController.java:779)
	at io.github.dsheirer.source.tuner.rtl.RTL2832TunerController.writeI2CRegister(RTL2832TunerController.java:650)
	at io.github.dsheirer.source.tuner.rtl.RTL2832TunerController$ControllerAdapter.writeI2CRegister(RTL2832TunerController.java:1550)
	at io.github.dsheirer.source.tuner.rtl.r820t.R820TEmbeddedTuner.writeR820TRegister(R820TEmbeddedTuner.java:487)
	at io.github.dsheirer.source.tuner.rtl.r820t.R820TEmbeddedTuner.setPLL(R820TEmbeddedTuner.java:398)
	at io.github.dsheirer.source.tuner.rtl.r820t.R820TEmbeddedTuner.setTunedFrequency(R820TEmbeddedTuner.java:148)
	at io.github.dsheirer.source.tuner.rtl.RTL2832TunerController.setTunedFrequency(RTL2832TunerController.java:142)
	at io.github.dsheirer.source.tuner.frequency.FrequencyController.setFrequency(FrequencyController.java:203)
	at io.github.dsheirer.source.tuner.frequency.FrequencyController.setFrequency(FrequencyController.java:175)
	at io.github.dsheirer.source.tuner.TunerController.setFrequency(TunerController.java:261)
	at io.github.dsheirer.source.tuner.manager.PolyphaseChannelSourceManager.getSource(PolyphaseChannelSourceManager.java:454)
	at io.github.dsheirer.source.tuner.manager.TunerManager.getSource(TunerManager.java:607)
	at io.github.dsheirer.source.tuner.channel.MultiFrequencyTunerChannelSource.getNextSource(MultiFrequencyTunerChannelSource.java:117)
	at io.github.dsheirer.source.tuner.channel.MultiFrequencyTunerChannelSource.rotate(MultiFrequencyTunerChannelSource.java:102)
	at io.github.dsheirer.source.tuner.channel.MultiFrequencyTunerChannelSource.process(MultiFrequencyTunerChannelSource.java:309)
	at io.github.dsheirer.source.SourceEventListenerToProcessorAdapter.receive(SourceEventListenerToProcessorAdapter.java:51)
	at io.github.dsheirer.source.SourceEventListenerToProcessorAdapter.receive(SourceEventListenerToProcessorAdapter.java:25)
	at io.github.dsheirer.sample.Broadcaster.broadcast(Broadcaster.java:128)
	at io.github.dsheirer.sample.Broadcaster.receive(Broadcaster.java:52)
	at io.github.dsheirer.source.tuner.channel.rotation.ChannelRotationMonitor.checkState(ChannelRotationMonitor.java:148)
	at io.github.dsheirer.source.tuner.channel.rotation.ChannelRotationMonitor.lambda$start$0(ChannelRotationMonitor.java:179)
	at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Unknown Source)
	at java.base/java.util.concurrent.FutureTask.runAndReset(Unknown Source)
	at java.base/java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(Unknown Source)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.base/java.lang.Thread.run(Unknown Source)
20230819 070426.833 [sdrtrunk scheduled thread 4] INFO  i.g.d.d.f.c.ComplexPolyphaseChannelizerM2 - Sample Rate [1200000.0] providing [48] channels at [25000.0] Hz each  [484MB/640MB 75%]
_

DSheirer, 2023-08-19:
Your tuner has an error (USB Error 9 - Pipe Error).  Unplug and replug your tuner may clear the issue.

KrumbEMT, 2023-08-20:
Sorry for the slow reply. Unfortunately unplugging/replugging the SDR (RTL-SDR) didn't help. Tried a restart of windows to no help. Let me know what info to provide and I'll try to be helpful

DSheirer, 2023-08-31:
The original issue that you posted where the channel to frequency map is missing was resolved under issue #1637 and PR #1646

closed 2023-08-31 by DSheirer