DSheirer/sdrtrunk #1637: DMR Trunking ‘Check Frequency Map’

kazzaw, 2023-08-21:
**sdrtrunk Version**
0.6.0b1

**Describe the bug**
Tested on DMR TIII network - when a call is granted, SDRTrunk reports CHANNEL GRANT - NO FREQUENCY - CHECK LSN CHANNEL MAP. Reprogrammed the system with the LCN instead of LSN but also yielded the same results

**To Reproduce**
Steps to reproduce the behavior:
1. Tune to DMR TIII control channel 
2. Upon call grant error will display 

**Expected behavior**
A channel grant will display and call audio will play 

**Screenshots**
https://imgur.com/a/0HS3Xh9

**Additional context**
Reverting to v0.5 alpha resolves issue

KrumbEMT, 2023-08-23:
I had the same issue ([1634](https://github.com/DSheirer/sdrtrunk/issues/1634)) with 6.0b1. Reverted back to 6.0a7 with no issues.

Seagravebuff60, 2023-08-27:
Were you able to get any audio when SDR Trunk was displaying the error "CHANNEL GRANT - NO FREQUENCY - CHECK LSN CHANNEL MAP"?

kazzaw, 2023-08-27:
No, no audio, SDRTrunk acted as if it had no channel map so didn’t even show the seperate channels

PR #1646 (2023-08-31): #1637 DMR Cap+/CapMax Channel Frequency Map / Trunking Issue
Closes #1637 

Resolves issue with DMR Cap+/CapMax trunking channel to frequency map.  Updates the DMR channel map to use LCN instead of LSN and aligns standard DMR with Tier-III DMR for channel identification.

Warning: breaking change.  Users will need to either recreate their DMR channel to frequency maps or simply delete and reload the channel from Radio Reference to correct the channel to frequency map.

Updates the DMR Decoder State to better track and update DMR events versus creating a new event and causing lots of scrolling  [... cut]
closed 2023-08-31 by DSheirer