DSheirer/sdrtrunk #1643: Support RTLSDR V4

alphafox02, 2023-08-29:
Ubuntu 22.04
Latest SDRTrunk Release

I'm testing various applications with the RTLSDR v4, and I noticed now with SDRTrunk it will not start w/ an error indicating Unsupported/Unrecognized Tuner Type. I have the latest Osmocom RTLSDR source compiled and installed w/ V4 support. 

I suspect something in here needs to be updated to account for the V4 changes?

https://github.com/DSheirer/sdrtrunk/tree/master/src/main/java/io/github/dsheirer/source/tuner/rtl

Happy to test anything since I have the V4 handy.

alphafox02, 2023-08-29:
The RTLSDR V4 is recognized as the "Rafael Micro R828D tuner"

Cpw302, 2023-08-31:
First off, thank you for your efforts on this product.  I have been a user for many years.  Unfortunately the rtl-sdr dongles I get keep changing.  I’ve purchased five new ones to try a Phase II system upload for Radio Reference and have not been able to get the tuners recognized.  Would it be possible to include some of the newer inexpensive tuners on a future release.  Specifically I’ve tried the aforementioned “Rafael Micro R828D” as well as the Fitpower “FC0012”, neither appear to be supported for SDRTrunk.  I really like your product.  Just having difficulty locating newer supported inexpensive dongles.  Thx.  Patrick

Cpw302, 2023-09-01:
Denny- I would also be willing to help test support for any of these tuners.  Thx. Patrick.

novaboxalarm, 2023-09-05:
It would be wonderful if the software could be updated to include some of the less costly applications.  I see the software looks like it wanted to include V4 and Fitpower but it does not recognize them.  I don’t know enough about programming to try to change the tuner name but as mentioned but my V4 is a “Rafael Micro R828D” with a space —- I think the program is looking for “RafaelMicro R828D” with no space.  In any event, the V3 is harder to find and I can’t really invest in the $200 versions.  I had been using an inexpensive V3 from China (blue USB) and it works great.  I need several more to run a P25 system can’t really invest in the $200 top of the line versions.

alphafox02, 2023-09-05:
Suggestion, although it may be faster to just add in the v4 supper to your code, but what about building a way for sdrtrunk to accept rtl_tcp input? Even if the user ran rtl_tcp on the same machine as sdrtrunk it would at least (I think?) push the usage of the rtlsdr dongle onto rtl_tcp itself and in turn would use the native driver from osmocom that’s been patched for the v4.

Could also be a good way to have an rtlsdr located away from the sdrtrunk PC. Just a thought.

grahampaull, 2023-10-19:
I am using V4 also, purchased 5 of them not realising they weren't supported. My bad.

DSheirer, 2023-10-19:
I have a v4 batch 2 on the way.

alphafox02, 2023-10-19:
Awesome! I fired up op25 yesterday and it did well with p25 phase 2 on 700ish MHz so I’m sure it’ll work great in sdrtrunk.

bburkett52, 2023-10-19:
How does OP25 do with simulcast P25 Phase 2?

bburkett52, 2023-10-19:
How does OP25 do with simulcast P25 Phase 2?

~Bob

jrhwr, 2023-10-20:
Do you plan to add Bias T control or let it ride with the eeprom option?

grahampaull, 2023-10-20:
> I have a v4 batch 2 on the way.

Brilliant.

closed 2023-10-22 by DSheirer

Cagier, 2023-10-22:
Thanks so much for adding this (and all the other work you do in this great project)

grahampaull, 2023-10-24:
Brilliant. Will give this a go. Thank you.

alphafox02, 2023-10-24:
That was quick! Thank you very much for adding the support. I’ll update things in my OS asap.

PR #1679 (2023-10-22): #1643 RTL2832/R828D aka RTL-SDR V4 Dongle & Bias-T Support
Closes #1643 
* Adds support for RTL2832 with embedded R828D tuner and RTL-SDR.com V4 dongles.
* Introduces Bias-T support for all RTL2832 dongles.