DSheirer/sdrtrunk #1645: DMR T3 LCN-LSN Mapping not working

Hermite123456, 2023-08-30:
**sdrtrunk Version**
Indicate which sdrtrunk version you are using.  Use 'master-branch' if you are running the latest code.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Application Log**
If applicable, include either the full application log,or a snippet of the error logging.

**Desktop (optional - complete the following information):**
 - OS: [e.g. iOS]
 - CPU Cores: [e.g. 4 cores]
 - RAM:[e.g. 16GB]

**Additional context**
Add any other context about the problem here.

Hermite123456, 2023-08-30:
SDRTrunk V6.B1 is not resolving the Frequency either by simply using the CC or by manualy populating the LSN mapping frequencies for a DMR Tier 3 Trunking TINY Network. DSDPlus and Winradio confirm the CH id (LSN), but SDRTrunk cannot resolve. Any clue? 

[image]

DSheirer, 2023-08-31:
Resolved as #1637 / #1646

closed 2023-08-31 by DSheirer