DSheirer/sdrtrunk #1659: Duplicate calls in Broadcastify Calls platform

jmp367, 2023-10-02:
I have been a provider to Broadcastify Calls for some time, however since using SDRTrunk v0.5.3 to stream two Broadcastify Feeds, I have noticed numerous duplicated calls in certain talkgroups in Broadcastify Calls. The duplicated calls seem to coincide with the talkgroups carried in the two feeds.

N8LOL, 2023-10-02:
This is a Broadcastify issue. They've allowed duplicates of sites, not sure
why. When you listen to archived recordings, you will hear calls from all
of the sites. It's disappointing, but I don't think they care anymore.

jmp367, 2023-10-02:
Indeed, right you are. I looked at the source node of each duplicated call, and one was sourced from a different node.

Thanks!

DSheirer, 2023-11-05:
Closing ... Broadcastify issue.

closed 2023-11-05 by DSheirer