DSheirer/sdrtrunk #1664: Pipe Error with new motherboard/CPU

Jeffre19, 2023-10-05:
**sdrtrunk Version**
v0.6.0-beta2

**Describe the bug**
Receive a pipe error when adjusting tuner settings, and "no tuner available" error when trying to start monitoring a system.

**To Reproduce**
Steps to reproduce the behavior:
Attempt to start a system, or change sample rate.

**Expected behavior**
Should be able to start monitoring a system.  I have tried multiple dongles, all having the same result.  I have also tried multiple USB ports, currently plugged in a USB 2.0 port.

**Screenshots**
See attached

**Application Log**
See attached

**Desktop (optional - complete the following information):**
 - OS: Windows 11
 - CPU Cores: 8 cores, 16 thread [Ryzen 7 3800X]
 - RAM:16GB

**Additional context**
I was a user of previous versions of SDRTrunk with no issues on my old setup.  After upgrading my motherboard and CPU, I am no longer able to select a tuner.  I have tried multiple dongles, multiple usb ports, reinstalling drivers to no success.

Randesign, 2023-10-20:
I would suggest attaching the entire text from the terminal window and not a screenshot of the error message. Information above what is shown in the screenshot would be useful.

DSheirer, 2026-11-07:
Closing.  sdrtrunk was updated with a new RTL driver library: jRTL

closed 2026-11-07 by DSheirer