DSheirer/sdrtrunk #1675: Sdrtrunk app Timed restart.

BUGGME22, 2023-10-20:
I run the app and stream to broadcastify.  For days on end. Sometimes I go offline like 2am to like 4am every other day.  The stream connected status show like "error" . I shut down the program and restart it..  

Maybe adding a way to restart the app on a timed schedule. 

I was thinking about doing Taskkill command to reboot the computer.  But SDRTrunk does auto launch when the PC comes up.

RonRN18, 2023-10-20:
I believe this is more of an OS-related issue. I too have issues where my SDRTrunk goes offline. As SDRTrunk runs on Java, which is relatively OS-agnostic, I run mine on an Ubuntu server (Linux). I have created a systemd service that starts SDRtrunk automatically and checks to make sure it hasn't completely crashed; completely is the key word there. If it has completely crashed, it will automatically restart itself. The problem is that sometimes it is not an entire crash but a tuner crash. 

I know it is probably silly of me to always be on the latest main branch version of SDRTrunk, when it normally is operating fine, but every once in a while, there are strings of builds that crashes the tuners. The last two updates seem to do this a couple of times a day. I then created a crontab event every two hours to restart the systemd services. I had slowed it down to every 12 hours a few months ago but returned it to two hours due to semi-frequent tuner crashes. Trying to do this in Windows is just a pain-in-the-behind, which is why I prefer tasks like this to be on a Linux machine.

JustApu, 2023-10-24:
@RonRN18 There is an sdr-trunk.bat in the release for Windows users. You may want to consider contributing your systemd setup.