DSheirer/sdrtrunk #1678: Calls Time Qualifier for Stream Uploads

Cpw302, 2023-10-21:
Thank you Denny for your wonderful program.

Requesting a Time qualifier to upload only Calls/Recordings exceeding a selectable specific duration (ie, greater than 10 seconds).  This could be added as an option within the “Streaming/Configuration” tab on SDRTrunk.

I stream a very busy P25 County Fire/EMS.  Would like to have a “Dispatch” only stream that eliminates non-dispatch communication (defined here as communications less than 10 seconds).  

This would allow listeners of very busy department to monitor “Dispatch” for the entire County and then switch over to a specific talk-group for detailed communications when an event of interest takes place.

thx

louisik1, 2023-10-21:
In the mean time, I think Trunking Recorder has that functionality.

Cpw302, 2023-10-22:
Thank you.  I looked but did not see it.  Will check again.  Currently using windows for SDRTrunk but could transition over to Linux if Trunk Recorder has that capability.  Thx again.

Cpw302, 2023-11-11:
Thank you for the Trunking Recorder recommendation.  Very nice program.  Unfortunately it appears to only allow an upload to Broadcastify “Calls” (which I already upload using SDRTrunk which works fine).   Trunking Recorder does indeed have a very nice recording time qualifier built into the program and it works well.  However, I need it to upload to a Broadcastify “Feed” (not “Calls) system which appears is not supported by Trunking Recorder.  Everything is a compromise :).

Cpw302, 2023-12-07:
Hi Denny.

Just a qualifier to this request.  I called it a “Calls Time Qualifier” but in reality it is a request to qualify individual Feeds to Broadcastify.  Sorry for any confusion.    I think uploads to Broadcastify Calls would not be qualified (ie, limited based on transmission duration).  I’m only requesting the option to place a limit on individual Feed uploads so that certain busy Feeds could have inconsequential traffic (responding, on scene, in-service) limited.  Hope that is clearer.

thx