DSheirer/sdrtrunk #1703: java.lang.OutOfMemory exceptions on macOS Sonoma 14.1

kf4lhp, 2023-11-05:
**sdrtrunk Version**
sdrtrunk 0.5.3 final

**Describe the bug**
Running sdrtrunk and rdio-scanner on a Mac Mini 2018 (Intel 3.6 GHz quad-core i3, 8 GB RAM). This system had been running macOS Ventura for over six months with sdrtrunk 0.5.1 final and rdio-scanner with no issues, only being restarted due to power failures or OS security patching. Upgraded from Ventura several days ago and am now having java.lang.OutOfMemory exceptions within five or six hours of starting the program. I've tried setting the JVM memory to four GB using the Xmx4096m and Xms4096m options, but get the same result in the same timeframe. Upgraded from 0.5.1 to 0.5.3 hoping the noted memory bug fixes would resolve the issue but no success there either. 

**Application Log**
mwsadler@lr01 bin % ./sdr-trunk   
WARNING: Using incubator modules: jdk.incubator.vector
WARNING: package com.sun.java.swing.plaf.windows not in java.desktop
2023-11-05 11:32:44.237 INFO  i.g.d.icon.IconModel - loading icons file [/Users/mwsadler/SDRTrunk/settings/icons.xml]  [16MB/4GB 0%]
2023-11-05 11:32:44.241 INFO  i.g.d.icon.IconModel - Icons file not found at [/Users/mwsadler/SDRTrunk/settings/icons.xml]  [17MB/4GB 0%]
2023-11-05 11:32:44.570 INFO  i.g.d.log.ApplicationLog - Application Log File: /Users/mwsadler/SDRTrunk/logs/sdrtrunk_app.log  [26MB/4GB 0%]
2023-11-05 11:32:44.603 INFO  i.g.d.log.ApplicationLog - SDRTrunk Version  : 0.5.3  [28MB/4GB 0%]
2023-11-05 11:32:44.604 INFO  i.g.d.log.ApplicationLog - Gradle Version    : Gradle 7.6  [28MB/4GB 0%]
2023-11-05 11:32:44.604 INFO  i.g.d.log.ApplicationLog - Build Timestamp   : 2023-04-12T03:48:32.618-0400  [28MB/4GB 0%]
2023-11-05 11:32:44.604 INFO  i.g.d.log.ApplicationLog - Build-JDK         : 19.0.1 (BellSoft 19.0.1+11  [28MB/4GB 0%]
2023-11-05 11:32:44.604 INFO  i.g.d.log.ApplicationLog - Build OS          : Linux (amd64 5.19.0-38-generic  [28MB/4GB 0%]
2023-11-05 11:32:44.605 INFO  i.g.d.log.ApplicationLog -   [28MB/4GB 0%]
2023-11-05 11:32:44.605 INFO  i.g.d.log.ApplicationLog - *******************************************************************  [28MB/4GB 0%]
2023-11-05 11:32:44.605 INFO  i.g.d.log.ApplicationLog - **** sdrtrunk: a trunked radio and digital decoding application ***  [28MB/4GB 0%]
2023-11-05 11:32:44.605 INFO  i.g.d.log.ApplicationLog - ****  website: https://github.com/dsheirer/sdrtrunk             ***  [28MB/4GB 0%]
2023-11-05 11:32:44.605 INFO  i.g.d.log.ApplicationLog - *******************************************************************  [28MB/4GB 0%]
2023-11-05 11:32:44.605 INFO  i.g.d.log.ApplicationLog - Memory Logging Format: [Used/Allocated PercentUsed%]  [28MB/4GB 0%]
2023-11-05 11:32:44.606 INFO  i.g.d.log.ApplicationLog - Host OS Name:          Mac OS X  [28MB/4GB 0%]
2023-11-05 11:32:44.606 INFO  i.g.d.log.ApplicationLog - Host OS Arch:          x86_64  [28MB/4GB 0%]
2023-11-05 11:32:44.607 INFO  i.g.d.log.ApplicationLog - Host OS Version:       14.1  [28MB/4GB 0%]
2023-11-05 11:32:44.607 INFO  i.g.d.log.ApplicationLog - Host CPU Cores:        4  [28MB/4GB 0%]
2023-11-05 11:32:44.607 INFO  i.g.d.log.ApplicationLog - Host Max Java Memory:  4 GB  [28MB/4GB 0%]
2023-11-05 11:32:44.607 INFO  i.g.d.log.ApplicationLog - Storage Directories:  [28MB/4GB 0%]
2023-11-05 11:32:44.608 INFO  i.g.d.log.ApplicationLog -  Application Root: /Users/mwsadler/SDRTrunk  [28MB/4GB 0%]
2023-11-05 11:32:44.608 INFO  i.g.d.log.ApplicationLog -  Application Log:  /Users/mwsadler/SDRTrunk/logs  [28MB/4GB 0%]
2023-11-05 11:32:44.608 INFO  i.g.d.log.ApplicationLog -  Event Log:        /Users/mwsadler/SDRTrunk/event_logs  [28MB/4GB 0%]
2023-11-05 11:32:44.608 INFO  i.g.d.log.ApplicationLog -  Playlist:         /Users/mwsadler/SDRTrunk/playlist  [28MB/4GB 0%]
2023-11-05 11:32:44.609 INFO  i.g.d.log.ApplicationLog -  Recordings:       /Users/mwsadler/SDRTrunk/recordings  [28MB/4GB 0%]
2023-11-05 11:32:44.691 INFO  i.g.d.util.ThreadPool - Application thread pool created SCHEDULED and CACHED executors threads  [31MB/4GB 0%]
2023-11-05 11:32:44.692 INFO  i.g.dsheirer.gui.SDRTrunk - Home path: /Users/mwsadler/SDRTrunk  [31MB/4GB 0%]
2023-11-05 11:32:44.693 INFO  i.g.d.p.SystemProperties - SystemProperties - loaded [/Users/mwsadler/SDRTrunk/SDRTrunk.properties]  [31MB/4GB 0%]
2023-11-05 11:32:44.693 INFO  i.g.d.p.SystemProperties - SystemProperties - application properties loaded [/Users/mwsadler/SDRTrunk/SDRTrunk.properties]  [31MB/4GB 0%]
2023-11-05 11:32:44.882 INFO  i.g.d.s.t.m.TunerManager - Discovering tuners ...  [22MB/4GB 0%]
2023-11-05 11:32:44.996 INFO  i.g.d.s.t.m.TunerManager - LibUsb API Version: 1.0.262  [22MB/4GB 0%]
2023-11-05 11:32:44.997 INFO  i.g.d.s.t.m.TunerManager - LibUsb Version: 1.0.22.11312  [22MB/4GB 0%]
2023-11-05 11:32:45.002 INFO  i.g.d.s.t.m.TunerManager - LibUsb - discovered [5] potential usb devices  [22MB/4GB 0%]
2023-11-05 11:32:45.006 INFO  i.g.d.s.t.m.TunerManager - Discovered tuner at USB Bus [20] Port [14] Tuner Class [Airspy]  [22MB/4GB 0%]
2023-11-05 11:32:45.008 INFO  i.g.d.s.t.m.TunerManager - Tuner: USB Tuner - Airspy USB Bus:20 Port:14 - Added / Starting ...  [23MB/4GB 0%]
2023-11-05 11:32:45.046 WARN  i.g.d.v.VectorUtilities - CPU supports maximum SIMD instructions of Species[float, 8, S_256_BIT]  [25MB/4GB 0%]
2023-11-05 11:32:45.079 INFO  i.g.d.d.f.c.ComplexPolyphaseChannelizerM2 - Sample Rate [10000000.0] providing [400] channels at [25000.0] Hz each  [27MB/4GB 0%]
2023-11-05 11:32:45.080 INFO  i.g.d.s.t.m.TunerManager - Tuner: USB Tuner - Airspy USB Bus:20 Port:14 - Applying Tuner Configuration  [27MB/4GB 0%]
2023-11-05 11:32:45.086 INFO  i.g.d.s.t.m.TunerManager - LibUsb - starting HotPlug detection  [29MB/4GB 0%]
2023-11-05 11:32:45.090 INFO  i.g.d.s.t.m.TunerManager - Tuner plug-in detected at USB Bus [20] Port [14] Tuner Class [Airspy]  [29MB/4GB 0%]
2023-11-05 11:32:45.093 INFO  i.g.d.s.SettingsManager - SettingsManager - loading settings file [/Users/mwsadler/SDRTrunk/settings/settings.xml]  [29MB/4GB 0%]
2023-11-05 11:32:46.014 INFO  i.g.d.p.PlaylistManager - Loading playlist [/Users/mwsadler/SDRTrunk/playlist/default.xml]  [26MB/4GB 0%]
2023-11-05 11:32:46.183 INFO  i.g.dsheirer.gui.SDRTrunk - starting main application gui  [36MB/4GB 0%]
2023-11-05 11:32:47.977 INFO  i.g.d.c.c.ChannelAutoStartFrame - Starting [1] channels now - user invoked  [79MB/4GB 1%]
2023-11-05 11:32:48.020 INFO  i.g.d.a.c.m.JmbeAudioModule - Loading JMBE library from [/Users/mwsadler/Documents/sdrtrunk/jmbe-1.0.9.jar]  [83MB/4GB 2%]
2023-11-05 11:32:48.026 INFO  i.g.d.a.c.m.JmbeAudioModule - JMBE audio conversion library loaded: JMBE Audio Conversion Library v1.0.9  [83MB/4GB 2%]
2023-11-05 11:32:48.026 INFO  i.g.d.a.c.m.ImbeAudioModule - JMBE audio conversion library IMBE CODEC successfully loaded - P25-1 audio will be available  [83MB/4GB 2%]
2023-11-05 11:32:48.035 INFO  i.g.d.d.f.c.ComplexPolyphaseChannelizerM2 - Sample Rate [10000000.0] providing [400] channels at [25000.0] Hz each  [84MB/4GB 2%]
2023-11-05 11:39:20.930 INFO  i.g.d.a.c.m.AmbeAudioModule - AMBE CODEC successfully loaded - P25-2/DMR/NXDN audio will be available  [574MB/4GB 14%]

Exception: java.lang.OutOfMemoryError thrown from the UncaughtExceptionHandler in thread "sdrtrunk audio output RIGHT thread 1"

Exception: java.lang.OutOfMemoryError thrown from the UncaughtExceptionHandler in thread "sdrtrunk scheduled thread 4"

Exception: java.lang.OutOfMemoryError thrown from the UncaughtExceptionHandler in thread "sdrtrunk scheduled thread 1"

Exception: java.lang.OutOfMemoryError thrown from the UncaughtExceptionHandler in thread "sdrtrunk polyphase ifft processor thread 1"

Exception: java.lang.OutOfMemoryError thrown from the UncaughtExceptionHandler in thread "AWT-EventQueue-0"

Exception: java.lang.OutOfMemoryError thrown from the UncaughtExceptionHandler in thread "sdrtrunk audio manager thread 1"

Exception: java.lang.OutOfMemoryError thrown from the UncaughtExceptionHandler in thread "sdrtrunk audio output LEFT thread 1"

Exception: java.lang.OutOfMemoryError thrown from the UncaughtExceptionHandler in thread "sdrtrunk dft processor thread 1"

Exception: java.lang.OutOfMemoryError thrown from the UncaughtExceptionHandler in thread "sdrtrunk scheduled thread 3"

Exception: java.lang.OutOfMemoryError thrown from the UncaughtExceptionHandler in thread "sdrtrunk dft processor thread 1"

Exception: java.lang.OutOfMemoryError thrown from the UncaughtExceptionHandler in thread "sdrtrunk polyphase channel thread 1"

Exception: java.lang.OutOfMemoryError thrown from the UncaughtExceptionHandler in thread "sdrtrunk USB tuner - bus [20] port [14]"

Exception: java.lang.OutOfMemoryError thrown from the UncaughtExceptionHandler in thread "sdrtrunk scheduled thread 2"

**Desktop (optional - complete the following information):**
 - OS: macOS Sonoma 14.1
 - CPU Cores: Intel i3, quad-core, 3.6 GHz
 - RAM: 8 GB

cfc62, 2023-11-07:
Not minimizing your problem by any means here. 

Having said that, running Sonoma on a platform like that is going to be problematic, and layering on a near real-time app like SDRTrunk is going to press things even further, which you're unfortunately finding. 

The amount of RAM that Sonoma uses just sitting there doing nothing is getting pretty extreme, as Apple tries to get AI and ML functions working locally. There are literally _hundreds_ of processes/agents/daemons running, with many being hungry for RAM. To see the list, run `launchctl list` and be prepared to see a huge list of processes you've likely never even suspected existed. 

Now, Apple _says_ the RAM on the 2018 mini isn't user upgradeable, but it most certainly is, and I've done it, it's not hard. In fact when I get a few moments I'm going to expand it again. 

[image]

You can find RAM upgrade steps in a lot of places, but iFixIt does a pretty good job: https://www.ifixit.com/Guide/Mac+mini+Late+2018+Memory+(RAM)+Replacement/115309

[image]

Even with as many iCloud, AI and ML and other services stopped as possible, this mini running Sonoma _idles_ at 7.5 Gb RAM usage, and CPU still spikes from time to time when yet another undiscovered Sonoma process lights up. 

I don't dislike Sonoma, I understand Apple's stance of trying to bring AI and ML to users in a privacy-focused way by keeping the analysis local to the machine, but it carries a penalty with it that's completely noticeable and impactful on older CPUs.

cfc62, 2023-11-07:
In tracking CPU and RAM usage over the last 30 days, it's easy to see when I upgraded from Ventura to Sonoma 14.1 directly. Again this machine is 32 GB RAM, and it's 6-core I5, and all of this usage is idle usage - it's all OS:

CPU:
[image]

RAM:
[image]

The reduction in CPU and RAM over the last few days is due to booting out unnecessary processes/agents/daemons on an experimental basis. So far I haven't broken much but it's early days.

DSheirer, 2023-11-10:
@kf4lhp if this is an OS issue, I'd like to close this ticket since it's not sdrtrunk related?  Thoughts?

kf4lhp, 2023-11-14:
After further research, I'm fine with closing it and attributing this to the OS. I've rolled it back to Ventura and these problems are persisting, so I'm going to try the 0.6.0 Beta, and failing that, I'll probably end up upgrading the RAM in this little Mac anyhow.

closed 2023-11-14 by kf4lhp