DSheirer/sdrtrunk #1737: Analog Squelch issues NBFM issues

krtz07, 2023-11-14:
**sdrtrunk Version**
Latest Verison - beta verison

**Describe the bug**
While the bug of the squatch opening briefly before transmissions seems to be corrected. When montioring the railroad frequencies 160mhz in the USA. It appears that on stronger transmissions, whenever it stops transmitting, the squelch will stay open for a while sometimes for as long as five seconds. 

**To Reproduce**
Need to listen for stronger transmissions on NBFM 
Here is an example: 
https://drive.google.com/file/d/1tOkN-Eb0Jgd3bVhPi4HeO6Eu2RHjstWC/view?usp=sharing

**Expected behavior**
The squlech would release quickly

I am using SDR Play RS1A

krtz07, 2024-03-31:
It's continuing on with the latest beta version.

doftheworld, 2024-04-27:
I have this same issue with squelch on anolog VHF. Squelch opens fine, but then "sticks" open for a couple seconds resulting in each transmission ending in "mush."

krtz07, 2024-04-27:
@doftheworld  That's exactly the issue that I am having and having it on a stream makes the feed quality differcult to listen to after a while.

grahampaull, 2024-09-12:
+1 on this.

krtz07, 2025-03-18:
Any updates on this issue?

liquidmayo, 2025-03-18:
Having the squelch as close as possible to the threshold eliminates the end of transmission squelch close delay. The more you over-squelch, the more "open" time there is at the end of transmission.

DSheirer, 2026-11-08:
Closing.  OBE with recent NBFM squelch redesign.

closed 2026-11-08 by DSheirer