DSheirer/sdrtrunk #1739: Calls Platform uploads missing back-to-back communications

ItsTechlogix, 2023-11-14:
**sdrtrunk Version**
master

**Describe the bug**
I'm using the latest 0.6.0 beta 5 of sdrtrunk. This issue has been going on since I started streaming back on version 0.5.0 beta 5. I update to the latest one each time just to see if it fixes this issue. The system is the WV SIRN system. I have 2 separate computers (each one in a separate location) running 2 different Calls nodes. Each node is for a different county and tower. The nodes do not have any overlap in talk groups nor do they overlap with other nodes in WV. One node is setup using two RTL-SDR v3 dongles. The other node is setup using two Airspy Minis. One computer is running Windows 10, while the other is using Windows 11. Each computer has at least 2 Broadcastify audio feeds and 1 Calls upload. All traffic, including back-to-back communications, are heard through the speakers of the computer and through the Broadcastify audio feeds. It's just the Calls uploads that miss those back-to-back communications. I thought this was originally a Broadcastify issue but I recently switched one of the nodes to use Trunk-Recorder and it works perfectly with uploading all traffic to the calls platform. The audio files do seem longer as it seems to wait for more traffic before stopping the recording. This could be the solution to sdrtrunk.

**Expected behavior**
Like trunk-recorder, I would like all traffic to be uploaded to the calls platform during back-to-back communications. I much prefer the easier-to-use interface of sdrtrunk.

**Application Log**
No errors show in the log.

**Desktop**
 - OS: [Windows 10 and Windows 11 (2 different machines/nodes)]
 - CPU Cores: [4]
 - RAM:[16GB]

Cpw302, 2023-11-17:
I have exactly the same issue.  Using windows 10, 0.6.0 5 Beta (although have experienced same issue with prior SDRTrunk versions).  I upload to a Feed and Calls.  Tested using the same upload for both and loose about 1/3 of the Calls uploads.  Have tried adjusting the duplicate calls settings but does not appear to have an impact.

ItsTechlogix, 2023-12-21:
Wanted to provide an update that I now have 3 separate sites being uploaded with sdrtrunk using the calls platform. All are using the latest 0.6.0 Final version of sdrtrunk. I'm still having the same issue of back-to-back communications not uploading to Broadcastify. I have a 4th site uploading to calls using trunk-recorder and it works perfectly and uploads all of the conversations. I really don't want to switch the other 3 to using trunk recorder. Please fix this issue soon

egftechman, 2024-01-22:
I'm thinking it is a Broadcastify Call dupe algorithm thing (I can't find many details on their algorithm). Trunk-recorder seems to make slightly longer recording than SDR Trunk, SDRTrunk really cuts the recording to exactly the call itself (going off of stuff published to broadcastify call, I don't have a working trunk-recording setup, only SDRTrunk),  I'll have to do some analysis of recording lengths, but my suspicion is two recording lengths of similar duration from the same talkgroup within a second or two  of each other are identified as dupes by Broadcastify Calls (I though they used unit id also, but not sure of that), so a "go ahead"..."10-8" type conversation might the the "10-8" response as a dupe..I'll maybe have to run a wireshark capture of the API calls for a while and see if I can observe that behavior, or at least verify SDRTrunk attempted to send the "missing" calls.

egftechman, 2024-01-22:
Wireshark capture won't work, it's https. I'm not finding any logging capabilities for BC API upload within SDRTrunk, am I missing them somewhere?

DSheirer, 2024-01-22:
The upload is a 2-step process.  sdrtrunk offers a call to the api and the api provides an upload link to send the audio.  If the api doesn't provide an upload link, because of duplicate detection or otherwise, then sdrtrunk doesn't send the call.

You could temporarily modify the code in the broadcastify calls broadcaster class to log out the status of each call that it attempts to upload and whether or not it uploaded the audio.