DSheirer/sdrtrunk #1740: DMR (MOT) Tier III Capacity Max - Group Calls Duration

Tbolt10033, 2023-11-15:
**sdrtrunk Version**
Version 0.6.0 Beta 5

**Describe the bug**
DMR Tier III Capacity Max system - Duration of group calls show 0 or 0.1 in length while group calls are coming in. 

**Expected behavior**
Group calls to properly show the duration.

**Screenshots**
[image]
[image]
[image]

**Application Log**
This shows on the command prompt window log when a group call happens.
2023-11-14 23:45:01.366 DEBUG i.g.d.r.AudioRecordingManager - Audio Segment detected with NO TO identifiers  [148MB/540MB 27%]

Nokoa, 2023-11-15:
I also noticed this issue with a T3 that I found while traveling.  It's as if no continuation call grants are processed?

If you start monitoring the control during an already active group call, will SDRTrunk tune to that call, or will SDRTrunk only tune to T3 calls at their very beginning?

Tbolt10033, 2023-11-15:
For me, group calls work as normal and come in except for the duration on all say either 0 or 0.1.

DSheirer, 2024-02-10:
Resolved in issue #1787

closed 2024-02-10 by DSheirer