DSheirer/sdrtrunk #1746: Convential DMR Simplex does not decode

oniell121, 2023-11-18:
**sdrtrunk Version**
v0.6.0-beta-5

**Describe the bug**
A channel setup as a conventional, simplex DMR channel does not decode. Events appear to be jumbled and it isn't sure what timeslot to use.

**To Reproduce**
Steps to reproduce the behavior:
1. Playlist editor, add channel for 146.520MHz setup as DMR.
2. Create alias list with DMR Talkgroup ID rage of 1-MAX.
3. Play channel.
4. Key up Motorola SL300 setup with the VHF calling Freq. as Digital.
5. Watch waterfall for signal.

**Expected behavior**
SDRTrunk will see a call and decode the voice.

**Desktop (optional - complete the following information):**
 - OS: WIndows 10 Pro
 - CPU Cores: 8 cores
 - RAM: 16GB

**Additional context**
Multiple .bits demodulated baseband recordings attached.
[recordings.zip](https://github.com/DSheirer/sdrtrunk/files/13401662/recordings.zip)

oniell121, 2023-11-29:
To add clarity, I have recorded a screen grab and simultaneous bitstream recording.

Here's the video: https://youtu.be/mMKyndsLwYc

The bitstream is attached. There were not filters applied to the view at all in the GUI.

Also, there is NO repeater involved. This is literally between to Motorola SL300/3500 handhelds on a purely simplex setup. Forgive the power of the transmission. The Airspy is in the same room as the handheld and even on a rubber duck with low power, it's nice and powerful.
[20231129_110927_146520000_9600BPS_DMR_52_Simplex_SITE_52_Simplex_3.zip](https://github.com/DSheirer/sdrtrunk/files/13504213/20231129_110927_146520000_9600BPS_DMR_52_Simplex_SITE_52_Simplex_3.zip)

DSheirer, 2023-11-30:
@oniell121 Thanks!  I think I have the issue with assigning the voice frames to the correct timeslot (1 or 2) resolved and this should allow the calls to be correctly decoded.  I'm going to finish testing tomorrow and merge this branch to master and then I'll have a new version (ie the nightly build) that you can try out to see if it resolves this issue.

However, I'm also concerned that the DMR decoder may not be decoding the symbols optimally.  I have a half-complete effort to redesign the DMR decoder that uses an equalizer and it seems very promising.  However, I don't have good sample baseband recordings to test against for this simplex repeater scenario.  Would you be able to generate a couple channel baseband recordings and attach here?  You can do this in the same way that you turned on the (.bits) recordings ... just enable the (control) channel baseband recorder.

DSheirer, 2023-12-01:
@oniell121 the changes to support DMR simplex repeater are merged into master.  If you have an opportunity, can you please test with the nightly build release to see if this fixes the issue for you?

closed 2023-12-01 by DSheirer

Nokoa, 2023-12-02:
@DSheirer I tested the latest with a TYT MD380 with DMR Simplex, I was not able to get continuous voice decoded. During multiple tests I did, I was sometimes able to make out a word of what was being transmitted. Issue [#1260](https://github.com/DSheirer/sdrtrunk/issues/1260) might be related. 

Attached are baseband recordings made in DMR simplex mode, as well as DMR Simplex with DCDM on and set to leader. Slot 1 is programmed for both. TGID 11 RID 311287 CC 5
[dmr-simplex-baseband-446006250.zip](https://github.com/DSheirer/sdrtrunk/files/13537673/dmr-simplex-baseband-446006250.zip)

oniell121, 2023-12-03:
Agreed. While I now get an occasional momentary blip of decode, I cannot even make out a word. Forward progress from before, but only a bit.

I have also attached recordings of the bitstream and baseband.
[new-recordings-inc-baseband.zip](https://github.com/DSheirer/sdrtrunk/files/13538108/new-recordings-inc-baseband.zip)