DSheirer/sdrtrunk #1747: Feature request: Save Window layouts

bburkett52, 2023-11-19:
**Is your feature request related to a problem? Please describe.**
This feature request is not related to any problem, just user convenience. Window layouts are not saved and must be re-done after each startup.

**Describe the solution you'd like**
I would like to suggest that all window layouts for window size and position as well as column layouts and window section sizes be saved for next program launch. For example I do not use the spectrum or waterfall displays and shrink them vertically to allow more space for events, etc.

**Describe alternatives you've considered**
It is necessary to resize all layouts after each start up.

**Additional context**
None

Thank you for all you work on this project!

GTR8000, 2023-11-20:
+1 on this suggestion. I also don't use the spectrum/waterfall displays, and each time I start up SDRTrunk I have to drag them smaller again.

LCEbay, 2023-11-29:
+1 from me as well.  Especially disabling the spectrum/waterfall display, since they take additional CPU resources.  I only use the spectrum display when I'm troubleshooting something specific, or looking for new signals.

Brad-git-man, 2023-11-30:
It is extremely annoying every time you start SDRT to have to play "musical columns" b\c it won't save your column positions & sizes on close.

anotherbob67, 2024-01-26:
Yes, I’m running on an Orange Pi3 and want to have a single on/off button with an optimized display.