DSheirer/sdrtrunk #1752: P25 UDT Short Data Packets - Unrecognized Format (IP Address)

DSheirer, 2023-11-28:
User reported instances of unrecognized format.  Investigate to see if we can add support for these:

2023-11-27 08:25:41.370 WARN  i.g.d.m.d.d.m.d.p.UDTShortMessageService - Unrecognized UDT Short Data Format: IP ADDRESS - Please send this to the sdrtrunk developer - Hex:03000001341F9EF6D668513EF89A7FC3BD691805266C1250562C0000000000000AFF4EFB As String:4∩┐╜∩┐╜∩┐╜hQ>∩┐╜∩┐╜├╜i&lPV,
∩┐╜N∩┐╜  [380MB/1GB 34%]

2023-11-27 08:52:44.067 WARN  i.g.d.m.d.d.m.d.p.UDTShortMessageService - Unrecognized UDT Short Data Format: IP ADDRESS - Please send this to the sdrtrunk developer - Hex:03000001341F9EF6DD28513F118608C3C95EF701216C0334568600000000000025A22B58 As String:4∩┐╜∩┐╜∩┐╜(Q?∩┐∩┐╜∩┐╜^∩┐╜!l4V∩┐╜%∩┐╜+X  [675MB/1GB 61%]

2023-11-27 09:46:51.818 WARN  i.g.d.m.d.d.m.d.p.UDTShortMessageService - Unrecognized UDT Short Data Format: IP ADDRESS - Please send this to the sdrtrunk developer - Hex:03000001341F9EF6EBAC513F00A68DC3C7686B007C6C1350562C0000000000000B1C5647 As String:4∩┐╜∩┐╜∩┐╜Q?∩┐╜∩┐╜∩┐╜∩┐╜hk|lPV,
VG  [891MB/1GB 81%]

infecticide, 2024-01-03:
I'll start adding my UDT packets here if that's ok!

Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:4A297AA67F0C56542219A549DDEFF33BAFA310057AA67F0C56542219A549DDEFF33BAFA3 As String:J)z� VT" �I���;�� z� VT" �I���;��

Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:020B0C0000030FA50FA5000C0D5C0002BF300587000000000000000000000000345A82FA As String:  � �  \ �0 � 4Z��

Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:0160030000015147BF45E1B59FC93F09596C025F0F5956860000000000000000A8E2E060 As String: ` QG�Eᵟ�? Yl _ YV� ���`

infecticide, 2024-01-05:
Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:4308336C040D562D83065066 As String:C 3l V-� Pf

infecticide, 2024-01-07:
Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:9A84DC2D32CC56542219A86BDD92C7E9C723EA29FAAA71B886BCB219A87D47FA9273A1A34A297AA5321C56542219A5490CBFF33BAFA3018441104157C60694E5AA2802131F161522005005005005005019DE87ED0C60BA23FEED12815E7D3888 As String:���-2�VT" �kݒ���#�)��q���� �}G��s��J)z�2 VT" �I��;�� �A AW� ��( " P P P އ�`�#�� �^}8�

Unrecognized UDT Short Data Format: VENDOR SPECIFIC-8 - Please send this to the sdrtrunk developer - Hex:2BE14B5AE0A53FE1F0C26E2089A86D702836 As String:+�KZ�?���n ��mp(6

Unrecognized UDT Short Data Format: VENDOR SPECIFIC-8 - Please send this to the sdrtrunk developer - Hex:375C6D304B95A86A5129AC0888A86E902287 As String:7\m0K��jQ)� ��n�"�

Unrecognized UDT Short Data Format: VENDOR SPECIFIC-8 - Please send this to the sdrtrunk developer - Hex:2C619E31ABB13DA92B836A0C5AEA6E902811825C5AE368171A84222B6507CAEDA0CA0699825C5AE368171A84222B6507CAEDA0CA06998D28DA035810D4A904F82CA518476080869E8D28DA035810D4A904F82CA518476080869E8D28DA035810D4A904F82CA518476080869E8D28DA035810D4A904F82CA518476080869E8D28DA035810D4A904F82CA518476080869E8D28DA035810D4A904F82CA518476080869E8D28DA035810D4A904F82CA518476080869E8D28DA035810D4A904F82CA518476080869E As String:,a�1��=�+�jZ�n�( �\Z�h �"+e ��� ��\Z�h �"+e ��� ��(� X ԩ �,� G`����(� X ԩ �,� G`����(� X ԩ �,� G`����(� X ԩ �,� G`����(� X ԩ �,� G`����(� X ԩ �,� G`����(� X ԩ �,� G`����(� X ԩ �,� G`���

Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:15602B029A6F4A99404A558B4A29FAB75A8C56542219A549DDEFF33BAFA3 As String: `+ �oJ�@JU�J)��Z�VT" �I���;��

Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:330403389A0B95CB51731309AB91270D000283080CD08AAF As String:3 8� ��Qs ��' � Њ�

Unrecognized UDT Short Data Format: VENDOR SPECIFIC-8 - Please send this to the sdrtrunk developer - Hex:2BE14B5AE0A53FE1F0C26E2089A86D702836 As String:+�KZ�?���n ��mp(6

Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:4A297AA5321C56542219A86BDD92C7E9C723A2A5D5AE73A63A16053A7EB170B2CBF430AA14D292FFA153215EDBC591C522C5E463CE80DE7FD7EADB53ABE805DC9E3992338FD21608D15776BCFFE63A16053A7EB160502DA9DF7A0C60BA23FEED92805E753888EA006C550A465384588D4643 As String:J)z�2 VT" �kݒ���#��ծs�: :~�p���0� Ғ��S!^�ő�"��c΀� ���S�� ܞ9�3�� �Wv���: :~�`P-��z`�#�풀^u8�� lU

Unrecognized UDT Short Data Format: VENDOR SPECIFIC-8 - Please send this to the sdrtrunk developer - Hex:84100050176C0202CE028A2F8D28DA035810D4A904F82CA518476080869E370D9A994DB651A904C208251841606A069E23883832EE15B4A820C36BA9514038175B8C As String:� P l � �/�(� X ԩ �,� G`���7 ��M�Q� � % A`j �#�82� �� �k�Q@8 [�

Unrecognized UDT Short Data Format: VENDOR SPECIFIC-8 - Please send this to the sdrtrunk developer - Hex:2E84DE8A0817ECA9112A06A14511CFCA3E2C As String:.�ފ � * �E ��>

Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:80A44514592E0A00D324A5E18302160C18139621DF8041F00396D2FD5B27FE7FC702DA13D5290D39DBC0A00547141114401DC9A9CB7B As String:��E Y.

Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:D26CA337F38E38B38B38B38B38BCCCF0BD8040954714E830065AF061B0840C60BA23FEED12815E7D38884A29FAA5323886BD163085490CBFF33BAFA340C5431441284017FAA1962A As String:�l�7�8��8��8���@�G �0 Z�a��`�#�� �^}8�J)��28�� 0�I��;��@�C A(@ ���*

Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:9A84DB775A8C56542219A549DD92C7E9C723 As String:���wZ�VT" �Iݒ���#

Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:8085471411186C12E9E9933B8B9800449B13B0019502B40C As String:��G l ��;�� D� � � �

Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:4A29FAB75A8C56542219A87D47FA9273AFA3A670090050146FF5082C0D1EB65E2FE82400667918E90D06B17E38FC8C421B742FEA217ADFDF263A63A63A63A63A63A1731757F2B6952D31610B841F9E0A0068C0AA210F1A06A9FF57D75DF5DAEA14E30902141D9E0E606229BC200AA00689FF57D75DF1C2E2D1FF262C0BD0BD7EB16053A7F7FF2320DF33390A00385A139A01DB002A1A As String:J)��Z�VT" �}G��s���p P o� , �^/�$ fy � �~8��B t/�!z��&:c�:c�:c�s W�-1a � �

Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:A937652F247912979E79216D892A9F603E2A As String:�7e/$y ��y!m�*�`>*

Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:004342000033CB000500401648B42840500BC1C232241C03DE055063025F4A417AA7DA8C5654221A05490CBFF338EA01 As String: CB 3� @ H�(@P ��2$ � Pc _JAz�ڌVT" I��8�

Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:4A217AAA7A8C56542219A5490CBFF33BAFA3 As String:J!z�z�VT" �I��;��

Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:4A417AA7DA8C5654221A05490CBFF338EA014A417AA7DA8C5654221A05490CBFF338EA01DD06D32F32DEB042D7FFFEBFEAFE4BBFE3F14A29FAA5330CBCD4221A0549DDEFF33BAFA3 As String:JAz�ڌVT" I��8� JAz�ڌVT" I��8� � �/2ްB������K���J)��3��" I���;��

Unrecognized UDT Short Data Format: IP ADDRESS - Please send this to the sdrtrunk developer - Hex:3AE2D93A17170B2099DFAA242CB9EF4681F900244014710C4815E39493AC80A441146830461CE69AA2EB5946A267726FFD63A63A2000A00A49A7746A4A29FAA7DA8C56542219A5490CBFF33BAF7B4A29FAA45A8C56542219A549DDEFF33BAFA34A29FAB75A8C56542219A87D47FA9273AFA3200541140800045BFC83EA2E4024415470180E59E28C98A4 As  [... cut ...]
Unrecognized UDT Short Data Format: IP ADDRESS - Please send this to the sdrtrunk developer - Hex:4A297AA533B886BDA219A86D47FA5273AFA32025431448100469EC1DBE42B677A16F33A40707B1AA4D9A339CABD459C1A61A74F61BD0A742E1671671605B3CE12549 As String:J)z�3���� �mG�Rs�� %C H i� �B�w�o3� ��M�3���Y�� t� ЧB�g q`[<�%I 

Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:03150C00000E0FA50FA5000B588400017400043B00000000000000000000000035BC6CC5 As String:  � � X� t ; 5�l�

Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:C4E2C7FD5A06DA68F61B7E34F7E33C38E56931880B695947BF19808E44F3 As String:����Z �h� ~4��<8�i1� iYG� ��D�

infecticide, 2024-01-08:
Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:0DC21C0C18138E00D7421839BE4BE16C7FEA2A37762F170BD0BCE722B029 As String: �  � �B 9�K�l �*7v/ м�"�)

Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:4A29FAB75A8C56542219A5490CBFF33BAFA34A29FAA5333886BC6219A5490CBFF33BAFA310057AA533B886BDA219A54973FA5273AFA382780D633AC0C19017AE109480411044ADFF57D75DF3C0C19A84DC2D330C56542219A5490CBFF338E3A3 As String:J)��Z�VT" �I��;��J)��38��b �I��;�� z�3���� �Is�Rs���x c:��� � ��A D��W�]������-3VT" �I��8�

Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:A9EEC7FD166A2A37762F177FF7FF8C92FA82DC291D6B30F2FD7FD0BD7EB170B21362BC494A297AA67F0C56542219A549DD92C7E9C723D23925EB33A3910A60D9C905E3416ED374490820BA23FEED12815E7D38880860BA23FEAD12815E753888 As String:���� j*7v/ �������) k0�� н~�p� b�IJ)z� VT" �Iݒ���#�9%�3��

Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:9A84DC2D32CC56542219A5490CBFF33BAFA31040DB7727B886BDA219A86D47FA5273AFA3 As String:���-2�VT" �I��;�� @�w'���� �mG�Rs��

Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:C733A7EB3308DA27C2AF00004048EF9E5E1900C44514C9204213DB01673B35A30A68163B100EB3C401C14A297AA5321C56542219A5490CBFF33BAFA34A297AA5321C56542219A5490CBFF33BAFA3 As String:�3��3 �'¯ @H�^ �E � B � g;5�

Unrecognized UDT Short Data Format: VENDOR SPECIFIC-8 - Please send this to the sdrtrunk developer - Hex:225D8FE356A4504170ED46B95BEC542A320C As String:"]��V�PAp�F�[�T*2

Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:14A309006C05FE0B90081084F14150068DFF57D75DF1D2E8 As String: � l � � ��AP ��W�]���

infecticide, 2024-01-08:
Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:EA417AAA269086BCB481A843DFFAB329C88181402EE45C639E18D2961B36EA29FAAA5A3886BCB219A5490CB787E9C8A9 As String:�Az�&������C���)ȁ�@.�\c� Җ 6�)��Z8��� �I���ȩ

Unrecognized UDT Short Data Format: MS OR TG ADDRESS - Please send this to the sdrtrunk developer - Hex:2003333035000000BD6F2796 As String: 305 �o'�

Unrecognized UDT Short Data Format: VENDOR SPECIFIC-8 - Please send this to the sdrtrunk developer - Hex:52E094794BB0D7282597613B8040016A06A152E094794BB0D7282246613B8040016A06A152E094794518D7282597613B8040016A06A152E094794BB0D7282597613B8040016A06A1 As String:R��yK��(%�a;�@ j �R��yK��("Fa;�@ j �R��yE �(%�a;�@ j �R��yK��(%�a;�@ j �

Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:030000015147C20F7DB5A23E As String: QG� }��>

infecticide, 2024-01-10:
Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:0C60BA23FEAD02905E7D38880860BA23FEED02805E7D38980C60BA23FEED12805E753888C05A5CF72373BE96D970B38B38B231C66E980C60BA23FEED12805E7538980C60BA23FEED02805E753888EA29FAAA71B886BC6219A5490CB9333955A30820BA23FEAD12905E7D3A881BFFED2CE8CDB2968CC6CEA7AB0E319E46EADE7DA2205EA63A17A16053A170B8E137E37FF004DB64323886BC6219A5490CBFF338E2A91004DB64339E1CD42219A54973FA8DB3AA3B082001493C059E2FE0005053A41B1016A9FE57D65DF5DAC8083051087C05E62FC00CB053001C2406A9FE57D65DF5D2C8DCDA12E3A01474AAB1859578EAB7691567C9 As String:`�#�� �^}8� `�#�� �^}8�`�#�� �^u8��Z\�#s���p��8�1�n�`�#�� �^u8�`�#�� �^u8��)��q���b �I�39U� �#�� �^}:� ��,�Ͳ���Χ� 1�F��}� ^�: �`S�p��7� � �d28��b �I��8� �d3� �" �Is����; I< �/� PS� ��W�]��� 0Q | �/��S $ ��W�]����� � t����x�i g�

Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:0C20BA23FEAD12905E7D3A884D080110DF12BC81D54446FAF004DC2D33B886BC6219A54973FA95B3A7230C20BA23FEAD12905E7D3A880C20BA23FEED12905E7D3A980C20BA23FEAD12905E7D3A88EA418E90330886BCB21B85490CB787E9C8810C60BA23FEAD12805E753888A62A05AF1BD0BD5BD0BD7E90A05178F231E114A38B610ECCB998B30F02368084E216B8FE57D65DF5D1E41004D8177A8C56542219A5490CB2C7E9BA3B1004DC2D5A8C56542219A5490CB2C7E9BA3B As String: �#�� �^}:�M � ���DF�� �-3���b �Is����# �#�� �^}:� �#�� �^}:� �#�� �^}:��A��3 ��� �I���ȁ`�#�� �^u8��* � н[н~��Qx�1� ��a ̹�� 6��� ��W�]��� � z�VT" �I���; �-Z�VT" �I���;

Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:EA29CE90331E1CD42219A5490CB787E9C8A9EA29FAAA7E3886BC6219A5490CBFF33AC8A9EA29FAAA5A3886BCB219A5490CB787E9C8A9 As String:�)ΐ3 �" �I���ȩ�)��~8��b �I��:ȩ�)��Z8��� �I���ȩ

Unrecognized UDT Short Data Format: UNKNOWN - Please send this to the sdrtrunk developer - Hex:030000015147C2065EB5A23C As String: QG� ^��<

Unrecognized UDT Short Data Format: IP ADDRESS - Please send this to the sdrtrunk developer - Hex:EA417AAA5A3886BCB219A5490CB7876348A9 As String:�Az�Z8��� �I��cH�

Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:4A2F7AB75A8C56542219A5490CB2C7E9BA05C85205A06752FD5FD7FD0AA971991E57CB47EA29CE90331E1CD42219A5490CB787E9C8A90860BA23FEAD12805E7538881004D8177AAADCD42219A5490CBFF33BAEA90860BA23FEAD12815E7538884A29FAB75A8C56542219A5490CBFF33BAFA34A29FAB75A8C56542219A5490CBFF33BAFA30C60BA23FEAD92805E753898EA29FAAA5A3886BCB219A5490CB787E9C8ABBEDF232F15562E16DE37FEAFEAA6DEE69F884A29EC2D5A8C56542219A5490CB2C7E9BA3B0C60BA23FEED92805A7538980C60BA23FEED92805E753898EA29FAAA71B886BC6219A549DDEFF338E3A34A217AAA71B886BCB219A86D47FA926CE3A3911C08909F138003D4C097ACA2A09BE67270FFA17FF7FEA973F060A9C7C90860BA23FEAD12805E753988 As String:J/z�Z�VT" �I��� �R �gR�_��

Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:0C60BA23FEED12805E7538980C60BA23FEAD12805E753888F004DC2D5A3886BC6219A5490CB933298A3B0C60BA23FEAD92805E753898F004DC2D5A3886BC6219A5490CB2C7E98A3B1004DC2D5A8C56542219A5490CB2C7E9BA3BC94E1EE11458A732DE37FEA973F190DC74AF2C34EAE90A648B179E792E85037015A6CA5D As String:`�#�� �^u8�`�#�� �^u8�� �-Z8��b �I�3)�;`�#����^u8�� �-Z8��b �I���; �-Z�VT" �I���;�N � X�2�7��s��t�,4��

Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:EA417AABDA3886BCB219A5490CB787E9C8AB As String:�Az��8��� �I���ȫ

Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:0C60BA23FEAD12805E7D38880C60BA23FEAD02805E7538880860BA23FEAD12805E7D38880C60BA23FEED12805E7538880860BA23FEAD12905E7538889A84DC2D32CC56542219A5490CB93329BA05 As String:`�#�� �^}8�`�#�� �^u8� `�#�� �^}8�`�#�� �^u8� `�#�� �^u8����-2�VT" �I�3)�

Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:0C60BA23FEAD02905E7538880860BA23FEED02805E7538980860BA23FEAD02905E7D3888C732D7FD5A028C42C802CAB270A7BC547551 As String:`�#�� �^u8� `�#�� �^u8� `�#�� �^}8��2��Z �B� ʲp��TuQ

Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:EA290E90330886BCB219A5490CB7876348A9 As String:�) �3 ��� �I��cH�

Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:F004DB7B7E3886BD16C885490CB7876348AEEA417AA8330886BCB219A5490CB787E9C8A9 As String:� �{~8�� ȅI��cH��Az�3 ��� �I���ȩ

infecticide, 2024-01-10:
Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:030000015147CD9EC0B5979853090C6C00035600551D177E As String: QG͞����S l V U ~

Unrecognized UDT Short Data Format: IP ADDRESS - Please send this to the sdrtrunk developer - Hex:022809421C0786184032386CC12EE206A9FE57D75DF5DAEAA1AA716716716716F16F1605EAA1E49CB97D08302100984218B9501800FF85B69116B8FE57D75DF5D0E11040DB7727B886BDA219A54973FA5273AFA3A0DFC7FE4D18E97EAFEBFE8558B9F80BA6C3 As String: ( B � @28l�.� ��W�]��ꡪqg qg �o �䜹} 0! �B �P ���� ��W�]��� @�w'���� �Is�Rs������M �~����X�� ��

infecticide, 2024-01-12:
Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:4A297AA533B886BDA219A54973FA5273AFA3 As String:J)z�3���� �Is�Rs��

Unrecognized UDT Short Data Format: VENDOR SPECIFIC-8 - Please send this to the sdrtrunk developer - Hex:375C6D304B95A86A5129AC0888A86E902287 As String:7\m0K��jQ)� ��n�"�

Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:4A29FAB75A8C56542219A5490CBFF33BAFA340554514480C441AF12185884A29FAB75A8C56542219A5490CBFF33BAFA34A29FAB75A8C56542219A87D47FA9273AFA3C04C42FD516D0E5A855A7A7750742D5567D8 As String:J)��Z�VT" �I��;��@UE HD �!��J)��Z�VT" �I��;��J)��Z�VT" �}G��s���LB�Qm Z�ZzwPt-Ug�

Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:4A297AA67F0C56542219A549DDEFF33BAFA320A4451479003609FE80A6F94A297AB75A8C56542219A5490CB2C7E9C723 As String:J)z� VT" �I���;�� �E y 6 ����J)z�Z�VT" �I����#

Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:C4C17C6DDB0BC07EA8EA6390037005204A5D As String:��|m� �~��c� p J]

Unrecognized UDT Short Data Format: UNKNOWN - Please send this to the sdrtrunk developer - Hex:EACFEBC74A05F47EEE6791575BC2486CAD9B As String:����J �~�g�W[�Hl��

Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:030000015147BD60AAB5AAC03E07196C00075600117CD8D0 As String: QG�`����> l V |��

Unrecognized UDT Short Data Format: VENDOR SPECIFIC-8 - Please send this to the sdrtrunk developer - Hex:375C6D304B95A86A5129AC0888A86E902287 As String:7\m0K��jQ)� ��n�"�

Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:011E030000015147C5BF1FB5A5FEAE02006C036A0838568700000000000000003C4694A9 As String: QGſ ���� l j 8V�

infecticide, 2024-01-14:
Unrecognized UDT Short Data Format: UNKNOWN - Please send this to the sdrtrunk developer - Hex:470C0C56510FDA39C384C941 As String:GVQ �9Ä�A

Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:FB07596C0000560026CA3D59 As String:� Yl V &�=Y

Unrecognized UDT Short Data Format: UNKNOWN - Please send this to the sdrtrunk developer - Hex:14C313155FFB686C99DFA62376B9968C29EC As String: � _�hl�ߦ#v���)�

Unrecognized UDT Short Data Format: VENDOR SPECIFIC-8 - Please send this to the sdrtrunk developer - Hex:375C6D304B95A86A5129AC0888A86E902287 As String:7\m0K��jQ)� ��n�"�

Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:030000015147BAC8FDB59ABD4104196C0479568528191784 As String: QG������A l yV�( �

 Unrecognized UDT Short Data Format: VENDOR SPECIFIC-8 - Please send this to the sdrtrunk developer - Hex:375C6D304B95A86A5129AC0888A86E902287 As String:7\m0K��jQ)� ��n�"�

Unrecognized UDT Short Data Format: VENDOR SPECIFIC-8 - Please send this to the sdrtrunk developer - Hex:23883832EE15B4A820C36BA9514038175B8C As String:#�82� �� �k�Q@8 [�

Unrecognized UDT Short Data Format: VENDOR SPECIFIC-9 - Please send this to the sdrtrunk developer - Hex:00DCAB00010C5147C20778BB2978094F7F6C01048587563A48000000000003A49048B254 As String: ܫ QG� x�)x O l ��V:H ��H�T

infecticide, 2024-01-16:
Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:030000015147C1B165B5A23FC20D336C000556002E08764D As String: QG��e��?� 3l V . vM

Unrecognized UDT Short Data Format: UNKNOWN - Please send this to the sdrtrunk developer - Hex:0197450000337C2300004011EA910C00000B02980D00FAFA0FA10FA1001FEDDD0D152203052F0000015147BD3CF0B5A4B26703116C000751025650000000000000000000FD896993 As String: �E 3|# @ � � �� � � �� " / QG�<𵤲g l Q VP ��i�

Unrecognized UDT Short Data Format: MS OR TG ADDRESS - Please send this to the sdrtrunk developer - Hex:EB0501E01D00140A0110E3282CA00ACF0723D93849C496B08AE103214030C9305047D420013100000000000010D35E95 As String:� �,

Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:3F1150800C0001310D00FAFA0FA50FA50012A7C30008F0200333303500000000A14BD992 As String:? P� 1 �� � � �� � 305 �Kْ 

 Unrecognized UDT Short Data Format: VENDOR SPECIFIC-9 - Please send this to the sdrtrunk developer - Hex:031507CB3228400D49B33683 As String: �2(@ I�6�

Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:030000015147C206C6B5A23E As String: QG� Ƶ�>

Unrecognized UDT Short Data Format: UNKNOWN - Please send this to the sdrtrunk developer - Hex:002A45C008FB2F2A40004001F0784D00FAFA As String: *E� �/*@ @ �xM ��

Unrecognized UDT Short Data Format: MS OR TG ADDRESS - Please send this to the sdrtrunk developer - Hex:6FD04AF15E63429246499395410ED9ABD1C0417E8C871800D2B1FC14504C00241DCA5A160A99E79B2E41A84F4D21E9F202313500000000007ADFE4ED As String:o�J�^cB�FI��A ٫��A~�� ұ� PL $ �Z,

Unrecognized UDT Short Data Format: UNKNOWN - Please send this to the sdrtrunk developer - Hex:03BA0D00FAFA0FA10FA1001FE5180D15220305B20000015147BD40B4B5A4B44206376C000796025650000000000000000000510148B8 As String: � �� � � � " � QG�@����B 7l � VP Q H�

Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:3F119C650C000A114D80FAFA3EA00F874940BE2B4007F03002323000000000004FE6362E As String:? �e,

Unrecognized UDT Short Data Format: UNKNOWN - Please send this to the sdrtrunk developer - Hex:0029450000337A1C00004011EC980C00000B03870D00FAFA0FA10FA1001FF8170D15220305B30000015147BD3A64B5A4B47205586C0006610356430000000000000000006A752E86 As String: )E 3z @ � � �� � � � " � QG�:d���r Xl a VC ju.�

Unrecognized UDT Short Data Format: UNKNOWN - Please send this to the sdrtrunk developer - Hex:A9BF33BD5BD53A7EB170BD0BD0BA868DD25C6E640F59E39D98066801903304B39949FC16EE13B850B983881A300689FE57D65DF5D2E0 As String:��3�[�:~�p� к���\nd Y㝘 h �3 ��I� � �P��� 0 ��W�]���

Unrecognized UDT Short Data Format: VENDOR SPECIFIC-8 - Please send this to the sdrtrunk developer - Hex:F004DC30330886BCB219A5490CB787E9C8AE As String:� �03 ��� �I���Ȯ

 Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:35C06A128E7D08104CD19FEF1900F8EA0C2A01B0030112AD9160210045084C85287A18063F11398B0C0001900D00FAFA0FA50FA50012A7680008F0200334303000000000BFAAD961 As String:5�j �} Lџ� ��* � ��`! E L�(z ? 9� � �� � � �h � 400 ���a

 Unrecognized UDT Short Data Format: UNKNOWN - Please send this to the sdrtrunk developer - Hex:0021C9016807D6099040B00442001A96A9FE579759F5DAC8 As String: !� h � �@� B ���W�Y���

Unrecognized UDT Short Data Format: UNKNOWN - Please send this to the sdrtrunk developer - Hex:025F0C0000245401112000008A8A45000857C6A74FA50FA50019CDE80006E14727AEE000A77202BD68AC54002685E000000A4B4D2DFC As String: _ $T ��E WƧO� � �� �G'�� �r �h�T &��,

Unrecognized UDT Short Data Format: UNKNOWN - Please send this to the sdrtrunk developer - Hex:04948E4E00003F11D97F0C0000010D00FAFA073A0FA50FA50010DB300006F02001310000097A00000000031D0E9800054AF5E01C8F42 As String: ��N ? �  �� : � � �0 � 1 z � J�� �B

infecticide, 2024-01-18:
Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:252136805A13AB21108583F74A40DC2D5A9086BD163085490CBFF33BAF7B4A28AC2D5A8C56542219A5490CBFF33BAF7B9A84DC2D32CC56542219A549DD92C7E9BA0591126B089E0B9640123E70BA As String:%!6�Z �! ���J@�-Z��� 0�I��;�{J(�-Z�VT" �I��;�{���-2�VT" �Iݒ�� � k � �@ >p�

 Unrecognized UDT Short Data Format: VENDOR SPECIFIC-8 - Please send this to the sdrtrunk developer - Hex:2E84DE8A0817ECA9112A06A14511CFCA3E2C As String:.�ފ � * �E ��>,

Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:0C60BA23FEAD92805A75389869042D550D27DB1CC6CA2C0FBFAA33324BD8BD7EB170BD0BD0B94388875210057AA67F0C56542219A5490CBFF33BAFA310057AA67F0C56542219A549DDEFF33BAFA3 As String:`�#����Zu8�i -U '� ��, ��32Kؽ~�p� йC��R z� VT" �I��;�� z� VT" �I���;��

Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:0C60BA23FEED12805A7539980860BA23FEAD92805E75388810057AA67F0C56542219A549DDEFF33BAFA310057AA5330C56542219A54973FA5273AFA3C4EF3170DA029A2D932FD2006ECCF086698910057AA67F0C56542219A549DDEFF33BAFA3 As String:`�#�� �Zu9� `�#����^u8� z� VT" �I���;�� z�3VT" �Is�Rs����1p� �-�/� n���i� z� VT" �I���;��

 Unrecognized UDT Short Data Format: VENDOR SPECIFIC-8 - Please send this to the sdrtrunk developer - Hex:2E84DE8A0817ECA9112A06A14511CFCA3E2C53350803581480A9F37C22A2CA85603013DC As String:.�ފ � * �E ��>,S5 X ���|"�ʅ`0 �

Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:030000015147D35E1BB590E75F0A666C00085600D41DA839 As String: QG�^ ���_,

Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:EA29FAAA5A3886BC6219A5490CB787E9C8BBEA29FAAA5A3886BCB219A54973FA8DB3A0DD4A29EC2D323886BC6219A5490CBFF338E2A94A29FAA45A8C56542219A549DD92C7E9C723 As String:�)��Z8��b �I���Ȼ�)��Z8��� �Is�����J)�-28��b �I��8�J)��Z�VT" �Iݒ���#

Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:00240C02E402F00AD05AE002E1412014B9FE57D75DF5D0E600240881A002900CC06CA079A1282014B9FE57D75DF5D0E000240C81B002900CC06CA079A1282016B8FE57D75DF5D06400241C02F402F00AD05AE002E1412016B8FE57D75DF5D0E6 As String: $ � �,

Unrecognized UDT Short Data Format: UNKNOWN - Please send this to the sdrtrunk developer - Hex:3342D5FD5FD5FD5FD5FD0BD7EAA7E4595919 As String:3B��_��_�� ���YY

 Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:4A2F7AB75A8C56542219A549DDEFF33BAFA34A10DC2D5A8C56542219A5490CBFF33BAEA9 As String:J/z�Z�VT" �I���;��J �-Z�VT" �I��;��

Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:14E309020805A60E60549429B69BE296A9FE57D75DF5DACA4A40DC2D5A9086BD163085490CBFF33BAF7B4A28AC2D5A8C56542219A5490CBFF33BAEA94A29EC2D5A8C56542219A5490CBFF33BAF7B9A84DC2D5A8C56542219A5490CBFF33BAEA9A54094F082204E080301458B9A84DC2D32CC56542219A549DDEFF33BAFA30C60BA23FEED92805E7538987BA462FD54066D2AA3C2D0431B0203BD67C9 As String: � � `T�)��▩�W�]���J@�-Z��� 0�I��;�{J(�-Z�VT" �I��;��J)�-Z�VT" �I��;�{���-Z�VT" �I��;���@��� N E����-2�VT" �I���;��`�#�풀^u8�{�b�T m*���C �g�

infecticide, 2024-01-24:
Unrecognized UDT Short Data Format: VENDOR SPECIFIC-8 - Please send this to the sdrtrunk developer - Hex:3E50AE7AE5064F08C40E02845040140A3FAB3E50AE7AE5064D15F0941FB0D704940A3FAB51FAAE3AE5049C45F09441306A04940A3FAB51F0AE3AE5049C45F09441045040140A3FAB As String:>P�z� O � �P@,

Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:0C60BA23FEAD92805E753888C4DBC0BF970C0B009D6D8772073038395B510C60BA23FEED12805A753998A90C00401B13D6091682F0A04A2F7AA5393886BDA219A54973FA5273AFA3 As String:`�#����^u8������ �m�r 089[Q`�#�� �Zu9�� @ � ��J/z�98��� �Is�Rs��

 Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:00D1030000015147C15993B59EF076015B6C035D0002566D000000000000000066315D35 As String: � QG�Y����v [l ] Vm f1]5

Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:D9130C6C000556003B309634 As String:� l V ;0�4

Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:03F531D2690FC9A54670EEB16AEF212963CE As String: �1�i ɥFp�j�!)c�

Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:6045471490384A5EE6CE90B503001909DD13820114408BA54A2F7AB75A8C56542219A549DDEFF33BAFA34A28AC2D5A8C5654221A05490CBFF33BAEA99A84DB77779E56542219A5490CBFF33BAEA90860BA23FEAD92815E7538880C60BA23FEED92805E7538880860BA23FEAD12815E7538880C60BA23FEAD92805E7538984A29FAA7DA8C56542219A5490CBFF33BAF7B0C60BA23FEED92805A7538980C60BA23FEED92805E7538980C60BA23FEED92805A7538980C60BA23FEED12805A7539986D8810244411B4011444501710057AA539B886BDA219A87D47FA733BAFA30C60BA23FEAD92805A753888 As String:`EG �8J^�ΐ� � � @��J/z�Z�VT" �I���;��J(�-Z�VT" I��;�����ww�VT" �I��;�� `�#����^u8�`�#�풀^u8� `�#�� �^u8�`�#����^u8�J)��ڌVT" �I��;�{`�#�풀Zu8�`�#�풀^u8�`�#�풀Zu8�`�#�� �Zu9�m� $D � DP z�9���� �}G�s;��`�#����Zu8�

 Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:F034DC2D33B886BC6219A5490CB2C7E9B723F004DC2D33B886BC6219A5490CB2C7E9B723A045D3C733A40B07ED38EEA00E5892EA68C9 As String:�4�-3���b �I���#� �-3���b �I���#�E��3� �8� X��h�

Unrecognized UDT Short Data Format: UNKNOWN - Please send this to the sdrtrunk developer - Hex:AA2CC4142207C06487317EAB As String:�,� " �d�1~�

Unrecognized UDT Short Data Format: VENDOR SPECIFIC-9 - Please send this to the sdrtrunk developer - Hex:00F34500003373ED00004011F2C70C00000B03720D00FAFA0FA10FA1001F84C80D15220304B30000015147C20EA4B5A241C705106C00076E0156570000000000000000006C198287 As String: �E 3s� @ �� r �� � � �� " � QG� ���A� l n VW l ��

Unrecognized UDT Short Data Format: VENDOR SPECIFIC-8 - Please send this to the sdrtrunk developer - Hex:23883832EE15B4A820C36BA9514038175B8C As String:#�82� �� �k�Q@8 [�

Unrecognized UDT Short Data Format: UNKNOWN - Please send this to the sdrtrunk developer - Hex:D43109509428011E7A412887 As String:�1 P�( zA(�

Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:81DC98182F401680720025003A0C51369DFF57D74D75DCAC6E37FDBE8855E1976F1F91D2A901CE579A00FF9D15FD53B2A669A1F7DE7433F25B6DDF50 As String:�ܘ /@ �r % :Q6��W�Muܬn7���U�o �ҩ �W� �� �S��i���t3�[m�P

Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:030000015147C1B06FB5A23B430D4C6C0005560066039C0E As String: QG��o��;C Ll V f �

Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:030000015147AB8B29B539B6D402196C1F775625E651D25D As String: QG��)�9�� l wV%�Q�]

Unrecognized UDT Short Data Format: VENDOR SPECIFIC-8 - Please send this to the sdrtrunk developer - Hex:375C6D304B95A86A5129AC0888A86E902287 As String:7\m0K��jQ)� ��n�"�

infecticide, 2024-01-24:
Unrecognized UDT Short Data Format: VENDOR SPECIFIC-8 - Please send this to the sdrtrunk developer - Hex:23BF19E0F50131A822C02304C0A86D1756BB As String:#� �� 1�"�# ��m V�

Unrecognized UDT Short Data Format: VENDOR SPECIFIC-8 - Please send this to the sdrtrunk developer - Hex:21390C89AF07EEE15FC14A0E81408F305BAC As String:!9�� ��_�J �@�0[�

Unrecognized UDT Short Data Format: UNKNOWN - Please send this to the sdrtrunk developer - Hex:048ADF007B6B4A8E2485EE716A5CA21C5102FCD824555FB1D0E98E8436F1D0BD4CAAAE5D As String: �� {kJ�$��qj\� Q ��$U_��鎄6�нL��]

infecticide, 2024-01-25:
Unrecognized UDT Short Data Format: VENDOR SPECIFIC-8 - Please send this to the sdrtrunk developer - Hex:23883832EE15B4A820C36BA9514038175B8CAAF4806DB6380981D524960433F22FBCB7D048C0947943E65E28257C22A08040016A06A1DABFCC6A0814ED4170C364A0984020CA3284ADC719BF707185FC7FF2DBED7FA4FFEBA200A828A20A27AB375C6D304B95A86A5129AC0888A86E90228767659B6DB3BF08A55557DB0AA4557F6DDF40 As String:#�82� �� �k�Q@8 [���m�8 ��$� 3�/���H��yC�^(%|"��@ j �ڿ�j �Ap�d��@ �2��� �pq�� ��� ��� �(�, 

 Unrecognized UDT Short Data Format: VENDOR SPECIFIC-8 - Please send this to the sdrtrunk developer - Hex:23883832EE15B4A820C36BA9514038175B8C As String:#�82� �� �k�Q@8 [�

Unrecognized UDT Short Data Format: VENDOR SPECIFIC-8 - Please send this to the sdrtrunk developer - Hex:53506EA9AB80C841F0460BA959A9320A3B8C535D602CC300C845F0460BA959A9326A3B8C53506EA9AB80C841F0460BA959A9320A3B8C53506EA9AB80C841F0460BA959A9320A3B8C825C5AE368171A84222B6507CAEDA0CA069923883832EE15B4A820C36BA9514038175B8C As String:SPn����A�F �Y�2,

 Unrecognized UDT Short Data Format: VENDOR SPECIFIC-8 - Please send this to the sdrtrunk developer - Hex:9908EE822175744354199941151AD68742C1482E0C234111B42EA309F97A143D09C473B800000000003F4744A57B737B As String:� �!utCT �A ևB�H.#A �.� �z = �s� ?GD�{s{

Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:030000015147BFA176B5BFAFB303006C165E5686AC9E68A3 As String: QG��v���� l ^V���h�

DSheirer, 2024-01-28:
The log said to send this to the developer.

I'm listening to a single channel DMR frequency for the highways department in my area and 99 percent of the traffic is GPS pings for their salt trucks:

2024-01-13 10:40:22.576 WARN  i.g.d.m.d.d.m.d.p.UDTShortMessageService - Unrecognized UDT Short Data Format: NMEA GPS CODED - Please send this to the sdrtrunk developer - Hex:0C0000220D0001F40FA50FA500119F110007F02002333400400D9992323208060∩┐╜∩┐╜2♠╜☼∩┐╜ ◄∩┐╜◄ ∩┐╜ ☻34 @
!:∩┐╜  [1GB/1GB 53%]

infecticide, 2024-02-08:
Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:CB93C16733A407095436BC64A87B142D8049 As String:˓�g3� T6�d�{ -�I

Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:0A214100FC16D61B904E508BE05B1016A9FE57D75DF5DAC8 As String:,

Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:82780D63AEE2110B4288A20C00149204ADFF57D75DF1C0C3 As String:�x c�� B�� � ��W�]���

 Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:BB994AFE31D9A3B259A2CAED82294D22520D6813E015002880BB1014B9FE57D75DF5D0E6A84842FD59FCB473555A740315166D2F2479A892C7FD71A40701EE864CA0E18030ABCEC9C4436AFD5393A5BAAF02DD01325DB639DBD0 As String:��J�1٣�Y���)M"R h � (�� ��W�]���HB�Y��sUZt m/$y����q� �L��0����Cj�S���� � 2]�9��

Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:C92800219A12DE00DE8293ECB52036241B6BAACE14456A3AA8C5D330E058BD26716717A7E8545BAAD269 As String:�( !� � ނ�� 6$ k�� Ej:���0�X�&qg ��T[��i

Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:011A1905180B82021E400236 As String: � @ 6 

Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:9A84DC2D32CC56542219A5490CBFF33BAFA3 As String:���-2�VT" �I��;�� 

 Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:C912EAFF17A4C1BAAAA80BFFA838AF7A6788 As String:� �� ����� ��8�zg�

Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:AB15C7FD71A40700824A10A24D8406D00E1920A44314F1384E0473F582A44A29FAA45A8C56542219A549DD92C7E9BA050038190B0004BE283816F84C310322068DFF57D75DB1C2EA18B6E63A17A16053A7EB170BD0B8D06F73C7A29BD2FD62C0A74FD5BD53A7EA9374353061 As String:� ��q� �J �M� � �C �8N s���J)��Z�VT" �Iݒ�� 8 �(8 �L1 " ��W�]��� ��: �`S�� и�osǢ���b��OսS��t50a

Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:A1B203605A070B4F78127B109021FC0120C9C7300336725C2C5EA970B38B38B393D8C3D14A29FAA5323886BD163085490CBFF33BAFA3 As String:�� `Z Ox { �!� ��0 6r\,^�p��8�����J)��28�� 0�I��;��

Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:00240C02E402F00AD05AE002E1412006B8FE57D75DF5D0E6 As String: $ � �,

Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:C51828CC9A7B8891B31211E54A29FAB75A8C56542219A86BDD92C7E9C723EA29FAB75A8C56542219A549DDEFF33BAFA3 As String:� (̚{��� �J)��Z�VT" �kݒ���#�)��Z�VT" �I���;��

Unrecognized UDT Short Data Format: UNKNOWN - Please send this to the sdrtrunk developer - Hex:16AFA7CBF7520F6E3C0ADB1A52E094794BB0D7282597613B8040016A06A1 As String: ����R n<,

infecticide, 2024-10-03:
As you guys can see, I get a lot of these (200 over the past 30 days).   Is there some automated way I can get these to the devs without manually posting them here?

Do you even still want them?