DSheirer/sdrtrunk #1767: DMR UDT Short Data Packets - Unrecognized Formats

szechyjs, 2023-12-15:
**sdrtrunk Version**
`nightly`

**Describe the bug**
I've been monitoring a new DMR Tier-III large network. Here are all the unrecognized UDT short data packets from the logs...
```
Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:9DB1C03B2718A9B105802AB81C93C03B3750A9B105803AB87E844B04479CA12E70438192F847389EADFF57C75DF1C6D87B9C3C0446DCA12E78438192F847309CFDFF57F55C64D3081D10C43AE518A8B105803AB819300108F68C1D00EA0A29F1
Unrecognized UDT Short Data Format: UNKNOWN - Please send this to the sdrtrunk developer - Hex:9AE30F0104E2296B4C12C006BE46A89EFDFF57D25DF4934AEA45B8151C7284B1A10B0419E15AA62AE8FF53D75DF4D798E85528451C3284B1C10E045AE11AA64AA96F57D75DF5D6D81C90C03B6608A9B10D803AB876DA80BFFC4E08D5099C5E9BDC1555BD12F80CB380393714A1B10BC0B8B87111443A39C099800170B90EC511C03937D1A1B105C0BEA81C11C0392608A9B101003AB8
Unrecognized UDT Short Data Format: UNKNOWN - Please send this to the sdrtrunk developer - Hex:9C01C831FAA8A12100506DD8D4781839867AB037010FB606C945420AA96E57D75DF5D6D08103443D3A80BBA102F3F4319C91C0323712A9B10DC0B8A81D11443B261AA99105903AB09D10C83A6218B9A100003BBC1809C83BF8B8A1B106406CB80079D9031C99286BE946881BF149650CB9FFD7D75DF5D69B0C01C82BB828A0B30C4078F81D91C039E708A8B104003AB04490C03A3714A1B005403BB977338B36D84FCF990E59737A1E15D68612F81C93C43B3714A0B104C03AB8
Unrecognized UDT Short Data Format: MS OR TG ADDRESS - Please send this to the sdrtrunk developer - Hex:FA112C111E72A4B9C14B041BF912320EFDFF55974DF5D6989D10403B6718A9B104803EB8
Unrecognized UDT Short Data Format: UNKNOWN - Please send this to the sdrtrunk developer - Hex:0883C8337828A1B1084078F0E84438110C3384B1C10E064BF11AA66AB8FE57D75DF5D798
Unrecognized UDT Short Data Format: UNKNOWN - Please send this to the sdrtrunk developer - Hex:5501C83B6618A9B105803FB8F010047870C010A001709C8271EE4937F0EA43BB0E2F1AC70133238736E0
Unrecognized UDT Short Data Format: UNKNOWN - Please send this to the sdrtrunk developer - Hex:1D81C4396700A9B1040038B8FA112C111B328439A10F241B791AB60EB8F757D74DF5D789FA112C111E7A84A1A00F064BE11AA66AE8FE57F75DF5D7981491842B2310A0B105803AB819000809E00AB5B002402D92
Unrecognized UDT Short Data Format: MS OR TG ADDRESS - Please send this to the sdrtrunk developer - Hex:66844B04479A292E3853C102FC46A99AECFF57C75DF3C6DAFA112C100368A539204B0612F94A360EA9FF57D75DF1D69854934C3B6318A8B005909AB87A412C111B78A4B9A10B041BE91AB20AA9FF57D75DF586D83C01C039E808A0B1844068C06EDA2E0444DAA16B6852C104BE54CDAAACBD57D77DB5969800328C09867AB0A6812FA004E945100AEDFF57D75DD1D7D00493842B2754A1B105C0B2A8
Unrecognized UDT Short Data Format: UNKNOWN - Please send this to the sdrtrunk developer - Hex:05820C1831809BA00DB0C965DC11C83A3718A8B105003AB80D13C43B3710A9B10580BAB85C10C03B2618A9B104103AA0400B4C393980BAA386F1B2BB1EE30F0151C7284F4D10E106FA46A89ABDFF57D35DE4938A1C01C839B8A8A1B38E4078C0FA113C044C2284B1810E064BE11AA66AE8FE57975DF5D79800B140F6F84F63982A41134598A9259F8BD404B3843B3710A1B105C0BAA8E42E19920084C1286008860F8094840CB9FF53D75DF5D69999100900F68C3D0000002BF1
Unrecognized UDT Short Data Format: UNKNOWN - Please send this to the sdrtrunk developer - Hex:DD300018628DAD9104003AB09C11C43B3610A1B105103BB8DC11803B3710A1B106803AA81FA112C111C7284B0C12C006BE46A88AAC9F17D75DF5D7986A4438450D2284B4C10E064BE11AA24AB8FE57975DF5D6981C01C8293008A1B100406CE85D81C03B2318A8B10D003AB821001C0831809BA04DB8496590304918B2F5D92B9156F046A08AC40AECFE57D359F5D7D09D00403A628AA9110410BAB0
Unrecognized UDT Short Data Format: UNKNOWN - Please send this to the sdrtrunk developer - Hex:9030419A00E6C1A82008826B26148E6AA8FE57D65DF5D7900801C833B808A0B28E4068D8
Unrecognized UDT Short Data Format: UNKNOWN - Please send this to the sdrtrunk developer - Hex:218EDE3829C08B900DB91B612D024E383A809BA225B84A6501890C0970A8B9A106F394A971EE4937F0F819FF3A0CBE73C93323C07BC89511CC316290B98100803BB81C31E4386E19A9B12D801EB8EE30F0111F2284B9C10E065BE11AB62AE8FE55D75DF5D798180188307828A0B1844078807011447831C0�-N8:���%�Je�C11C0382708A8B101103AB8
Unrecognized UDT Short Data Format: UNKNOWN - Please send this to the sdrtrunk developer - Hex:1C03C8393838B1B10C407CE00C11C03B369AA9B10D003AB81801C8333838B0B10C0028980106DE3C3BC01F800DB91B65
Unrecognized UDT Short Data Format: UNKNOWN - Please send this to the sdrtrunk developer - Hex:2102543838809F802DB04865BE944E145328C1B920490413F912B60AEDFF53F74DF5D7D00036899E6861292BA106201FB1090C0AA9FED7D75DF7DE93FA5128114E7284B1C10E064BE11AA60AB9FE57975DF5C6981101C838E88AB1A10C506D89
Unrecognized UDT Short Data Format: UNKNOWN - Please send this to the sdrtrunk developer - Hex:F1ABEFB0BB2A66B98E9C527342130FE87BE0
Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:9C10C4392318F8B187903EA8
Unrecognized UDT Short Data Format: UNKNOWN - Please send this to the sdrtrunk developer - Hex:007A9C19863AB4A7012F2406E945420AA9FE57D655F5D7D821824E383A809FA00DB068650D114439660EA9A105903FB95C11C0396619A9B104D03AB8
```

DSheirer, 2023-12-19:
file:///home/denny/Downloads/sdrtrunk.png

DSheirer, 2023-12-19:
[image]

DSheirer, 2023-12-21:
20231219 171825.433 [sdrtrunk polyphase channel thread 1] WARN i.g.d.m.d.d.m.d.p.UDTShortMessageService - Unrecognized UDT Short Data Format: UNKNOWN - Please send this to the sdrtrunk developer - Hex:152CA7400157516DFA3BD997B73B73936E47 As String: ,�@ WQm�;ٗ�;s�nG [339MB/2GB 16%]

20231219 154518.536 [sdrtrunk polyphase channel thread 1] WARN i.g.d.m.d.d.m.d.p.UDTShortMessageService - Unrecognized UDT Short Data Format: BINARY - Please send this to the sdrtrunk developer - Hex:030000015147C5F8EBB5A48A As String: QG��뵤� [242MB/2GB 11%]

20231218 214641.058 [sdrtrunk heterodyne channel 460550000 thread 1] WARN i.g.d.m.d.d.m.d.p.UDTShortMessageService - Unrecognized UDT Short Data Format: UNKNOWN - Please send this to the sdrtrunk developer - Hex:030000015147C20AEFB5A244780F006C0002560039E6ADFB As String: QG� 
 
20231218 115134.223 [sdrtrunk heterodyne channel 460550000 thread 1] WARN i.g.d.m.d.d.m.d.p.UDTShortMessageService - Unrecognized UDT Short Data Format: UNKNOWN - Please send this to the sdrtrunk developer - Hex:030000015107C1B72FB58B66 As String: Q ��/��f [1GB/2GB 52%]

2023-12-18 11:51:34.223 WARN i.g.d.m.d.d.m.d.p.UDTShortMessageService - Unrecognized UDT Short Data Format: UNKNOWN - Please send this to the sdrtrunk developer - Hex:030000015107C1B72FB58B66 As String: Q ��/��f [1GB/2GB 52%]