DSheirer/sdrtrunk #1774: GPS Map units text issue and Play buttons and TG hold

Soupbones, 2023-12-23:
Merry Christmas to you all!
 I just recently obtained all the LCNs for my local DMR trunk system and have been playing with it. I see it has GPS data and this comes up on the map great, but the text showing the units number is in a very hard to read yellow and a thin font. Can this be changed to something that contrasts the map colours and a bolder font to make it easier to read? Maybe add this in as a setting somewhere on the map screen?
 Another thing that bugs me is when you say start up multiple channels and for some reason you need to shut down one, say for bad reception that day as its not decoding. Its a pain to have to find the playlist editor again and find the channel, open it and hit stop. Can there be a way to stop and start a channel on the Now playing screen? Maybe right click then a menu with this option?
 While I was thinking of this rigth click idea maybe a right click could be added in, say when you click on an active TG it pops up a Hold option, and then the SDR trunk could hold that TGs and the audio till you unhold it with a right click again? Other things likemrecording of other activity can still happen while holding a TG. Maybe for Radio IDs as well. I though this after trying to listen and write down TGs on this PMR system I have started listenin too. Would be easy to nutt out the TGs and who is using them is you could hold on them.

DSheirer, 2023-12-23:
Hi @Soupbones, please use the supplied template(s) for new issues and please use the search button to see if someone has already asked for the same feature(s).

Generally, when you create an issue for either a bug or a feature-request, you limit the scope to just one item.  That said, I'll address your (user questions) and (feature request):

> I just recently obtained all the LCNs for my local DMR trunk system and have been playing with it. I see it has GPS data and this comes up on the map great, but the text showing the units number is [...]

The map plotting functionality is very basic, for now.  When/if I have time, I'll work on enhancing this feature.  

In the mean-time, you can alias the FROM radio ID value and assign a color to the alias and the map will use that color.  If the radio is not aliased, it will default to yellow.

> Another thing that bugs me is when you say start up multiple channels and for some reason you need to shut down one, say for bad reception that day as its not decoding. Its a pain to have to find [...]

Have you tried right-clicking on the channel row in the Now Playing tab?

> While I was thinking of this rigth click idea maybe a right click could be added in, say when you click on an active TG it pops up a Hold option, and then the SDR trunk could hold that TGs and the [...]

There's already several open-issues requesting this feature.  I'm working on  #1723 to address this.

Closing this issue.

closed 2023-12-23 by DSheirer

Soupbones, 2023-12-24:
Hi @DSheirer.
OK i'm fairly new to this Githu stuff. I am not sure if this was a bug or a suggestion.. Or a feature request. Probably that.
OK changing a Unit ID colour doesnt change it on the map. Is there a way to simply change the tex colour on the map?
the yellow text is pretty unreadable on that map.
 Have a good Christmas.