DSheirer/sdrtrunk #1775: Support for SDRplay RSPdx

bi3mw, 2023-12-23:
Hello,
can you implement support for SDRplay RSPdx ? Thank you very much.

DSheirer, 2023-12-24:
The RSPdx is supported in code, but i don't have that model to test against.

Have you tried using the RSPdx to see if the software works correctly?

bi3mw, 2023-12-24:
Yes, I have connected the RSPdx. The API and libusb are installed. However, the device is not recognized (Ubuntu 22.04 ). SDRconnect works perfectly.

mivok, 2024-01-01:
FWIW, my SDRPlay RSPDx works fine with sdrtrunk (I've tested on linux and mac). I do find that I have to restart the api service sometimes (`sudo systemctl restart sdrplay.service` on linux) and I need to have the RSPdx connected before I start sdrtrunk.

bi3mw, 2024-01-01:
Yes, restarting the API service did the trick. Now sdrtrunk recognizes the device. Thanks for the hint.

However, the sound output is muted. Is that intentional?

closed 2024-01-21 by DSheirer

MNolan5, 2024-02-03:
Hello.  
I am having the same problem with my RSPdx.  SDR Trunk 0.6.0 doesn't see my SDR (runs great using other SW).
[image]

From the above, the solution appears to be restarting API.
Apologies for the possibly very stupid question, but...

How do I restart API (Windows 10)?

DSheirer, 2024-02-03:
@MNolan5 you're using the latest API (3.14) which is not yet supported by the release version.

If you'd like to test with the 'nightly' release, I just pushed in changes to support API 3.14 and the new RSP1B here: https://github.com/DSheirer/sdrtrunk/releases/tag/nightly

MNolan5, 2024-02-03:
Fixed.
Thank you!