DSheirer/sdrtrunk #1778: Perhaps the protocol is not complete

ganjaden, 2024-01-06:
[image]

WARNING: Using incubator modules: jdk.incubator.vector
2024-01-07 00:47:20.395 INFO  i.g.d.icon.IconModel - loading icons file [C:\Users\╨┤╨╡╨╜╨╕╤Б\SDRTrunk\settings\icons.xml]  [15MB/256MB 6%]
2024-01-07 00:47:20.929 INFO  i.g.d.log.ApplicationLog - Application Log File: C:\Users\╨┤╨╡╨╜╨╕╤Б\SDRTrunk\logs\sdrtrunk_app.log  [12MB/34MB 35%]
2024-01-07 00:47:20.967 INFO  i.g.d.log.ApplicationLog - SDRTrunk Version  : 0.6.0  [14MB/34MB 43%]
2024-01-07 00:47:20.968 INFO  i.g.d.log.ApplicationLog - Gradle Version    : Gradle 8.2  [14MB/34MB 43%]
2024-01-07 00:47:20.968 INFO  i.g.d.log.ApplicationLog - Build Timestamp   : 2023-12-16T06:54:18.883-0500  [14MB/34MB 43%]
2024-01-07 00:47:20.968 INFO  i.g.d.log.ApplicationLog - Build-JDK         : 20.0.1 (BellSoft 20.0.1+10  [14MB/34MB 43%]
2024-01-07 00:47:20.968 INFO  i.g.d.log.ApplicationLog - Build OS          : Linux (amd64 6.2.0-37-generic  [14MB/34MB 43%]
2024-01-07 00:47:20.968 INFO  i.g.d.log.ApplicationLog -   [14MB/34MB 43%]
2024-01-07 00:47:20.968 INFO  i.g.d.log.ApplicationLog - *******************************************************************  [14MB/34MB 43%]
2024-01-07 00:47:20.968 INFO  i.g.d.log.ApplicationLog - **** sdrtrunk: a trunked radio and digital decoding application ***  [14MB/34MB 43%]
2024-01-07 00:47:20.968 INFO  i.g.d.log.ApplicationLog - ****  website: https://github.com/dsheirer/sdrtrunk             ***  [14MB/34MB 43%]
2024-01-07 00:47:20.969 INFO  i.g.d.log.ApplicationLog - *******************************************************************  [14MB/34MB 43%]
2024-01-07 00:47:20.969 INFO  i.g.d.log.ApplicationLog - Memory Logging Format: [Used/Allocated PercentUsed%]  [14MB/34MB 43%]
2024-01-07 00:47:20.969 INFO  i.g.d.log.ApplicationLog - Host OS Name:          Windows 10  [14MB/34MB 43%]
2024-01-07 00:47:20.969 INFO  i.g.d.log.ApplicationLog - Host OS Arch:          amd64  [14MB/34MB 43%]
2024-01-07 00:47:20.970 INFO  i.g.d.log.ApplicationLog - Host OS Version:       10.0  [14MB/34MB 43%]
2024-01-07 00:47:20.970 INFO  i.g.d.log.ApplicationLog - Host CPU Cores:        12  [14MB/34MB 43%]
2024-01-07 00:47:20.970 INFO  i.g.d.log.ApplicationLog - Host Max Java Memory:  3 GB  [14MB/34MB 43%]
2024-01-07 00:47:20.970 INFO  i.g.d.log.ApplicationLog - Storage Directories:  [14MB/34MB 43%]
2024-01-07 00:47:20.971 INFO  i.g.d.log.ApplicationLog -  Application Root: C:\Users\╨┤╨╡╨╜╨╕╤Б\SDRTrunk  [14MB/34MB 43%]
2024-01-07 00:47:20.971 INFO  i.g.d.log.ApplicationLog -  Application Log:  C:\Users\╨┤╨╡╨╜╨╕╤Б\SDRTrunk\logs  [14MB/34MB 43%]
2024-01-07 00:47:20.971 INFO  i.g.d.log.ApplicationLog -  Event Log:        C:\Users\╨┤╨╡╨╜╨╕╤Б\SDRTrunk\event_logs  [14MB/34MB 43%]
2024-01-07 00:47:20.972 INFO  i.g.d.log.ApplicationLog -  Playlist:         C:\Users\╨┤╨╡╨╜╨╕╤Б\SDRTrunk\playlist  [14MB/34MB 43%]
2024-01-07 00:47:20.973 INFO  i.g.d.log.ApplicationLog -  Recordings:       C:\Users\╨┤╨╡╨╜╨╕╤Б\SDRTrunk\recordings  [14MB/34MB 43%]
2024-01-07 00:47:20.986 INFO  i.g.d.util.ThreadPool - Application thread pool created SCHEDULED and CACHED executors threads  [15MB/34MB 45%]
2024-01-07 00:47:20.994 INFO  i.g.d.p.SystemProperties - SystemProperties - loaded [C:\Users\╨┤╨╡╨╜╨╕╤Б\SDRTrunk\SDRTrunk.properties]  [15MB/34MB 45%]
2024-01-07 00:47:21.046 INFO  i.g.d.s.t.m.TunerManager - Discovering tuners ...  [18MB/34MB 53%]
2024-01-07 00:47:21.145 INFO  i.g.d.s.t.m.TunerManager - LibUsb API Version: 1.0.262  [18MB/34MB 55%]
2024-01-07 00:47:21.146 INFO  i.g.d.s.t.m.TunerManager - LibUsb Version: 1.0.22.11312  [18MB/34MB 55%]
2024-01-07 00:47:21.162 INFO  i.g.d.s.t.m.TunerManager - LibUsb - discovered [7] potential usb devices  [18MB/34MB 55%]
2024-01-07 00:47:21.167 INFO  i.g.d.s.t.m.TunerManager - Discovered tuner at USB Bus [2] Port [8] Tuner Class [RTL-2832]  [19MB/34MB 56%]
2024-01-07 00:47:21.170 INFO  i.g.d.s.t.m.TunerManager - Tuner: USB Tuner - RTL-2832 USB Bus:2 Port:8 - Added / Starting ...  [19MB/34MB 56%]
2024-01-07 00:47:22.869 INFO  i.g.d.s.t.r.r.R8xEmbeddedTuner - PLL is not locked.  Increasing VCO current  [23MB/34MB 67%]
2024-01-07 00:47:22.880 INFO  i.g.d.s.t.m.DiscoveredTuner - Tuner Error - Stopping - RTL-2832 USB Bus:2 Port:8 Error: Unable to initialize R828D  [23MB/34MB 67%]
2024-01-07 00:47:22.881 INFO  i.g.d.s.t.m.DiscoveredTuner - Stopping Tuner: RTL-2832 USB Bus:2 Port:8  [23MB/34MB 67%]
2024-01-07 00:47:22.882 ERROR i.g.d.s.t.m.DiscoveredUSBTuner - Unable to start tuner [RTL-2832] - error: Unable to initialize R828D  [23MB/34MB 67%]
2024-01-07 00:47:22.884 INFO  i.g.d.s.t.m.TunerManager - LibUsb Hotplug event notification Is Not Supported on this platform.  [23MB/34MB 69%]
2024-01-07 00:47:22.892 INFO  i.g.d.s.t.s.api.SDRplay - Loading API Library from default install path: C:\Program Files\SDRplay\API\x64\sdrplay_api  [24MB/34MB 71%]
2024-01-07 00:47:22.893 ERROR i.g.d.s.t.s.api.SDRplay - Unable to load SDRplay API library from default install path.  Loading from java system library path  [24MB/34MB 71%]
2024-01-07 00:47:22.894 WARN  i.g.d.s.t.s.api.SDRplay - SDRPlay API library not found/installed on this system.  Ensure the API is installed either in the default install location or the install location is included in the 'java.library.path' JVM property contains path to the library file [sdrplay_api.dll].  Current library path property contents: c:\Program Files\SDRplay\API\x64  [24MB/34MB 71%]
2024-01-07 00:47:22.894 INFO  i.g.d.s.t.s.api.SDRplay - API library was not loaded  [24MB/34MB 71%]
2024-01-07 00:47:22.894 INFO  i.g.d.s.t.s.api.SDRplay - API library is not available.  [24MB/34MB 71%]
2024-01-07 00:47:22.897 INFO  i.g.d.s.SettingsManager - SettingsManager - loading settings file [C:\Users\╨┤╨╡╨╜╨╕╤Б\SDRTrunk\settings\settings.xml]  [24MB/34MB 71%]
2024-01-07 00:47:23.220 WARN  i.g.d.v.VectorUtilities - CPU supports maximum SIMD instructions of Species[float, 8, S_256_BIT]  [13MB/38MB 35%]
2024-01-07 00:47:23.789 INFO  i.g.d.p.PlaylistManager - Loading playlist [C:\Users\╨┤╨╡╨╜╨╕╤Б\SDRTrunk\playlist\default.xml]  [24MB/38MB 65%]
2024-01-07 00:47:23.859 INFO  i.g.dsheirer.gui.SDRTrunk - starting main application gui  [27MB/38MB 73%]

bradjtrammell, 2024-01-29:
Hi there. 

Posting a log with no context or information otherwise isn't really helpful to those who monitor this to provide assistance to others, instead it requires us to dig through your log and try to piece together what we can best assume is your issue. Please make sure that when asking for help, or even just posting a log, that you're providing context. :-)

With that being said, while looking at your logs, I'm able to pick out that you have plugged in an RTL-SDR "New Gen", most likely one of these devices...

[image] 

These devices are not presently supported by SDR Trunk, however I wish they would be as I have about 10 of them that I used to use for ADS-B and now are sitting in storage. These don't use the same chipset as say the RTL-SDR Blog v3/v4, or any of the chips from Nooelec, they instead use the less popular R828D chip.

Perhaps the software's author (@DSheirer) may add support for these devices in the future. Looking at both your logs above, and logs I can collect from my own attempt at using these in SDR Trunk, SDR Trunk seems to fail during the init stage of the software launch. 

It doesn't appear to be an issue with driver support as at least on my end, I am able to use this software in SDR++, SDR#, SDR Console and a whole host of other software. Perhaps @DSheirer can look into it and provide some insight.

DSheirer, 2024-01-29:
I have one of these older R828D tuners (just recently rediscovered it in my box of tuners) that's not the same as the newer V4 Blog variant and I intend to add support for the older R828D tuners at some point.

ganjaden, 2024-01-29:
> Всем привет.
> 
> Публикация журнала без контекста или информации в противном случае не очень полезна для тех, кто отслеживает его, чтобы оказать помощь другим, вместо этого нам нужно покопаться в вашем журнале и попытаться собрать воедино то, что, как мы можем лучше всего предположить, является вашей проблемой. Пожалуйста, убедитесь, что, обращаясь за помощью или даже просто публикуя журнал, вы предоставляете контекст. :-)
> 
> При этом, просматривая ваши журналы, я могу определить, что вы подключили RTL-SDR «Нового поколения», скорее всего, одно из этих устройств...
> 
> [image]
> 
> Эти устройства в настоящее время не поддерживаются SDR Trunk, однако мне бы хотелось, чтобы они были такими, поскольку у меня есть около 10 из них, которые я использовал для ADS-B и сейчас находятся в хранилище. В них не используется тот же набор микросхем, что, скажем, в RTL-SDR Blog v3/v4 или в каких-либо чипах от Nooelec, вместо этого они используют менее популярный чип R828D.
> 
> Возможно, автор программы (@DSheirer) может добавить поддержку этих устройств в будущем. Глядя на ваши журналы выше и журналы, которые я могу собрать в результате моей собственной попытки использовать их в SDR Trunk, кажется, что SDR Trunk дает сбой на этапе инициализации запуска программного обеспечения.
> 
> Кажется, это не проблема с поддержкой драйверов, поскольку, по крайней мере, с моей стороны, я могу использовать это программное обеспечение в SDR++, SDR#, SDR Console и целом ряде других программ. Возможно@DSheirerможет изучить это и дать некоторое представление.

Yes, that is right. Sorry for not mentioning it right away.

bradjtrammell, 2024-01-29:
> I have one of these older R828D tuners (just recently rediscovered it in my box of tuners) that's not the same as the newer V4 Blog variant and I intend to add support for the older R828D tuners at some point.

Perfect. I believe we have our answer then. Thanks for the quick feedback @DSheirer

DSheirer, 2024-02-10:
Closing.  Duplicate to #227

closed 2024-02-10 by DSheirer