DSheirer/sdrtrunk #1781: RRDB Enterprise Account Import

BrendanMan3, 2024-01-06:
Hello I am running v0.6.0 and have tried to link to my RRDB account which is an enterprise account. However I receive the error that my account is not a premium RRDB account in sdrtrunk, although it is.

DSheirer, 2024-01-07:
You may want to try logging into the webpage and verify that you can browse through the frequency database.  If that's ok, then check with the radio reference folks to see why your account login doesn't reflect premium.

BrendanMan3, 2024-01-10:
My account is not Premium, it is Enterprise. So while it is also premium it also includes a few extra perks.
Enterprise and Premium should be the same for importing from the RRDB.

DSheirer, 2024-01-11:
The sdrtrunk application attempts to use the radio reference API using the login credentials that you provide.  If the API service is reporting that your account is not enabled to use the API, then that's something you'll have to resolve with the radio reference team.

Even though sdrtrunk reports the error as 'you don't have a premium account', the reality is that your login credentials are not authorized to access the API.

I don't believe this is a bug in the sdrtrunk software.

blantonl, 2024-06-27:
Dennis, can you advise how SDRTrunk is determining premium subscription status and authorization?  What API call are you using, and what attributes are you keying off of to make the determination?

This seems to be localized to SDRTrunk and I'm trying to determine why... other API consumers appears to be working ok (this person is able to use ProScan for instance just fine with his enterprise account)

DSheirer, 2024-07-24:
In discussion with @blantonl the API ```getUserDetails``` method can return the following values:

* An expiration date
* Never - Feed Provider
* Never - Admin
* Never - Enterprise

Will update sdrtrunk to accept any non-date value strings that start with 'Never' as valid expiration dates.

DSheirer, 2024-07-24:
After inspecting the code, it should be handling the optional response values for any of the 'Never - ' expiration status and allowing the user to use the RR API.

I tested each of these response values and confirmed that the code functions correctly.

@BrendanMan3 can you post a screen shot of the radio reference tab in the playlist editor?  This should display the 'Expiration Date' value that was returned from the radio reference API for your account.

closed 2024-07-29 by DSheirer

BrendanMan33, 2024-07-29:
[image]
That did the trick.

Thank you all for the help.

PR #1943 (2024-07-29): #1781 Radio Reference API Login Enhanced Logging
Closes #1781 Radio Reference API - adds logging to log results of test connection or user login attempt for troubleshooting RR enterprise user login issues.