DSheirer/sdrtrunk #1782: RSPduo spectrum expanded

ah1102, 2024-01-07:
sdrtrunk 0.6.0 release
macOS 10.15
RSPduo

I observe that RSPduo's spectrum is expanded compared to hackRF and rtl-sdr. It can be seen that adjacent P25p1 channels overlap each other. On hackRF the channels are clearly visible.

[image]
[image]

ah1102, 2024-01-07:
Or here is the NBFM channel 12.5 kHz. The spectrum goes beyond.

[image]
[image]

ah1102, 2024-09-30:
I installed Nightly available the version now and nothing has changed in the spectrum. SDRPLAY API 3.15

DSheirer, 2024-10-01:
I can't fix this.

closed 2024-10-01 by DSheirer

ah1102, 2024-10-28:
> I can't fix this.

Look. This screenshot is made today. The version of the program is the last **vnightly**. I have the impression that the width of the spectrum depends on the signal level. Very weak signals contain the width of the spectrum.

[image]