DSheirer/sdrtrunk #1783: P25P2 L3Harris Message Parsing Error

DSheirer, 2024-01-08:
2024-01-06 17:14:58.325 ERROR i.g.d.sample.Broadcaster - Error while
broadcasting [class
io.github.dsheirer.module.decode.p25.phase2.message.mac.MacMessage] to
listeners [Log Suppress 1/5]  [61MB/148MB 41%]
java.lang.IndexOutOfBoundsException: fromIndex: 192 > toIndex: 168
        at java.base/java.util.BitSet.checkRange(Unknown Source)
        at java.base/java.util.BitSet.get(Unknown Source)
        at
io.github.dsheirer.bits.BinaryMessage.get(BinaryMessage.java:317)
        at
io.github.dsheirer.bits.CorrectedBinaryMessage.getSubMessage(CorrectedBinaryMessage.java:93)
        at
io.github.dsheirer.module.decode.p25.phase2.message.mac.structure.l3harris.L3HarrisTalkerAlias.getAlias(L3HarrisTalkerAlias.java:82)
        at
io.github.dsheirer.module.decode.p25.phase2.message.mac.structure.l3harris.L3HarrisTalkerAlias.getIdentifiers(L3HarrisTalkerAlias.java:113)
        at
io.github.dsheirer.module.decode.p25.phase2.message.mac.MacMessage.getIdentifiers(MacMessage.java:131)
        at
io.github.dsheirer.alias.action.AliasActionManager.receive(AliasActionManager.java:58)
        at
io.github.dsheirer.alias.action.AliasActionManager.receive(AliasActionManager.java:43)
        at
io.github.dsheirer.sample.Broadcaster.broadcast(Broadcaster.java:128)
        at
io.github.dsheirer.sample.Broadcaster.receive(Broadcaster.java:52)
        at
io.github.dsheirer.module.decode.Decoder$MessageDistributor.receive(Decoder.java:115)
        at
io.github.dsheirer.module.decode.Decoder$MessageDistributor.receive(Decoder.java:108)
        at
io.github.dsheirer.module.decode.p25.phase2.P25P2MessageProcessor.receive(P25P2MessageProcessor.java:104)
        at
io.github.dsheirer.module.decode.p25.phase2.P25P2MessageProcessor.receive(P25P2MessageProcessor.java:42)
        at
io.github.dsheirer.module.decode.p25.phase2.P25P2SuperFrameDetector.broadcast(P25P2SuperFrameDetector.java:231)
        at
io.github.dsheirer.module.decode.p25.phase2.P25P2SuperFrameDetector.broadcastFragment(P25P2SuperFrameDetector.java:190)
        at
io.github.dsheirer.module.decode.p25.phase2.P25P2SuperFrameDetector.checkFragmentSync(P25P2SuperFrameDetector.java:255)
        at
io.github.dsheirer.module.decode.p25.phase2.P25P2SuperFrameDetector.receive(P25P2SuperFrameDetector.java:153)
        at
io.github.dsheirer.module.decode.p25.phase2.P25P2MessageFramer.receive(P25P2MessageFramer.java:173)
        at
io.github.dsheirer.module.decode.p25.phase2.P25P2MessageFramer.receive(P25P2MessageFramer.java:57)
        at
io.github.dsheirer.sample.Broadcaster.broadcast(Broadcaster.java:128)
        at
io.github.dsheirer.sample.Broadcaster.receive(Broadcaster.java:52)
        at
io.github.dsheirer.dsp.psk.PSKDemodulator.broadcast(PSKDemodulator.java:87)
        at
io.github.dsheirer.dsp.psk.DQPSKGardnerDemodulator.calculateSymbol(DQPSKGardnerDemodulator.java:88)
        at
io.github.dsheirer.dsp.psk.PSKDemodulator.receive(PSKDemodulator.java:146)
        at
io.github.dsheirer.dsp.psk.PSKDemodulator.receive(PSKDemodulator.java:121)
        at
io.github.dsheirer.module.decode.p25.phase2.P25P2DecoderHDQPSK.receive(P25P2DecoderHDQPSK.java:142)
        at
io.github.dsheirer.module.decode.p25.phase2.P25P2DecoderHDQPSK.receive(P25P2DecoderHDQPSK.java:48)
        at
io.github.dsheirer.sample.Broadcaster.broadcast(Broadcaster.java:128)
        at
io.github.dsheirer.sample.Broadcaster.receive(Broadcaster.java:52)
        at
io.github.dsheirer.dsp.filter.channelizer.PolyphaseChannelSource.receive(PolyphaseChannelSource.java:143)
        at
io.github.dsheirer.dsp.filter.channelizer.PolyphaseChannelSource.receive(PolyphaseChannelSource.java:40)
        at
io.github.dsheirer.dsp.filter.channelizer.output.TwoChannelOutputProcessor.process(TwoChannelOutputProcessor.java:124)
        at
io.github.dsheirer.dsp.filter.channelizer.output.ChannelOutputProcessor.lambda$new$0(ChannelOutputProcessor.java:57)
        at io.github.dsheirer.util.Dispatcher.process(Dispatcher.java:171)
        at
io.github.dsheirer.util.Dispatcher$ProcessorWithHeartbeat.run(Dispatcher.java:213)
        at
java.base/java.util.concurrent.Executors$RunnableAdapter.call(Unknown
Source)
        at
java.base/java.util.concurrent.FutureTask.runAndReset(Unknown Source)
        at
java.base/java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(Unknown
Source)
        at
java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
        at
java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
        at java.base/java.lang.Thread.run(Unknown Source)

closed 2024-01-10 by DSheirer

louisik1, 2024-02-01:
Hey Denny, I've seen this a couple times. Do you know what kind of side effect it would manifest as?  I'm just curious.

DSheirer, 2024-02-01:
This error is being caught, handled and logged, so there shouldn't be any side effect other than polluting the log file.

PR #1786 (2024-01-10): #1783 L3H P25P2 Talker Alias Guard Against Negative Length Aliases
Closes #1783