DSheirer/sdrtrunk #1790: OpenMHz feeder connection

jmp367, 2024-01-10:
*SDRTrunk versions:
0.6.0 and nightly 2023-12-17T12:02:28:146+0000

*Bug description:
When using 0.6.0 to feed OpenMHz, the feed periodically becomes disconnected and will not reconnect to OpenMHz automatically. SDRTrunk shows the OpenMHz feed status cycle between "Connecting" and "Error" (in red) when this issue occurs and there are no messages related to this issue in the log file.

*Steps to reproduce behavior:
1. Disconnect Ethernet cable or WiFi connection.
2. From View menu item, select option to display Streaming Status.
3. Observe OpenMHz streaming feed connection status, which should show it cycle between Connecting and Error.
4. Reconnect Ethernet cable or WiFi connection.
5. Observe OpenMHz streaming feed connection status once again, as it will only cycle between Connecting and Error, never reconnecting automatically.
6. To reconnect to the OpenMHz server, go to Playlist Editor -> Streaming tab -> select OpenMHz feed
7. In Configuration tab, toggle "Enabled" toggle off then on.
8. Check Stream Status window once again, OpenMHz feed will be reconnected.

*Expected behavior:
I expect upon a disconnection from the OpenMHz server for SDRTrunk to automatically reconnect the feed, as the software does for my Broadcastify Calls nodes when internet connectivity resumes. Additionally, logging the OpenMHz connection/disconnection status to the SDRTrunk log would be helpful.

*Application log:
N/A nothing was logged.

*Desktop
- OS: Windows 10 Professional 64-bit
- CPU Cores: 4 cores
- RAM: 16GB

tadscottsmith, 2024-01-10:
I was able to confirm there is an issue with automatic reconnection after your comment yesterday. I've identified the problem and am currently testing out the fix. Good catch!

jmp367, 2024-01-10:
Very good, thanks! Hopefully the fix can be rolled out in the next nightly build.

PR #1791 (2024-01-12): #1791 Resolves OpenMHz Broadcaster Auto-Reconnection Issue
Closes #1790
closed 2024-01-12 by DSheirer