DSheirer/sdrtrunk #1800: Recording channels only when call is active

mike6789, 2024-01-20:
Hello,

It seems that when enabling recording on a channel (I am using P25 channels), the wav file grows as long as the program is running. This is not the ideal behavior for me; I'd like it so that a new wav file is made for each time someone talks on the channel and stops when there is silence.

Is this possible to do in the current software? Or would it need to be implemented?

Thank you!
Mike

DSheirer, 2024-01-21:
You may be confusing recordings.  It sounds like you're enabling baseband recording which records the raw (undecoded) signal for the channel and this is a continuous recording.

If you're looking to record the call audio, you'll need to setup aliases and then mark the aliases for recording.

https://github.com/DSheirer/sdrtrunk/wiki/Playlist-Editor

Read the section on aliases & how-to record audio.

Can you please close this issue if you don't have a discrepancy to report.

Cheers, Denny

mike6789, 2024-01-21:
Whoops! You are totally right. Thanks so much

closed 2024-01-21 by mike6789