DSheirer/sdrtrunk #1803: Difficulties decoding DMR DMO/Talkaround and uplink from Chinese radios (TYT MD-UV380)

adryd325, 2024-01-22:
**sdrtrunk Version**
0.6.0

**Describe the bug**
When tuning to a DMR DMO/Talkaround channel transmitted by a Chinese radio, SDRTrunk struggles to decode anything. 

**To Reproduce**
1. Tune to a channel transmitted from a MD-UV380
2. Observe difficulties decoding audio

**Expected behavior**
Clear audio from the channel, 

**Actual behavior**
Decoding is spotty and voice barely comes through

FM baseband recording from sdrtrunk: https://adryd.co/public/20240122_141253_446800000_BrandMeister_UP_24_baseband.wav

adryd325, 2024-01-23:
I think my original baseband recording was messed up, so I've made some new recordings

https://adryd.co/public/20240123_000340_446800000_BrandMeister_UP_25_baseband_probably_gain_overload.wav
https://adryd.co/public/20240123_000734_446800000_BrandMeister_UP_25_baseband_reduced_gain.wav

DSheirer, 2024-01-23:
I'm working on an enhancement to the DMR decoder that incorporates an equalizer and I think that needs to be completed before I investigate this, because any enhancements for this issue would be better suited on top of the DMR decoder enhancements.

lwvmobile, 2024-01-24:
Just for what's its worth, I was only able to decode the very first sample submitted in DSD-FME, so if you do any work on this Denny, you might just focus your effort on the first sample named:

20240122_141253_446800000_BrandMeister_UP_24_baseband.wav

The other two were completely unusable for me, so whatever settings you had on the first sample, those were best for what you were doing. I wonder if using the radio in close proximity to your SDR dongle is causing issues, I often see front end overload occur when users send me samples from their own radios that are neigh undecodable.

```
20240122_141253_446800000_BrandMeister_UP_24_baseband.wav

11:25:18 Sync: +DMR MS/DM MODE/MONO | VLC  
 SLOT 1 TGT=9990 SRC=3025209 FLCO=0x03 FID=0x00 SVC=0x00 Private Call  
 DMR PDU Payload [03][00][00][00][27][06][2E][29][39][D7][C6][2B]
11:25:18 Sync: +DMR MS/DM MODE/MONO | VC* 
 AMBE F86D28598C0C80 err = [0] [0] 
 AMBE F88D2858440980 err = [0] [0] 
 AMBE F87D28586CA880 err = [0] [0] 
 AMBE F88D285844A900 err = [0] [0] 
 AMBE F88C782F849300 err = [0] [0] 
 AMBE F88C785F849B00 err = [0] [0] 
 AMBE F87C785F849900 err = [0] [0] 
 AMBE F87C782E4C9100 err = [0] [0] 
 AMBE F87C72578C7B00 err = [0] [0] 
 AMBE F87632564C6900 err = [0] [0] 
 AMBE F87C78564CB900 err = [0] [0] 
 AMBE F87C785F8C9900 err = [0] [0] 
 AMBE F87C785E4CB980 err = [0] [0] 
 AMBE F88C785844B980 err = [0] [0] 
 AMBE F88C7858449980 err = [0] [0] 
 AMBE F88C7858449980 err = [0] [0] 
 AMBE F88D283F848980 err = [0] [0] 
 AMBE F87D2839AC8A00 err = [0] [0] 
 SLOT 1 TGT=9990 SRC=3025209 FLCO=0x03 FID=0x00 SVC=0x00 Private Call  
 DMR PDU Payload [03][00][00][00][27][06][2E][29][39] SB: 00000000000 - 000; 
 SLOT 1 SB/RC (FEC ERR) E:12; I:1193FADD D:000;  

```
[adry-decoded-audio.zip](https://github.com/DSheirer/sdrtrunk/files/14041180/adry-decoded-audio.zip)

adryd325, 2024-01-24:
I had my RF gain at nearly zero for the third recording, but my radio was fairly close to my SDR so that makes sense.

I can make some more recordings if needed

lwvmobile, 2024-01-24:
I actually had something cross my mind, I went back and played the other two samples that I couldn't decode originally, and was able to decode them when I inverted the polarity manually. I hadn't considered it earlier, but a lot of times when you have a really strong signal, you can see its reflection as a harmonic in another part of the spectrum, and that is usually always inverted in polarity from the original signal. I believe on the second batch of samples, you may have been tuned to the harmonic and not the actual frequency of broadcast. To be honest, I think the sample with the reduced gain may have performed worse than the other one, but again, with you tuned on the harmonic presumably, its hard to say what the correct gain should have been. Another thing you might try is to remove the antenna from the dongle when capturing, see if that helps at close range.

Here are the decoded audio samples, ignore that the recordings start in the middle of the transmission, I had the samples running on a loop and had to get the decode going first before starting the wav file, so starts about half way and then loops back around and repeats.

[ardy-decoded-audio2.zip](https://github.com/DSheirer/sdrtrunk/files/14042713/ardy-decoded-audio2.zip)

adryd325, 2025-02-10:
Fixed in nightly

closed 2025-02-10 by adryd325