DSheirer/sdrtrunk #1811: Compilation instructions

poggs, 2024-01-28:
I'm looking to contribute to sdrtrunk, but having trouble compiling it as it needs JavaFX and Java 20, the latter not an LTS release so tricky to download with `sdkman`.

Do you have a compilation guide?  I'm happy to contribute one with a few pointers!

DSheirer, 2024-01-29:
The wiki has a page with instructions for how to compile via the gradle command line: 
 https://github.com/DSheirer/sdrtrunk/wiki/Developing

I just updated the page to indicate you need JDK 20 with JavaFX (vs JDK 19).

The download links for the Liberica JDK are here: https://bell-sw.com/pages/downloads/?version=java-20

Make sure you change from 'Standard JDK' to 'Full JDK' before you download.  The Full JDK contains the integrated JavaFX libraries.

poggs, 2024-01-29:
Thank you!  Expect some PRs from me in the coming weeks!

closed 2024-01-29 by poggs