DSheirer/sdrtrunk #1815: Event and Message sending to MQTT

poggs, 2024-01-30:
There are occasions where I want to analyse the events and messages being decoded from a channel and use that data in other applications.  For example, to trigger an action if a GPS position report from a radio is within a certain geofence, or to collect statistics on the number of calls placed between users and to talkgroups.

The data in the Events and Messages tab scrolls rapidly and although it is logged to files on disk, picking up these files and sending them elsewhere for processing is cumbersome, so I would like a means of sending different types of message, as well as data from undefined messages, to an external MQTT or similar server for analysis.

This feature could further be extended to send MP3 recordings of voice messages over MQTT, and also extended to support other protocols such as AMQP, Stomp and OpenWire if there is interest.

I am actively working on this feature now and should have a PR ready in a week or so.

tadscottsmith, 2024-01-31:
This would be a great addition!

poggs, 2024-02-05:
For information, my WIP fork is at https://github.com/poggs/sdrtrunk/tree/mqtt-support

poggs, 2024-02-19:
Work on this feature is going well - I have the preference panes and MQTT connection logic working.  I need to make logging more granular than just a string generated by toString().

Is it a good idea to extend IMessage to add a 'toDetailedLoggableMessage' which outputs an instance of a new class, DetailedLoggableMessage, which is populated with some basic fields (timestamp, message type, string representation) but also a HashMap which each class populates with variables of interest?  We would get basic message details logged and could go through the codebase and implement a toDetailedLoggableMessage() in each relevant message class to produce enhanced logging with data that's parseable by code.

@DSheirer, any thoughts?

DSheirer, 2024-02-20:
It may help to better understand what you're planning to implement and the scope of messages and events that you're planning to target.

If you're planning to target every decoded message for export via MQTT, then it may make sense to modify the IMessage interface.  However, that's a lot of classes and adding code to each of those classes to create a custom formatted representation of that message content is a significant amount of effort.  You might consider using Jackson JSON annotations and serialize the messages to JSON so that you can take advantage of the existing class structures.

However, if you're only looking to export a subset of the messages using a custom export format, then it may make sense to create a new interface (IExportable) with the method signatures that you need to create representational instances of your MQTT messages and that way you can simply add the new interface to those messages that you want to export.
 
Same questions apply to events.  Are you going to export all events or a subset?