DSheirer/sdrtrunk #1818: Can't shrink playlist editor, Can't reach bottom options anymore. Makes playlist unusable.

Ctpolicelive, 2024-02-03:
**sdrtrunk Version**

Masterbranch

**Describe the bug**
I cannot adjust the size of the playlist editor and it goes below my screen. I tried everything to be able to get to the bottom border so i can pull up the bottom options. I no longer can.

**To Reproduce**
Happens everytime, no matter what I do. 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (optional - complete the following information):**
 - Linux Mint

**Additional context**
Very frustrating. I cannot enable streams since I can't get to the bottom. I have feeds off and no way I can turn them back on.  Playlist editor window has no controls to adjust it to fit the screen. 

[image]

Ctpolicelive, 2024-02-03:
Found a work around for now. I change my display rotation setting to Portrait Right or Left. Then I can slide the slider "up" , then I must go back to display rotation Landscape and it fixes it, until I restart anyway.

hans109h, 2024-02-04:
1.  Are you able to maximize/minimize the window using keyboard shortcuts super+up and super+down?
2.  Is this on a multi-monitor setup?  Does it display correctly if you move it to an alternate display?  You might need to use shift+super+{left,right}
3.  Are you unable to resize the window at all by using the mouse at the edge of the window?
4.  If you open the playlist editor and close it, does reopening it behave the same?

KmanOz, 2024-02-12:
The workaround for this is to move the Window to another desktop on Linux. I have the same issue but seems to be better on the nightly builds compared top the release.

disonesto, 2024-05-04:
The best work around I have found for this is alt + spacebar.   Then use the arrows to select move.    Then click enter.  Then use the arrows again to move the window around.  Then press enter to keep the window in that spot you moved the window too .   Then to move it back same thing again alt + spacebar........ repeat steps

transorsmth, 2024-10-03:
I am also affected by this issue. Not only can I not resize it with any means other than dragging the edge of the window with my cursor, but there seems to be a minimum size of the window, that is taller than my screen. Other application windows work fine, and I can post a recording of attempting to resize the window for debugging purposes if it'd help.

DSheirer, 2024-10-27:
Fixed.

closed 2024-10-27 by DSheirer