DSheirer/sdrtrunk #1823: Event log grows past settings threshold

IMG20191023, 2024-02-04:
**sdrtrunk Version**
0.6.0

**Describe the bug**
Event log continues to grow even when i have a limit set in the settings.

**To Reproduce**
Set event log size in settings,
watch event log grow past this limit as calls come in.

**Expected behavior**
Event log to be flushed when limit reached.

**Screenshots**
[image]

**Application Log**
[20240202_sdrtrunk_app.log](https://github.com/DSheirer/sdrtrunk/files/14156457/20240202_sdrtrunk_app.log)

DSheirer, 2024-02-04:
The 'Event Logs' and 'Recordings' settings correspond to the indicators at the bottom of the application with the same names.  They're simply meant to show the user when their logs and recordings sizes are using a lot of disk space so that the user can take care of it and not run out of disk space.

I can see how the labeling of the settings in the user preferences isn't quite clear ... I'll work to clarify that wording.

IMG20191023, 2024-02-06:
Great, thanks for the feedback.
How do i go about reducing the logging so the event log doesn't grow so big so quickly.?

DSheirer, 2024-02-09:
I don't think you can selectively filter the log file output.

Closing.

closed 2024-02-09 by DSheirer