DSheirer/sdrtrunk #1826: Cannot decode MPT-1327 FILES

EmmaChissit, 2024-02-07:
**sdrtrunk Version**
Indicate which sdrtrunk version you are using.  Use 'master-branch' if you are running the latest code.
V0.6.0

**Describe the bug**
A clear and concise description of what the bug is.
Have set up everything to decode MPT-1327 SIGNALS
When I press the green play icon, I would expect an arrow to appear in the playing column appropriate to that site. this does not happen

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.
[image]

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Application Log**
If applicable, include either the full application log,or a snippet of the error logging.

**Desktop (optional - complete the following information):**
 - OS: [e.g. iOS]
 - CPU Cores: [e.g. 4 cores]
 - RAM:[e.g. 16GB]

**Additional context**
Add any other context about the problem here.
Dennis replied to my facebook page with how to fix problem:Disable SIMD & vector processing in user preferences. This action fixed the problem
[image]

D

DSheirer, 2024-02-07:
Reopening the issue.  The stack trace (error printout) indicates there's an issue in one of the components that needs resovled.

DSheirer, 2024-02-09:
@EmmaChissit the fix was merged into the master branch and once the nightly build updates, you can use the new version and turn your Vector / SIMD processing back on to make the application run more efficiently.  Please let me know if it continues to generate errors for you.

PR #1832 (2024-02-09): #1826 SIMD-512 Systems Fail When Decoding MPT1327
Resolves issue in MPT1327 decoder for users with vector SIMD enabled on a SIMD-512 machine.

The Correlator was unable to generate the reference samples for the specified sample period.  Updated code to generate a sampe array aligned to the architecture and then simply trucate the returned array size.

Closes #1826
closed 2024-02-09 by DSheirer